evcc-io/evcc #647: Phase switches from 1p to 3p

martink173, 2021-02-04:
**Describe the bug**
If I do read the log correctly, the charging process started using single phase charging (1p), after pausing due to threshold, charging continued in triple phase mode (3p).

**To Reproduce**
Steps to reproduce the behavior:
1. Configure 1p mode in yaml
2. Start evcc

**Expected behavior**
A clear and concise description of what you expected to happen.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
<- paste here

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc log output with <code>--log trace</code>:</summary><br><pre>
Feb  4 10:34:31 charger evcc[3636]: [lp-1  ] DEBUG 2021/02/04 10:34:31 max charge current: 6.5A = 0A + 6.5A (-1492W @ 1p)
Feb  4 10:34:31 charger evcc[3636]: [lp-1  ] DEBUG 2021/02/04 10:34:31 pv max charge current: 7A
Feb  4 10:34:45 charger evcc[3636]: [lp-1  ] DEBUG 2021/02/04 10:34:45 charge power: 1300W
Feb  4 10:34:47 charger evcc[3636]: [lp-1  ] DEBUG 2021/02/04 10:34:47 charger status: C
Feb  4 10:34:49 charger evcc[3636]: [lp-1  ] DEBUG 2021/02/04 10:34:49 charge start energy: 258kWh
Feb  4 10:34:49 charger evcc[3636]: [lp-1  ] INFO 2021/02/04 10:34:49 start charging ->
Feb  4 10:34:52 charger evcc[3636]: [lp-1  ] DEBUG 2021/02/04 10:34:52 vehicle soc: 85%
Feb  4 10:34:52 charger evcc[3636]: [lp-1  ] DEBUG 2021/02/04 10:34:52 vehicle range: 227km
Feb  4 10:34:55 charger evcc[3636]: [lp-1  ] DEBUG 2021/02/04 10:34:55 detected phases: 3p [6.58 6.56 6.61]A
Feb  4 10:34:55 charger evcc[3636]: [lp-1  ] DEBUG 2021/02/04 10:34:55 max charge current: 6.8A = 7A + -0.18A (124W @ 3p)

</pre></details>

**If using Docker:**
Show output of <code>docker run andig/evcc -v</code>:
```
<- paste here

```

<details><summary>Show evcc log output with <code>docker logs <container id></code>:</summary><br><pre>
<- paste here

</pre></details>

closed 2021-02-04 by andig

premultiply, 2021-02-04:
EVCC erkennt anhand der Zählerdaten dass der Nutzer hier die falsche Vorgabe eingestellt hat.
Funktioniert perfekt.

martink173, 2021-02-04:
Wird der Modus ausschließlich durch den Ladekabeladapter vom NRGKick bestimmt?

premultiply, 2021-02-04:
Nein, aber durch die vom Fahrzeug auf den einzelnen Phasen bezogenen Ströme.
Wenn du ein dreiphasiges Fahrzeug mit drei Phasen versorgst wird es auch mit drei Phasen laden.
Wenn du nur eine Phase versorgst schaltet das Fahrzeug auf einphasige Ladung um (empfohlen!).
Die verwendeten Steckverbinder sind dabei nur nebensächlich. Natürlich kann man aber mit einem einphasigen Steckverbinder grundsätzlich keine drei Phasen versorgen.

martink173, 2021-02-05:
Ich dachte mit der Einstellung in EVCC gebe ich das vor ("1p"). Wozu dient "phases" dann? Wie erreiche ich einphasiges Laden?

premultiply, 2021-03-03:
In dem du L2 und L3 abklemmst.

andig, 2025-10-06:
Was auch immer hier das Thema war ist seit mindestens 4 Jahren gelöst ;)