evcc-io/evcc #656: Hyundai API sendet sporadisch SoC=0

xerion3800, 2021-02-11:
**Describe the bug**
Zweitweise sendet die Hyundai API einen SoC von Null zurück. Diese führt dazu, dass EVCC diesen Wert entsprechend als Valide einstuft. Dadurch wird dann beim Ladestart nicht mit dem PV-Überschuss geladen sondern mit der `maxCurrent `da der Wert unter `minSoC` liegt.

**To Reproduce**

- PollMode: charging oder connected einstellen
- EV verbunden
- Laden starten

**Expected behavior**
Ein SoC von Null sollte als nicht valide eingestuft werden und nicht verarbeitet werden. Wenn Null zurück kommt, sollte EVCC diesen SoC Wert verwerfen und mit dem zuletzt bekannten SoC weiter arbeiten. Nach kurzer Wartezeit sollte EVCC erneut versuchen den SoC abzufragen.

**EVCC details:**
evcc version 0.44 (8b3ad1a)

zusätzliche Infos unter https://evccgroup.slack.com/archives/C01321PUJAD/p1612954732324400

andig, 2021-02-11:
> Ein SoC von Null sollte als nicht valide eingestuft werden und nicht verarbeitet werden.

Leider ist das spezifisch für Hyundai.

Lösungsansatz: bei 0 selektiv einen Fehler zurück geben und API erneut abfragen.

xerion3800, 2021-02-11:
> Leider ist das spezifisch für Hyundai.
> 
> Lösungsansatz: bei 0 selektiv einen Fehler zurück geben und API erneut abfragen.

Wichtig ist aber, dass die 0 nicht von EVCC verarbeitet wird, sonst greift sofort `minSoC`.
Wenn es Hyundai Spezifisch ist vielleicht wäre nur hier der Workaround angebracht?

andig, 2021-02-11:
Ich überlege, die Lösung gemeinsam mit https://github.com/andig/evcc/issues/488 für Nissan zu machen. Es sollte für ein vehicle API möglich sein, EVCC mitzuteilen, dass die Aktualisierung wiederholt werden muss.

andig, 2021-02-28:
Seit https://github.com/andig/evcc/pull/699 leider obsolet

closed 2021-02-28 by andig