evcc-io/evcc #661: Nissan Leaf zeigt kein SOC nach erneuten Anstecken an. Ladevorgang beginnt nicht.

martinez81, 2021-02-11:
**Describe the bug**
Fahrzeug: Nissan Leaf
Nach erneuten Anstecken des Fahrzeuges wird der SOC nicht angezeigt und wahrscheinlich nicht aktualisiert. Dadurch beginnt der Ladevorgang nicht.
Erst wenn man das Ladeziel auf 100% setzt beginnt der Ladevorgang. Auch jetzt wird der SOC weder angezeigt noch aktualisiert. In der Nissan App kann der SOC aktualisiert werden.
EVCC Neustart bewirkt, dass es wieder normal funktioniert.

**Expected behavior**
SOC sollte bei angesteckten Fahrzeug immer angezeigt und aktualisiert werden.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.44 (8b3ad1a)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
Unsupported Config Type ""

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

# log settings
log: error
levels:
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions
meters:
- name: sma-grid
  type: sma
  serial: 3006138723
- name: sma-pv
  type: modbus
  model: sunspec
  uri: 192.168.0.165:502
  id: 126
  power: Power
  energy: Sum

# charger definitions
chargers:
- name: GO-eCharger1
  type: go-e # GO-eCharger
  uri: http://192.168.0.192
- name: GO-eCharger2
  type: go-e
  uri: http://192.168.0.193

# vehicle definitions
vehicles:
 - name: model3
   type: tesla
   title: Model 3
   capacity: 79
   user: xxx # your email address used on Tesla website
   password: xxx # your password used on Tesla website password
   vin: xxx # the VIN of you car (can be obtained from Tesla website)
   tokens:
     access: xxx
     refresh: xxx
   cache: 5m

 - name: nissan
   type: nissan
[... 2115 lines cut ...]
Feb 11 16:59:08 Docker evcc[3547]: [lp-1  ] WARN 2021/02/11 16:59:08 updating soc failed: unexpected status: 401
Feb 11 16:59:08 Docker evcc[3547]: [lp-1  ] ERROR 2021/02/11 16:59:08 vehicle error: unexpected status: 401
Feb 11 16:59:08 Docker evcc[3547]: [lp-1  ] DEBUG 2021/02/11 16:59:08 detected phases: 1p [16 0 0]A
Feb 11 16:59:18 Docker evcc[3547]: [lp-2  ] DEBUG 2021/02/11 16:59:18 charge power: 0W
Feb 11 16:59:18 Docker evcc[3547]: [lp-2  ] DEBUG 2021/02/11 16:59:18 charger status: B
Feb 11 16:59:28 Docker evcc[3547]: [lp-1  ] DEBUG 2021/02/11 16:59:28 charge power: 3560W
Feb 11 16:59:28 Docker evcc[3547]: [lp-1  ] DEBUG 2021/02/11 16:59:28 charger status: C
Feb 11 16:59:28 Docker evcc[3547]: [lp-1  ] WARN 2021/02/11 16:59:28 updating soc failed: unexpected status: 401
Feb 11 16:59:28 Docker evcc[3547]: [lp-1  ] ERROR 2021/02/11 16:59:28 vehicle error: unexpected status: 401
Feb 11 16:59:28 Docker evcc[3547]: [lp-1  ] DEBUG 2021/02/11 16:59:28 detected phases: 1p [15.9 0 0]A
Feb 11 16:59:38 Docker evcc[3547]: [lp-2  ] DEBUG 2021/02/11 16:59:38 charge power: 0W
Feb 11 16:59:38 Docker evcc[3547]: [lp-2  ] DEBUG 2021/02/11 16:59:38 charger status: B
Feb 11 16:59:48 Docker evcc[3547]: [lp-1  ] DEBUG 2021/02/11 16:59:48 charge power: 3560W
Feb 11 16:59:48 Docker evcc[3547]: [lp-1  ] DEBUG 2021/02/11 16:59:48 charger status: C
Feb 11 16:59:48 Docker evcc[3547]: [lp-1  ] WARN 2021/02/11 16:59:48 updating soc failed: unexpected status: 401
Feb 11 16:59:48 Docker evcc[3547]: [lp-1  ] ERROR 2021/02/11 16:59:48 vehicle error: unexpected status: 401
Feb 11 16:59:48 Docker evcc[3547]: [lp-1  ] DEBUG 2021/02/11 16:59:48 detected phases: 1p [15.8 0 0]A
Feb 11 16:59:58 Docker evcc[3547]: [lp-2  ] DEBUG 2021/02/11 16:59:58 charge power: 0W
Feb 11 16:59:58 Docker evcc[3547]: [lp-2  ] DEBUG 2021/02/11 16:59:58 charger status: B
lines 1980-2009/2009 (END)
pi@Docker:~ $ 

</pre></details>

**systemd zum ausführen von Evcc benutzt**

closed 2021-02-11 by martinez81

andig, 2021-05-13:
@martinez81  ich würde gerne den SoC Refresh asynchron machen, dafür bräuchte ich Zugangsdaten zum Testen. Falls Du helfen willst gerne per Mail melden (cpuidle@gmx.de).

martinez81, 2021-05-13:
Hi Andreas!

Ja, du kannst den Zugang nochmal nutzen.

E-Mail:
***@***.***

Passwort:
Evcc!2021

Viele Grüße 
Martin

andig, 2021-05-13:
Super- PW habe ich entfernt. Weisst Du evtl, wie ich den Status eines Refreshes anhand der ID abfragen kann?

    {"data":{"type":"RefreshBatteryStatus","id":"5638e62f-8aa2-4ed8-9575-39445bf36cbc","attributes":{}}}

Falls nein müsste ich mich einfach am Timestamp des Ergebnisses orientieren:

    {
        "data": {
            "type": "Car",
            "id": "XXX",
            "attributes": {
                "rangeHvacOn": 163,
                "chargeStatus": 1,
                "timeRequiredToFullNormal": 450,
                "timeRequiredToFullSlow": 0,
                "plugStatusDetail": 1,
                "rangeHvacOff": 169,
                "batteryLevel": 65,
                "plugStatus": 1,
                "chargePower": 2,
                "lastUpdateTime": "2021-05-13T11:13:08Z",
                "timeRequiredToFullFast": 0,
                "batteryCapacity": 40000,
                "batteryBarLevel": 156
            }
        }
    }

martinez81, 2021-05-13:
Nein, tut mir leid.
Wenn du die ID zum einloggen meinst, geht das nicht.
Der Login mit einer ID ist ein altes System BJ vor 2019.

Gruß 
Martin