evcc-io/evcc #670: Can we add details on KEBA support

swernerx, 2021-02-15:
I was wondering whether C and X-series of KEBA work equally well. We are planning to buy two wallboxes with each 11KW power. These typically work in some master-slave fashion e.g. one x-series, one c-series. Is that approach required for EVCC or does it work equally well when using two of the cheaper c-series?

That detail should probably be added to the docs as well.

Thanks for your excellent work. Very much appreciated for sure!

andig, 2021-02-16:
If both have the UDP api (and I think they have) then both should work. If you need master-slave then abviously its the X as evcc at this time does not manage per-phase currents.

andig, 2021-02-21:
Think this has been answered.

closed 2021-02-21 by andig