evcc-io/evcc #672: SHM: PlanningRequest auch wenn disconnected

premultiply, 2021-02-18:
**Describe the bug**
Wenn kein Fahrzeug (mehr) verbunden ist wird an den Sunny Home Manager via SEMP (/semp) trotzdem weiterhin ein PlanningRequest anhand des zuletzt bekannten SOC gestellt.

**To Reproduce**
1. Fahrzeug mit SOC laden lassen
2. Fahrzeug trennen
3. unter /semp den weiter vorhandenen PlanningRequest sehen

**Expected behavior**
Wenn kein Fahrzeug mit dem Ladepunkt verbunden ist (**disconnected**) wird **kein PlanningRequest** gemeldet.
Während der der Zustände **connected** und **charging** ist der PlanningRequest vorhanden.

andig, 2021-02-18:
Good catch. Brauchst Du Hilfe?

premultiply, 2021-02-18:
Ja, bitte. Ich habs zwar angesehen und ich vermute dass nur irgendwo ein minimaler Eingriff notwendig wäre aber ich sehe es nicht.

PR #673 (2021-02-23): SEMP: check connected
Fixes #672 when ready to merge
closed 2021-02-23 by andig