evcc-io/evcc #674: Steuerung per REST API

ilosuna, 2021-02-19:
Laut [README](https://github.com/andig/evcc#api) müsste sich EVCC folgendermaßen per REST API mit cURL steuern lassen (z.B. auf "Nur PV" stellen; andere Modi mit "off", "now" und "minpv" statt "pv"):

`curl -s -X POST http://localhost:7070/api/mode/pv`

Ich meine, das ging in einer der vorigen Versionen auch mal so. Im aktuellen Release (0.46) habe ich bemerkt, dass eine solche Anfrage mit "404 page not found" beantwortet wird.

Stattdessen geht es mit z.B. mit folgendem Befehl:

`curl -s -X POST http://localhost:7070/api/loadpoints/0/mode/pv`

Damit wird, laut README, nur der Loadpoint 0 gesteuert. Eine allgemeine Steuerung ("global charge mode") scheint jedoch nicht mehr möglich zu sein. Ist das so beabsichtigt und nur in der README noch erwähnt?

andig, 2021-02-19:
Fehler in der README, danke für den Hinweis!

andig, 2021-02-19:
https://github.com/andig/evcc/commit/b2ddbd4ff7f029445de0f11883e61cb24c26ac71

closed 2021-02-19 by andig