evcc-io/evcc #679: Need advice on how to start evcc on macOS

bicaluv, 2021-02-21:
**Is your feature request related to a problem? Please describe.**
I downloaded the evcc_0.46_macOS_64-bit.tar.gz file and unpacked it.
I moved too the directory created in the Downloads folder and tried to start ivcc with
`./evcc -h` and even with
`sudo ./evcc -h`
All I get is this warning dialog:
[image]

**Describe the solution you'd like**
Having a more detailed description for macOS would be great.
Or even better: get a verified /trusted evcc application.

**Additional context**
Using macOS 10.15.7

andig, 2021-02-21:
Shift-Right Mouse-Open?

andig, 2021-02-21:
Another option would be Apple developer account (100$/a) and notarizing during build process.

bicaluv, 2021-02-21:
> Shift-Right Mouse-Open?

Works when simply trying to start, but not when trying to pass parameters.

bicaluv, 2021-02-21:
> Another option would be Apple developer account (100$/a) and notarizing during build process.

Maybe any of the other contributors is already an Apple dev?

A simply solution would still be to add some more words for starting using macOS

andig, 2021-02-21:
> Maybe any of the other contributors is already an Apple dev?

Not that I'm aware of. If you want to become a Github Sponsor I would look into notarizing the build.

andig, 2021-02-28:
No sponsor available, closing.

closed 2021-02-28 by andig