evcc-io/evcc #685: Add VW/Skoda ID Charger (EVBox) support

swernerx, 2021-02-22:
I think it would be great if the wallboxes of the VW Group (ID Charger, https://charging-energy.elli.eco/) from the manufacturer ELLI (hardware supplier is the company EVBox) would also work with EVCC. These are very inexpensive for the range of functions offered. It is certainly not entirely unrealistic to believe that these will represent a decent share of the market in the near future. Starting with the "Connect" wallbox (starting at 600€), these are also controllable. The devices are connected like WLAN or LAN. The control interface is OCPP 2.0-J (JSON API) or later EEBus 4. Sounds good, doesn't it?

DerAndereAndi, 2021-02-22:
Hi,

As of right now this is sadly not possible (I own such a box myself):

- OCCP: This interface is used for their mobile app and there is no option available (Web UI, API, ...) to change the server to be used to a different one
- EEBUS: Support for EEBUS for the devices is currently only announced, previously for 2020, now for somewhen in 2021. Also there is no open source EEBUS support available anywhere yet, and that protocol is not an easy task. But this might be the only option to be viable for supporting these chargers. Maybe this is something we can achieve, but it will also take quite some time.
- There is no other way to control these boxes, the web interface doesn’t provide any options to change the amperage nor is there any other means of API to do so.

So obviously these devices are on the future wish list, but as of now it is impossible to say when EVCC will be able to support them.

andig, 2021-02-22:
https://github.com/andig/evcc/issues/591#issuecomment-760892209

swernerx, 2021-02-22:
I understand that there might not be any influence of you @andig - still we could keep that open for tracking any progress.

I talked today with the official support which lead me to this document: https://uploads-ssl.webflow.com/5f2271e9820e0bb93e6f3499/5f6a7e1113e3e52cf745fda1_OverloadProtection_Guideline.pdf

> The Overload Protection feature of the Elli Charger is aiming to allow customers to use as much of their facility’s power capacity as possible.

...

> With the Overload Protection, the Elli Charger offers a comfort feature to enable customers to always charge their vehicles at the fastest speed possible. By installing additional current sensors (Current Transformer coils = “CT coils”) in the house’s electric installation and configuring the Elli Charger according to the local restrictions, the Elli Charger will charge the vehicle with as much power as possible while at the same time ensuring to stay within these local restrictions.

That's not related to EVCC anyhow. Still this functionality might allow configuring the Wallbox to offer some kind of flexible loading based on PV availability.

andig, 2021-02-22:
What we'd need is a) an open programming interface and b) vendor support. None of both is unfortunately provided in the pdf. As long as that's the case the only sensible solution is to chose a wallbox according to personal need and if need is PV-steering and vehicle integration then not to chose Elli.

closed 2021-03-10 by andig

nkrepo, 2022-09-16:
Hi @andig, 
recently in the [software release logs](https://www.elli.eco/en/changelog/wallbox) of the ID.Charger software, I noticed that the EEbus protocol was implemented.

 `EEBus communication interface implemented (allows you to connect your wallbox to EEBus-enabled devices. For more information on EEBus please see the FAQs in our shop or visit the official EEBus website which you can find at [eebus.org](https://www.eebus.org/))`

Also at the [charger product page](https://www.elli.eco/en/charger#techspecs) now reports:
`Communication protocols : OCPP 2.0 & EEBUS`