evcc-io/evcc #694: KIA SoC not loaded

querdenker2k, 2021-02-24:
**Describe the bug**
When i start evcc, it's trying to get the soc from the car, but fails with 401.
2 Days ago it worked. Over the UVO App it's still working.

**To Reproduce**
Steps to reproduce the behavior:
Configure Kia Car and start evcc

**Expected behavior**
SoC should be loaded

**EVCC details:**
Show output of <code>evcc -v</code>:
```
<- paste here

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
vehicles:
- name: e-Niro
  type: kia
  title: e-Niro # display name for UI
  capacity: 64 # kWh
  user: hidden
  password: hidden

</pre></details>

<details><summary>Show evcc log output with <code>--log trace</code>:</summary><br><pre>

[lp-1  ] WARN 2021/02/24 22:02:30 updating soc failed: unexpected status: 401
[lp-1  ] ERROR 2021/02/24 22:02:30 vehicle error: unexpected status: 401

</pre></details>

**If using Docker:**
Show output of <code>docker run andig/evcc -v</code>:
```
evcc version 0.47 (1634d4e)

```

<details><summary>Show evcc log output with <code>docker logs <container id></code>:</summary><br><pre>

[lp-1  ] WARN 2021/02/24 22:02:30 updating soc failed: unexpected status: 401
[lp-1  ] ERROR 2021/02/24 22:02:30 vehicle error: unexpected status: 401

</pre></details>

anrichter, 2021-02-25:
Das könnte ein Fehler in der BlueLink Verbindung sein. Das gleich Verhalten beobachte ich seit 2 Tagen bei meinem Hyundai.

andig, 2021-02-25:
Depends on https://github.com/Hacksore/bluelinky/issues/101

closed 2021-02-26 by andig

andig, 2021-03-10:
See https://github.com/andig/evcc/pull/741

querdenker2k, 2021-03-10:
Ich bin begeistert dass das so schnell wieder gefixt wurde.

andig, 2021-03-11:
Wenn du magst kannst du das Projekt bei Github Sponsors unterstützen!

PR #699 (2021-02-26): Withdraw Kia and Hyundai
Fix #694