evcc-io/evcc #706: SEMP: Disable Control without vehicle

premultiply, 2021-02-28:
Wird der SHM mit allowcontrol: true eingebunden aber kein Vehicle konfiguriert ist kein PV-Laden mehr möglich, da EVCC ohne SoC keinen PlanningRequest stellen kann (unbekannte Energiemenge) und der SHM ohne PlanningRequest die Ladung sperrt (kein Bedarf).

Ohne konfiguriertes Vehicle muss daher via SEMP EMSignalsAccepted immer als false zurückgeben werden (wie wenn allowcontrol: false).

andig, 2021-02-28:
Wie wärs mit festem Wert, z.B. 1kWh?

andig, 2021-02-28:
@premultiply machst Du einen PR?

premultiply, 2021-03-09:
Ich glaube ich bin ich damit überfordert das angemessen zu lösen.

andig, 2021-03-11:
Hier fehlt Feedback. Grundsatzfrage: was ist zu planen wenn wir den soc eines Fahrzeugs nicht kennen sondern nur den Zustand des Chargers? Bitte um Spezifikation.

premultiply, 2021-03-11:
Oh, hab ich irgendwie übersehen.
Naja im Prinzip wie oben beschrieben. Wenn kein Fahrzeug konfiguriert dann muss nur immer EMSignalsAccepted false zurückgeben. Eigentlich ganz simpel.

premultiply, 2021-03-11:
https://github.com/andig/evcc/blob/8b794382849f206c678a1cd84d55ef53712f8c8c/hems/semp/semp.go#L371

PR #745 (2021-03-11): Disable remote control without configured vehicle
Fix #706
closed 2021-03-11 by andig