evcc-io/evcc #707: go-e charger amp and amx values set to control charging current

profigriller, 2021-02-28:
I am testing evcc with go-e charger.
I noticed that both values amp and amx are set to control current. It would be good not to write to amp but only to amx to protect the eeprom
I have verified this looking at the modbus values reported by the go charger in openhab reading the modbus values 299 and 300
In the log it is possible to see that amp value is changing as soon as amx changes

2021-02-28T07:45:43.568658348Z {"version":"B","tme":"2802210845","rbc":"15","rbt":"14413103","car":"4","amp":"16","err":"0","ast":"0","alw":"0","stp":"0","cbl":"32","pha":"8","tmp":"2","tma":[12.38,12.13,11.88,12.50],"amt":"32","dws":"0","dwo":"0","adi":"1","uby":"0","eto":"280","wst":"3","txi":"0","nrg":[221,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0],"fwv":"040.0","sse":"02ABCD","wss":"JLFXXXX","wke":"*********","wen":"1","cdi":"0","tof":"101","tds":"1","lbr":"56","aho":"3","afi":"7","azo":"0","ama":"16","al1":"6","al2":"8","al3":"10","al4":"12","al5":"16","cid":"255","cch":"65535","cfi":"65280","lse":"1","ust":"0","wak":"3090a6849b","r1x":"2","dto":"0","nmo":"0","sch":"AAAAAAAAAAAAAAAA","sdp":"0","eca":"0","ecr":"0","ecd":"0","ec4":"0","ec5":"0","ec6":"0","ec7":"0","ec8":"0","ec9":"0","ec1":"0","rca":"04AB4552","rcr":"","rcd":"","rc4":"","rc5":"","rc6":"","rc7":"","rc8":"","rc9":"","rc1":"","rna":"","rnm":"","rne":"","rn4":"","rn5":"","rn6":"","rn7":"","rn8":"","rn9":"","rn1":"","loe":0,"lot":0,"lom":0,"lop":0,"log":"","lon":0,"lof":0,"loa":0,"lch":0,"mce":1,"mcs":"192.168.123.12","mcp":1883,"mcu":"JLE","mck":"lobster1","mcc":1}
2021-02-28T07:45:43.568841754Z [lp-1  ] TRACE 2021/02/28 08:45:43 charge currents: [0 0 0]A
2021-02-28T07:45:43.569245120Z [lp-1  ] DEBUG 2021/02/28 08:45:43 max charge current: 6.8A = 0A + 6.8A (-1564W @ 1p)
2021-02-28T07:45:43.569312193Z [lp-1  ] DEBUG 2021/02/28 08:45:43 site power -1564W < enable threshold 0W
2021-02-28T07:45:43.569360600Z [lp-1  ] DEBUG 2021/02/28 08:45:43 pv enable timer elapsed
2021-02-28T07:45:43.569407136Z [lp-1  ] DEBUG 2021/02/28 08:45:43 pv max charge current: 6A
2021-02-28T07:45:43.569480617Z [lp-1  ] DEBUG 2021/02/28 08:45:43 max charge current: 6
2021-02-28T07:45:43.569527172Z [go-e  ] TRACE 2021/02/28 08:45:43 GET /mqtt?payload=amx=6 HTTP/1.1
2021-02-28T07:45:43.569573653Z Host: 192.168.123.83
2021-02-28T07:45:43.569779225Z [cache ] TRACE 2021/02/28 08:45:43 chargeCurrents: [0 0 0]
2021-02-28T07:45:43.569850131Z [cache ] TRACE 2021/02/28 08:45:43 timerSet: false
2021-02-28T07:45:43.631371207Z [go-e  ] TRACE 2021/02/28 08:45:43 HTTP/1.1 200 OK
2021-02-28T07:45:43.632188791Z Connection: close
2021-02-28T07:45:43.632395233Z Content-Length: 1152
2021-02-28T07:45:43.632444677Z Access-Control-Allow-Origin: *
2021-02-28T07:45:43.632540639Z Content-Type: application/json
2021-02-28T07:45:43.632565897Z 
2021-02-28T07:45:43.632605656Z {"version":"B","tme":"2802210845","rbc":"15","rbt":"14413162","car":"4","amp":"6","err":"0","ast":"0","alw":"0","stp":"0","cbl":"32","pha":"8","tmp":"2","tma":[12.38,12.13,11.88,12.50],"amt":"32","dws":"0","dwo":"0","adi":"1","uby":"0","eto":"280","wst":"3","txi":"0","nrg":[221,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0],"fwv":"040.0","sse":"02ABCD","wss":"JLFXXXX","wke":"*********","wen":"1","cdi":"0","tof":"101","tds":"1","lbr":"56","aho":"3","afi":"7","azo":"0","ama":"16","al1":"6","al2":"8","al3":"10","al4":"12","al5":"16","cid":"255","cch":"65535","cfi":"65280","lse":"1","ust":"0","wak":"3090a6849b","r1x":"2","dto":"0","nmo":"0","sch":"AAAAAAAAAAAAAAAA","sdp":"0","eca":"0","ecr":"0","ecd":"0","ec4":"0","ec5":"0","ec6":"0","ec7":"0","ec8":"0","ec9":"0","ec1":"0","rca":"04AB4552","rcr":"","rcd":"","rc4":"","rc5":"","rc6":"","rc7":"","rc8":"","rc9":"","rc1":"","rna":"","rnm":"","rne":"","rn4":"","rn5":"","rn6":"","rn7":"","rn8":"","rn9":"","rn1":"","loe":0,"lot":0,"lom":0,"lop":0,"log":"","lon":0,"lof":0,"loa":0,"lch":0,"mce":1,"mcs":"192.168.123.12","mcp":1883,"mcu":"JLE","mck":"lobster1","mcc":1}
2021-02-28T07:45:43.632670415Z [lp-1  ] DEBUG 2021/02/28 08:45:43 charger enable
2021-02-28T07:45:43.632692989Z [go-e  ] TRACE 2021/02/28 08:45:43 GET /mqtt?payload=alw=1 HTTP/1.1
2021-02-28T07:45:43.632714822Z Host: 192.168.123.83
2021-02-28T07:45:43.632736173Z [cache ] TRACE 2021/02/28 08:45:43 chargeCurrent: 0
2021-02-28T07:45:43.716096840Z [go-e  ] TRACE 2021/02/28 08:45:43 HTTP/1.1 200 OK
2021-02-28T07:45:43.716199042Z Connection: close
2021-02-28T07:45:43.716245505Z Content-Length: 1152
2021-02-28T07:45:43.716269505Z Access-Control-Allow-Origin: *
2021-02-28T07:45:43.716312226Z Content-Type: application/json
2021-02-28T07:45:43.716336967Z 
2021-02-28T07:45:43.716369485Z {"version":"B","tme":"2802210845","rbc":"15","rbt":"14413237","car":"4","amp":"6","err":"0","ast":"0","alw":"1","stp":"0","cbl":"32","pha":"8","tmp":"2","tma":[12.38,12.13,11.88,12.50],"amt":"32","dws":"0","dwo":"0","adi":"1","uby":"0","eto":"280","wst":"3","txi":"0","nrg":[221,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0],"fwv":"040.0","sse":"02ABCD","wss":"JLFXXXX","wke":"*********","wen":"1","cdi":"0","tof":"101","tds":"1","lbr":"56","aho":"3","afi":"7","azo":"0","ama":"16","al1":"6","al2":"8","al3":"10","al4":"12","al5":"16","cid":"255","cch":"65535","cfi":"65280","lse":"1","ust":"0","wak":"3090a6849b","r1x":"2","dto":"0","nmo":"0","sch":"AAAAAAAAAAAAAAAA","sdp":"0","eca":"0","ecr":"0","ecd":"0","ec4":"0","ec5":"0","ec6":"0","ec7":"0","ec8":"0","ec9":"0","ec1":"0","rca":"04AB4552","rcr":"","rcd":"","rc4":"","rc5":"","rc6":"","rc7":"","rc8":"","rc9":"","rc1":"","rna":"","rnm":"","rne":"","rn4":"","rn5":"","rn6":"","rn7":"","rn8":"","rn9":"","rn1":"","loe":0,"lot":0,"lom":0,"lop":0,"log":"","lon":0,"lof":0,"loa":0,"lch":0,"mce":1,"mcs":"192.168.123.12","mcp":1883,"mcu":"JLE","mck":"lobster1","mcc":1}
2021-02-28T07:45:43.716494891Z [lp-1  ] DEBUG 2021/02/28 08:45:43 charger enable

[8a90726ff53c.log.zip](https://github.com/andig/evcc/files/6056406/8a90726ff53c.log.zip)

andig, 2021-02-28:
> In the log it is possible to see that amp value is changing as soon as amx changes

Why shouldn't it? Please contact go-e support if you don't like that.

closed 2021-02-28 by andig

andig, 2021-02-28:
Imho it would be confusing if the api showed the physical register values instead of the effective logical value.

premultiply, 2021-02-28:
There is nothing wrong with EVCC writing only the amx value.