evcc-io/evcc #708: Bluelink Zugangsänderung

michaelkircher, 2021-02-28:
**Describe the bug**
Die Antwort von Bluelink führt zu einem HTTP Error 401. Scheinbar wurden die Authentifizierungsmechanismen verändert, es wird jetzt mind. noch eine App ID benötigt.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure vehicle of type hyundai
2. ./evcc -c evcc.yaml vehicle
4. See error

**Expected behavior**
Regular numerical value for SoC

**EVCC details:**
Show output of <code>evcc -v</code>:
```
[main  ] INFO 2021/02/28 09:11:48 evcc 0.47 (e47bb2b)
[main  ] INFO 2021/02/28 09:11:48 using config file evcc.yaml
SoC:           unexpected status: 401
Capacity:      64kWh
State:         unexpected status: 401
Vehicle range: unexpected status: 401
Finish time:   unexpected status: 401

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
loadpoint 1
===========

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
vehicles:
- name: hyundai
  type: hyundai
  title: Kona # display name for UI
  capacity: 64 # kWh
  user: xxx # user
  password: xxx # password
  cache: 60m # cache API response
</pre></details>

<details><summary>Show evcc log output with <code>--log trace</code>:</summary><br><pre>
[main  ] INFO 2021/02/28 09:39:32 evcc 0.47 (e47bb2b)
[main  ] INFO 2021/02/28 09:39:32 using config file evcc.yaml
[hyundai] TRACE 2021/02/28 09:39:32 POST /api/v1/spa/notifications/register HTTP/1.1
Host: prd.eu-ccapi.hyundai.com:8080
Ccsp-Service-Id: 6d477c38-3ca4-4cf3-9557-2a1929a94654
Content-Type: application/json;charset=UTF-8
User-Agent: okhttp/3.10.0

{"pushRegId":"1","pushType":"GCM","uuid":"7ab8f228-79a0-11eb-84b5-784f43898285"}
[hyundai] TRACE 2021/02/28 09:39:32 HTTP/1.1 401 Unauthorized
Content-Length: 138
Content-Type: application/json; charset=utf-8
Date: Sun, 28 Feb 2021 08:39:32 GMT
Etag: W/"8a-asbpEmAt3Fgq76jYBvw+mFph1RM"
Set-Cookie: TS0154565f=01cbaa6f70b668939f167bd34606cfa3977d4981e732a40fb130981f5432c3d55693bdfd6af62688f13b520ce70c8ddc8ada86ddc2; Path=/
X-Ratelimit-Limit: 0
X-Ratelimit-Remaining: 0
X-Ratelimit-Reset: 0

{"retCode":"F","resCode":"4017","resMsg":"Invalid request value - Invalid Application Id.","msgId":"a4894378-403c-435b-a39f-f14c0524bc5a"}
SoC:           unexpected status: 401
Capacity:      64kWh
State:         unexpected status: 401
Vehicle range: unexpected status: 401
Finish time:   unexpected status: 401
</pre></details>

andig, 2021-02-28:
https://github.com/andig/evcc/pull/699

closed 2021-02-28 by andig

andig, 2021-03-10:
See https://github.com/andig/evcc/pull/741