evcc-io/evcc #709: SEMP: PV Modi starten nicht trotz Planung durch SHM

tkempf, 2021-02-28:
EVCC startet bei eingeschaltetem allowcontrol: true in den PV-Modi nicht, trotz Planung durch SHM2

Ladung im Fahrzeug wird wegen zu wenig PV-Energie bei Soc=86% und targetSOC=90% unterbrochen. Trotz eines sichtbaren Planungsrequests im SunnyPortal, wird die Ladungnicht wieder aufgenommen, sondern das System bleibt im Zustand: Sunny Home Manager: Adaptives PV-Laden deaktiviert. Auch manuelles Umschalten der Modi in EVCC andert nichts am fehlenden Start der PV-Modi. Sofort Laden funktioniert, MinPV und PV nicht.
Nach umstellen des Loglevels von Info auf Debug und damit verbundenem Neustart startet dann das Laden auch wieder im MinPV.

**EVCC details:**
evcc version 0.47 (e47bb2b)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
[main  ] INFO 2021/02/28 10:47:45 evcc 0.47 (e47bb2b)
[main  ] INFO 2021/02/28 10:47:45 using config file /etc/evcc.yaml
config
======

grid: grid
----------
Power:          -130W
Current L1..L3: 16.3A 0.468A 14.1A

pv: pv
------
Power: 3680W

loadpoint 1
===========

charger: go-e
-------------
Power:          3170W
Current L1..L3: 14A 0A 0A
Status:         C
Enabled:        true
Charged:        6.3kWh
Charge status:  C

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
#uri: 192.168.178.122:80 # uri for ui
uri: 0.0.0.0:80
interval: 10s # control cycle interval

# log settings
log: debug
#levels:
#  core: info
#  lp-1: info
#  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
#sma: SMA Home Manager 2.0 and SMA Energy Meter. Power reading is configured out of the box but can be customized if necessary.
# To obtain specific energy readings define the desired Obis code (Import Energy: "1:1.8.0", Export Energy: "1:2.8.0").

meters:
- name: grid
  type: sma
#  uri: 192.168.178.42
  serial: 3006947168 # Serial number of the device
- name: pv
  type: default
  power:
    type: calc
    add:
    - type: modbus
      value: Power
      model: sunspec
      uri: 192.168.178.44:502 
      id: 126
    - type: modbus
      value: Power
      model: sunspec
      uri: 192.168.178.45:502
      id: 126

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/andig/evcc-config#chargers
chargers:
- type: go-e
  name: go-e
  uri: http://go-eCharger # either go-e local address

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/andig/evcc-config#vehicles
vehicles:
- name: zoe
  type: renault
  title: Zoe
  capacity: 60 # kWh
  user: moni@hompfs.de # user
  password: netsil4Zoe # password
  region:  de_DE
  vin: VF1AG000166410553
  cache: 5m

# site describes the EVU connection, PV and home battery
site:
  title: Hompfs # display name for UI
  meters:
    grid: grid # grid meter
    pv: pv # pv meter
#    charger: go-e

# Integration with SMA Homemanger 2.0
hems:
  type:  sma
  allowcontrol: true

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
- title: Garage # display name for UI
  charger: go-e # charger
#  meters:
#    charge: go-e # charge meter
  vehicle: zoe
  mode: minpv
#  targetSoC: 80 # charge to 100%
  soc:
    target: 80
    # pOlling defines usage of the vehicle APIs
    # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
    # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
    poll:
      # poll mode defines under which condition the vehicle API is called:
      #   charging: update vehicle ONLY when charging (this is the recommended default)
      #   connected: update vehicle when connected (not only charging), interval defines how often
      #   always: always update vehicle regardless of connection state, interval defines how often
      mode: charging
      # poll interval defines how often the vehicle API may be polled if NOT charging
      interval: 60m
    estimate: false # set true to interpolate between api updates
    levels: # target soc levels for UI
    - 30
    - 80
    - 90
    - 100
  onDisconnect: # set defaults when vehicle disconnects
    mode: minpv # switch back to pv mode
    targetSoC: 80 # charge to 100%
  phases: 1 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 10m # switch charger contactor not more often than this (default 10m)
  mincurrent: 10 # minimum charge current (default 6A)
  maxcurrent: 20 # maximum charge current (default 16A)

# mqtt message broker
#mqtt:
#  broker: localhost:1883
#  topic: evcc # root topic for publishing, set empty to disable
  # user:
  # password:

# influx database
#influx:
#  url: http://localhost:8086
#  database: evcc
  # user:
  # password:

# push messages
#messaging:
#  events:
#    start: # charge start event
#      title: Charge started
 #     msg: Started charging in "${mode}" mode
 #   stop: # charge stop event
 #     title: Charge finished
 #     msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
 #   connect: # vehicle connect event
 #     title: Car connected
 #     msg: "Car connected at ${pvPower:%.1fk}kW PV"
 #   disconnect: # vehicle connected event
 #     title: Car disconnected
 #     msg: Car disconnected after ${connectedDuration}
 # services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
</pre></details>

premultiply, 2021-02-28:
Ein vorhandenes Timeframe heisst ja nicht dass auch zwingend optionale Energie bereitgestellt wird.

tkempf, 2021-02-28:
Was meinst du in dem Fall mit optionaler Energie ?  Vlt. kapier steh ich ja hier echt auf der Leitung, aber ich denke es sollte wie folgt laufen: Der Evcc steht auf Modus Min+PV. Minimalstrom ist 10A Ladung ist 1phasig. Wenn der tatsächliche SOC jetzt unter dem targetSOC ist, sollte der EVCC eine Anforderung an den SHM machen mit der benötigten Energiemenge bis zum TargetSoc.
Wenn der SHM keine anderen Verbraucher einplanen muss, und die Ladung ab 10.00 angezeigt wird, dann sollte doch die Ladung beginnen (vgl. Screenshots), oder verstehe ich da echt was falsch?
[
[image]
](url)
[
[image]
](url)

andig, 2021-02-28:
/cc @premultiply

premultiply, 2021-03-01:
Mmmh schwer zu sagen. Der SHM scheint(!) es zumindest `enabled` eingeplant zu haben von der Grafik.
Ich fürchte da hilft nur ein entsprechendes Logfile...

tkempf, 2021-03-01:
das Logging habe ich ja leider erst danach auf Level debug gestellt und nach dem dsbzgl. Neustart hat er ja dann angefangen zu laden. Von daher habe ich leider hierzu kein log. Ich lasse jetzt aber erst mal core und lp-1 auf debug mitlaufen, reicht das dann, oder braucht Ihr ein trace log?

tkempf, 2021-03-02:
So, hier habe ich die etsprechenden DEBUG Log, nachdem er heute auch nicht gestartet hat. Kann es sein, dass da irgendwoe ein Vorzeichen falsch ist ? 12.40 hat meine Frau die Zoe angeschlossen und versucht auf minpv zu laden. Dann kommt irgendwann die Meldung site power -4680W < enable threshold 0W und er lädt nicht. 12:41:36 hat sie auf sofortladen umgestellt was geklappt hat und 12:43:21 dann noch ein Versuch mit minpv. Dieser wurde wieder gestoppt. Leider fehlen aber die SEMP Messages. 

Mar  2 12:40:55 hompfspi evcc[7747]: [lp-1  ] INFO 2021/03/02 12:40:55 car connected
Mar  2 12:40:57 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:40:57 vehicle soc: 55%
Mar  2 12:40:57 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:40:57 vehicle range: 180km
Mar  2 12:40:57 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:40:57 max charge current: 20A = 0A + 20A (-4603W @ 1p)
Mar  2 12:40:57 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:40:57 pv timer reset
Mar  2 12:40:57 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:40:57 pv max charge current: 20A
Mar  2 12:41:05 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:05 charge power: 0W
Mar  2 12:41:05 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:05 charger status: B
Mar  2 12:41:05 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:05 max charge current: 20A = 0A + 20A (-4608W @ 1p)
Mar  2 12:41:05 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:05 pv timer reset
Mar  2 12:41:05 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:05 pv max charge current: 20A
Mar  2 12:41:15 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:15 charge power: 0W
Mar  2 12:41:15 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:15 charger status: B
Mar  2 12:41:15 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:15 max charge current: 20A = 0A + 20A (-4677W @ 1p)
Mar  2 12:41:15 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:15 pv timer reset
Mar  2 12:41:15 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:15 pv max charge current: 20A
Mar  2 12:41:22 hompfspi evcc[7747]: [lp-1  ] INFO 2021/03/02 12:41:22 set charge mode: minpv
Mar  2 12:41:23 hompfspi evcc[7747]: [lp-1  ] INFO 2021/03/02 12:41:23 set charge mode: pv
Mar  2 12:41:23 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:23 charge power: 0W
Mar  2 12:41:23 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:23 charger status: B
Mar  2 12:41:23 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:23 max charge current: 20A = 0A + 20A (-4680W @ 1p)
Mar  2 12:41:23 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:23 site power -4680W < enable threshold 0W
Mar  2 12:41:23 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:23 start pv enable timer: 1m0s
Mar  2 12:41:23 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:23 pv enable timer remaining: 1m0s
Mar  2 12:41:23 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:23 pv max charge current: 0A
Mar  2 12:41:25 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:25 charge power: 0W
Mar  2 12:41:25 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:25 charger status: B
Mar  2 12:41:25 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:25 max charge current: 20A = 0A + 20A (-4682W @ 1p)
Mar  2 12:41:25 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:25 site power -4682W < enable threshold 0W
Mar  2 12:41:25 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:25 pv enable timer remaining: 58s
Mar  2 12:41:25 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:25 pv max charge current: 0A
Mar  2 12:41:26 hompfspi evcc[7747]: [lp-1  ] INFO 2021/03/02 12:41:26 set target soc: 90
Mar  2 12:41:27 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:27 charge power: 0W
Mar  2 12:41:27 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:27 charger status: B
Mar  2 12:41:27 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:27 max charge current: 20A = 0A + 20A (-4679W @ 1p)
Mar  2 12:41:27 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:27 site power -4679W < enable threshold 0W
Mar  2 12:41:27 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:27 pv enable timer remaining: 56s
Mar  2 12:41:27 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:27 pv max charge current: 0A
Mar  2 12:41:28 hompfspi evcc[7747]: [lp-1  ] INFO 2021/03/02 12:41:28 set charge mode: minpv
Mar  2 12:41:28 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:28 charge power: 0W
Mar  2 12:41:28 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:28 charger status: B
Mar  2 12:41:28 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:28 max charge current: 20A = 0A + 20A (-4683W @ 1p)
Mar  2 12:41:28 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:28 pv timer reset
Mar  2 12:41:28 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:28 pv max charge current: 20A
Mar  2 12:41:35 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:35 charge power: 0W
Mar  2 12:41:35 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:35 charger status: B
Mar  2 12:41:35 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:35 max charge current: 20A = 0A + 20A (-4684W @ 1p)
Mar  2 12:41:35 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:35 pv timer reset
Mar  2 12:41:35 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:41:35 pv max charge current: 20A
Mar  2 12:41:36 hompfspi evcc[7747]: [lp-1  ] INFO 2021/03/02 12:41:36 set charge mode: now
[... cut ...]
Mar  2 12:43:45 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:43:45 detected phases: 1p [3.3 0 0]A
Mar  2 12:43:55 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:43:55 charge power: 4260W
Mar  2 12:43:55 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:43:55 charger status: C
Mar  2 12:43:55 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:43:55 vehicle soc: 55%
Mar  2 12:43:55 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:43:55 vehicle range: 180km
Mar  2 12:43:55 hompfspi evcc[7747]: [lp-1  ] DEBUG 2021/03/02 12:43:55 detected phases: 1p [18.6 0 0]A

premultiply, 2021-03-02:
Ja, die SEMP-Nachrichten sieht man nur auf trace-Level

tkempf, 2021-03-02:
ok, dann lasse ich es mal mit trace level laufen. Ist ja zur Zeit schön sinnig, so dass man gut testen kann...
Woher kommt der "pv max charge current:  0A" Wird der Wert vom SHM übermittelt ? Ich habe aktuell im SunnyPortal den EVCC mit 100% Netzbezug eingestellt, dann müsste der Ladevorgang IMO doch auf jeden Fall starten ...
Mar 2 12:41:27 hompfspi evcc[7747]: [lp-1 ] DEBUG 2021/03/02 12:41:27 site power -4679W < enable threshold 0W
Mar 2 12:41:27 hompfspi evcc[7747]: [lp-1 ] DEBUG 2021/03/02 12:41:27 pv enable timer remaining: 56s
Mar 2 12:41:27 hompfspi evcc[7747]: [lp-1 ] DEBUG 2021/03/02 12:41:27 pv max charge current: 0A
Mar 2 12:41:28 hompfspi evcc[7747]: [lp-1 ] INFO 2021/03/02 12:41:28 set charge mode: minpv
Mar 2 12:41:28 hompfspi evcc[7747]: [lp-1 ] DEBUG 2021/03/02 12:41:28 charge power: 0W
Mar 2 12:41:28 hompfspi evcc[7747]: [lp-1 ] DEBUG 2021/03/02 12:41:28 charger status: B
Mar 2 12:41:28 hompfspi evcc[7747]: [lp-1 ] DEBUG 2021/03/02 12:41:28 max charge current: 20A = 0A + 20A (-4683W @ 1p)
Mar 2 12:41:28 hompfspi evcc[7747]: [lp-1 ] DEBUG 2021/03/02 12:41:28 pv timer reset

premultiply, 2021-03-02:
Der ganze SMA-SEMP-Kram ist leider mangels vollständiger Doku und auch mal Bugs seitens SMA sehr mit Vorsicht zu genießen.
Der SMA EV CHARGER tut ja auch noch nicht immer was er sollte...
Es kann auch ebenfalls gut sein, dass in EVCC diesbezüglich noch Bugs schlummern, da dieses Modul entwickelt wurde wo schon keine Sonne mehr geschienen hat ;)

tkempf, 2021-03-03:
So, ich habe gestern Abend - nachdem das Auto wieder da war - bei SOC 59%  auf MinPV targetSoc 80% gestartet. Der SHM hat den Start auf heute früh 9.20 geplant, obwohl wie gesagt 100% Netzbezug erlaubt sein sollten  - sprich die Ladung eigentlich gleich hätte beginnen können. Der PlanningRequest erlaubt hier als LatestEnd aber auch 24h (s.u.) und so ist halt dann vom SHM geplant.  EVCC weiss ja nichts von den erlaubten 100% Netzbezug im Sunnyportal, daher kann auch der MinPV Modus gar nicht wissen bis wann geladen sein sollte, ausser Zielladen wäre aktiv, dann sollte hier IMO die Endzeit des Zielladens angegeben sein. Insgesamt bin ich mir deshalb nicht wirklich sicher, ob ich nicht doch wieder auf allowControl false umschalten soll. Aber das führt glaube ich hier zu weit, Jetzt schaue ich erst mal, ob der SHM dann wenigstens um 9.20 den Einschaltbefehl gibt. Wenn nicht, funktioniert auf jeden Fall irgendwas nicht wie es soll...

<PlanningRequest>
<Timeframe>
<DeviceId>F-28081973-dca632872980-00</DeviceId>
<EarliestStart>0</EarliestStart>
<LatestEnd>86400</LatestEnd>
<MinEnergy>14000</MinEnergy>
<MaxEnergy>14000</MaxEnergy>
</Timeframe>
</PlanningRequest>

premultiply, 2021-03-03:
Diese Tests sind zumindest sehr hilfreich nachzuvollziehen was passiert (oder auch nicht). Danke!

Zunächst hat sich die Implementierung derzeit erstmal an das Verhalten der Amtron Xtra angelehnt. Muss nicht gut sein...

tkempf, 2021-03-03:
He, nichts zu danken. Wenn sich hier einer zu bedanken hat, dann bin ich das ! ;-)
Jedenfalls hat der SHM jetzt nochmal verschoben auf 10.56. Um das vielleicht doch früher an den Start zu bekommen, habe ich auf targetSOC 100 umgeschaltet. Dabei ist mir aufgefallen, dass im PlanningRequest immer noch die selbe Energiemenge angefragt wird. Müsste das jetzt nicht mehr sein? Immerhin will ich ja jetzt 20% mehr Energie...
`<PlanningRequest>
<Timeframe>
<DeviceId>F-28081973-dca632872980-00</DeviceId>
<EarliestStart>0</EarliestStart>
<LatestEnd>86400</LatestEnd>
<MinEnergy>14000</MinEnergy>
<MaxEnergy>14000</MaxEnergy>
</Timeframe>
</PlanningRequest>
`

tkempf, 2021-03-03:
So jetzt hat er nochmal umgeplant und um 10:43 dann angefangen mit dem konfigurierten Minimalstrom zu laden... Soll ich dir das Trace Log trotzdem mal zukommen lassen ?

tkempf, 2021-03-08:
So, heut wars wieder so weit. Aktuell sieht es mir sehr danach aus, als ob der SEMP PlanningRequest nach einer Änderung des targetSOC nicht (neu) generiert wird. Das hatte neulich schon so ausgesehen, als ob die angeforderte Energiemenge trotz Änderung des targetSOC nicht aktualisiert wird,sondern aus dem Cache kommt und heute wurde soweit ich sehe nach Ablauf der ersten erfolgreichen Ladung bis 80% und anschliessendem Umaschalten auf 100% kein gültiger PlanningRequest mehr erzeugt.
Um 14.52 sollte die Zoe nach vorherigem Laden auf targetSOC 80 nochmal weitergeladen werden bis auf 100% 
um 14.52.03 wurde der targetSOC auf 100 gesetzt. Die folgenden SEMP Messages enthalten aber nur einen leeren PlanningRequest. Wird der Cache bei einer Änderung des targetSOC hier nicht aktualisiert ?

Mar  8 14:52:02 hompfspi evcc[13067]: [lp-1  ] INFO 2021/03/08 14:52:02 set charge mode: pv
Mar  8 14:52:02 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:02 charge power: 0W
Mar  8 14:52:03 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:03 charger status: B
Mar  8 14:52:03 hompfspi evcc[13067]: [lp-1  ] TRACE 2021/03/08 14:52:03 charge currents: [0 0 0]A
Mar  8 14:52:03 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:03 targetSoC reached: 80.0 > 80
Mar  8 14:52:03 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:03 charge power: 0W
Mar  8 14:52:03 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:03 charger status: B
Mar  8 14:52:03 hompfspi evcc[13067]: [lp-1  ] INFO 2021/03/08 14:52:03 set target soc: 100
Mar  8 14:52:03 hompfspi evcc[13067]: [lp-1  ] TRACE 2021/03/08 14:52:03 charge currents: [0 0 0]A
Mar  8 14:52:03 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:03 max charge current: 15A = 0A + 15A (-3450W @ 1p)
Mar  8 14:52:03 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:03 site power -3450W < enable threshold 0W
Mar  8 14:52:03 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:03 start pv enable timer: 1m0s
Mar  8 14:52:03 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:03 pv enable timer remaining: 1m0s
Mar  8 14:52:03 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:03 pv max charge current: 0A
Mar  8 14:52:04 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:04 charge power: 0W
Mar  8 14:52:04 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:04 charger status: B
Mar  8 14:52:04 hompfspi evcc[13067]: [lp-1  ] TRACE 2021/03/08 14:52:04 charge currents: [0 0 0]A
Mar  8 14:52:04 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:04 max charge current: 15A = 0A + 15A (-3453W @ 1p)
Mar  8 14:52:04 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:04 site power -3453W < enable threshold 0W
Mar  8 14:52:04 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:04 pv enable timer remaining: 59s
Mar  8 14:52:04 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:04 pv max charge current: 0A
Mar  8 14:52:13 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:13 charge power: 0W
Mar  8 14:52:13 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:13 charger status: B
Mar  8 14:52:13 hompfspi evcc[13067]: [lp-1  ] TRACE 2021/03/08 14:52:13 charge currents: [0 0 0]A
Mar  8 14:52:13 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:13 max charge current: 15A = 0A + 15A (-3430W @ 1p)
Mar  8 14:52:13 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:13 site power -3430W < enable threshold 0W
Mar  8 14:52:13 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:13 pv enable timer remaining: 50s
Mar  8 14:52:13 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:13 pv max charge current: 0A
Mar  8 14:52:20 hompfspi evcc[13067]: [lp-1  ] INFO 2021/03/08 14:52:20 set charge mode: minpv
Mar  8 14:52:20 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:20 charge power: 0W
Mar  8 14:52:21 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:21 charger status: B
Mar  8 14:52:21 hompfspi evcc[13067]: [lp-1  ] TRACE 2021/03/08 14:52:21 charge currents: [0 0 0]A
Mar  8 14:52:21 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:21 max charge current: 15A = 0A + 15A (-3426W @ 1p)
Mar  8 14:52:21 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:21 pv timer reset
Mar  8 14:52:21 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:21 pv max charge current: 15A
Mar  8 14:52:23 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:23 charge power: 0W
Mar  8 14:52:23 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:23 charger status: B
Mar  8 14:52:23 hompfspi evcc[13067]: [lp-1  ] TRACE 2021/03/08 14:52:23 charge currents: [0 0 0]A
Mar  8 14:52:23 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:23 max charge current: 15A = 0A + 15A (-3404W @ 1p)
Mar  8 14:52:23 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:23 pv timer reset
Mar  8 14:52:23 hompfspi evcc[13067]: [lp-1  ] DEBUG 2021/03/08 14:52:23 pv max charge current: 15A
Mar  8 14:52:31 hompfspi evcc[13067]: [semp  ] TRACE 2021/03/08 14:52:31 send: {XMLName:{Space: Local:} Xmlns:urn:schemas-upnp-org:device-1-0 SpecVersion:{Major:1 Minor:0} Device:{DeviceType:urn:schemas-simple-energy-management-protocol:device:Gateway:1 FriendlyName:evcc Manufacturer:github.com/andig/evcc ModelName:EVCC SEMP Server 0.48 UDN:uuid:fa3f0375-7bfb-11eb-a153-dca632872980 PresentationURL:http://192.168.178.4:80 ServiceDefinition:{Xmlns:urn:schemas-simple-energy-management-protocol:service-1-0 Server:http://192.168.178.4:80 BasePath:/semp Transport:HTTP/Pull ExchangeFormat:XML WsVersion:1.1.0 [... cut ...]
Mar  8 14:54:33 hompfspi evcc[13067]: [semp  ] TRACE 2021/03/08 14:54:33 send: {XMLName:{Space: Local:} Xmlns:urn:schemas-upnp-org:device-1-0 SpecVersion:{Major:1 Minor:0} Device:{DeviceType:urn:schemas-simple-energy-management-protocol:device:Gateway:1 FriendlyName:evcc Manufacturer:github.com/andig/evcc ModelName:EVCC SEMP Server 0.48 UDN:uuid:fa3f0375-7bfb-11eb-a153-dca632872980 PresentationURL:http://192.168.178.4:80 ServiceDefinition:{Xmlns:urn:schemas-simple-energy-management-protocol:service-1-0 Server:http://192.168.178.4:80 BasePath:/semp Transport:HTTP/Pull ExchangeFormat:XML WsVersion:1.1.0} ServiceList:[]}}
`

andig, 2021-03-09:
> So, heut wars wieder so weit. Aktuell sieht es mir sehr danach aus, als ob der SEMP PlanningRequest nach einer Änderung des targetSOC nicht (neu) generiert wird. Das hatte neulich schon so [...]

Also. 

Der PlanningRequest wird immer erzeugt wenn der SHM ihn abfragt. Wird der `targetSoC` geändert, so ändert sich `chargeRemainingEnergy`. Das passiert aber _erst_ bei der nächsten Abfrage des `soc` vom Fahrzeug. Die kann je nach Einstellungen sehr spät kommen.

Jetzt gibts 2 Möglichkeiten:
- `chargeRemainingEnergy` und ggf. auch `chargeEstimate` aktualisieren wenn `targetSoC` geändert wird
- Fahrzeug bei geänderten `targetSoC` mal auslesen und die Logik dann der Verarbeitung dort überlassen

/cc @premultiply

andig, 2021-03-09:
PS.: Bitte Postings formatieren damit sie lesbar sind. Endlose Logfiles gerne auf Pastebin verlinken.

premultiply, 2021-03-09:
Variante 1 klingt für mich zweckdienlicher.

andig, 2021-03-09:
> Mir sieht das nach knapp 2 minuten aus, vom setzen des targetSoc auf 100% bis zum SEMP-Request.

Das ist Zufall da es vom Zeitpunkt der letzten Abfrage abhängt.

> Der Planning Request in der hier zitierten Message ist doch leer, oder deute ich das "PlanningRequest:[]" falsch ?

Korrekt. Ohne jetzt geschaut zu haben, aber wenn eins von Beiden leer ist, dann gibts keinen Request:

	if connected && maxEnergy > 0 {
		res = PlanningRequest{
			Timeframe: []Timeframe{{
				DeviceID:            s.deviceID(id),
				EarliestStart:       0,
				LatestEnd:           latestEnd,
				MinEnergy:           &minEnergy,
				MaxEnergy:           &maxEnergy,
				MaxPowerConsumption: &maxPowerConsumption,
				MinPowerConsumption: &minPowerConsumption,
			}},
		}
	}

tkempf, 2021-03-09:
> Das ist Zufall da es vom Zeitpunkt der letzten Abfrage abhängt.

Müsste nicht der SHM ca. minütlich den InformationRequest an das Gateway schicken ??

> Korrekt. Ohne jetzt geschaut zu haben, aber wenn eins von Beiden leer ist, dann gibts keinen Request:
> if connected && maxEnergy > 0 {

Hmm, das Auto war angeschlossen und durch das umschalten von 80% auf 100% targetSoc sollte maxEnergy 
eben auch (sofort beim umschalten) > 0 werden. Und das passiert scheinbar nicht.

premultiply, 2021-03-09:
Ich bin mir gerade da nicht so sicher ob das Timeframe bei `PlanningRequest:[]` im Log wirklich leer ist.
Dann wäre nämlich normalerweise auch `EMSignalsAccepted:false`
Sicherheitshalber mal mit der `/semp`-URL bei EVCC gegenchecken was da wirklich drin steht.
Nicht dass es am Ende nur nicht im Log angezeigt wird...

premultiply, 2021-03-09:
Was mir auf jedenfall die letzten Tage aufgefallen ist, dass der SHM im Modus PV (d.h. nach SEMP übersetzt MinEnergy:0) nur sehr sehr sparsam die Ladung überhaupt frei gibt. Ich mutmaße dabei, dass das eventuell mit der Prognose zu tun haben könnte.
Mit MinPV hingegen ist man sehr nah am nativen PV-Verhalten von EVCC dran.

andig, 2021-03-11:
Ich hab den Faden verloren. Wer muss hier was prüfen? Issue schliessen?

premultiply, 2021-03-11:
Seit dieses Issue geöffnet wurde gab es ein paar Änderungen an der SEMP-Schnittstelle die das Verhalten möglicherweise verbessern bzw. eventuell das Problem beheben.
@tkempf könntest du nochmal mit der aktuellen Version testen?

tkempf, 2021-03-11:
Klar mach ich gerne, allerdings habe ich keine eingerichtete GO Entwicklungsumgebung und kann das also erst testen, wenn die Version zum Download steht...
Oder bekomme ich die go Umgebung zum kompilieren von EVCC relativ schmerzfrei auf Windows eingerichtet ?

> Der PlanningRequest wird immer erzeugt wenn der SHM ihn abfragt. Wird der `targetSoC` geändert, so ändert sich `chargeRemainingEnergy`. Das passiert aber _erst_ bei der nächsten Abfrage des `soc` vom Fahrzeug. Die kann je nach Einstellungen sehr spät kommen, bei Dir oben sind das nur 3s.
> 
> Jetzt gibts 2 Möglichkeiten:
> 
>     * `chargeRemainingEnergy` und ggf. auch `chargeEstimate` aktualisieren wenn `targetSoC` geändert wird
> 
>     * Fahrzeug bei geänderten `targetSoC` mal auslesen und die Logik dann der Verarbeitung dort überlassen

Unabhängig davon, ob der SHM den Einschaltbefehl gibt, oder nicht, bin ich aber der Meinung, der aktualisiert Wert von chargeRemainingEnergy sofort beim Ändern des targetSoC im nächsten SEMP PlanningRequest erscheinen sollte...

premultiply, 2021-03-12:
@andig Ist das wirklich nur ein Thema des Caches hier https://github.com/andig/evcc/blob/05b5032e0b821c11588874203a760e3f0debdb7a/hems/semp/semp.go#L422 ?

Kann man den Cache für diesen Fall nicht einfach umgehen? Würde das etwas bringen?

andig, 2021-03-12:
Der Cache hält alle aktuellen Werte. Er wird nir nicht geschrieben 😳

premultiply, 2021-03-12:
Dann also Variante 1. Nur wie? 🤷🏻‍♂️
Da ist der Meister gefragt...

tkempf, 2021-03-13:
So, ich habe mir jetzt mal den aktuellen master compiliert und mit dem getestet. Lassen wir mal den SHM (fast) ganz ausser acht.
1. Das Auto ist auf 80% geladen und angeschlossen.
2. Nach Umschalten auf Min+PV und Anforderung auf 90% hat die Ladung begonnen. So wie es soll
3. Nach Zurückschalten auf 80% wird die Ladun beendet, wo wie es soll.
4. Nach erneutem Umschalten auf 90% wird die Ladung nicht wieder vom SHM freigegeben, weil EVCC keinen PlanningRequest erzeugt (zumindest nicht gleich).

andig, 2021-04-01:
Also. Der Lösungsvorschlag war ja:

- chargeRemainingEnergy und ggf. auch chargeEstimate aktualisieren wenn targetSoC geändert wird

Das Problem besteht also *nur* bei Änderung des targetSoC?

tkempf, 2021-04-01:
Ich denke schon. Das ist zumindest der gängigste usecase. Evtl. könnte man noch testen was passiert, wenn man bei gleichbleibendem targetSOC zwischen den Modi umschaltet. Allerdings bin ich aktuell auf allowControl=false gegangen, da das einfach reibungsloser für uns läuft.

closed 2021-04-30 by stale[bot]