evcc-io/evcc #715: PV enable timer starts only after car is connected

brunohaller, 2021-03-01:
**Describe the bug**

When running evcc in "pv" mode and plugging the cable in, the timer starts:
`[lp-1  ] DEBUG 2021/02/28 12:13:21 start pv enable timer: 30s`
even though the exported power was above the `enable.threshold` already long ago , before plugging in.

It seems like the timers are reset or only taken into account, as soon as the car is connected. 

P.S. With the default of 1 minute, I had multiple times the problem, that the charging did not start at all or went into an error state, as the car seemed to wait for a signal from the charging station, but it didn't happen. Currently I reduced my delay to 30 secs to workaround.

**Expected behavior**
Start charging immediately, if the `enable` settings have been fulfilled already before connecting the car.
Perform the "PV enable times" logic also when car is not connected.

andig, 2021-03-02:
> It seems like the timers are reset or only taken into account, as soon as the car is connected.

i haven‘t checked, but is that a problem?

brunohaller, 2021-03-02:
> i haven‘t checked, but is that a problem?

For me it was. Of course you can argue how "big" the issue is, but generally I would prefer to have a certain enable delay to prevent the charging to start on a short sunny moment. On the other hand, it should start charging on a sunny day, as soon as I connect the cable.

MrJayC, 2021-04-02:
I would support to do monitoring of the available energy independently from a connected vehicle, so that charging can start immediately after plugging in. For me, this has low priority, but would improve the user experience as you would get "expected behaviour".
If a user, who is not familar with how EVCC works internally, plugs in the vehicle on a sunny day, she/he would expect that charging starts right now. Took me a while to understand where the delay comes from.

andig, 2021-04-02:
> I would support to do monitoring of the available energy independently from a connected vehicle,

Off-topic. Please don't spam.

andig, 2021-04-23:
/reopen

andig, 2021-05-13:
Keep

andig, 2021-05-14:
Vorschlag: wir behandeln vehicle connected wie auch eine manuelle Auswahl des Modus über das UI oder Api so, als wären alle Timer abgelaufen und Schaltvorgänge könnten sofort stattfinden. Dadurch wird die Reaktivität für den Anwender erhöht. 

Passt?

andig, 2021-05-15:
Test wäre gut, sonst schieb ichs durch.

MrJayC, 2021-05-15:
- [x] Sofortiger Ladebeginn bei genügend PV-Strom: funktioniert

PR #1023 (2021-05-15): Disable pv mode countdown on vehicle connection and mode switch
Fix #715
closed 2021-05-15 by andig