evcc-io/evcc #723: Detect: Kostal Inverter auf Port 1502

premultiply, 2021-03-05:
Für die Erkennung von Kostal-Systemen werde es ggf. hilfreich die `detect`-Funktion bei SunSpec auf den Modbus/TCP-Port 1502 auszuweiten. Dort ist nur das EM auf Port 502 abfragbar. Die Inverter selber laufen auf Port 1502.

andig, 2021-03-05:
Mach mal. Wäre eine weitere Erkennung speziell für Kostal?

andig, 2021-03-06:
SolarEdge hat wohl optional auch 1502. sollten wir also einfach bei der SunSpec-Erkennung inkludieren?

premultiply, 2021-03-06:
Ja, exakt. „Normales“ SunSpec, nur auf Port 1502 statt 502 dort.

andig, 2021-03-08:
Beim aktuellen `detect` wird immer nur die IP Adresse zum nächsten Test mit durchgereicht. Was wir hier machen müssten, wäre die Testlogik so umzubauen, dass die Ergebnisse angereichert werden, z.B. auch um den Port und das angereicherte Ergebnis von Test zu Test gereicht. Dann könnte der Folgetest (=modbus) auf den vorigen (=TCP connect) auf dem richtigen Port zurück greifen.

closed 2021-04-04 by stale[bot]

premultiply, 2021-05-02:
Ich glaub hier sollte etwas Zeit investiert werden.

RandomRam2023, 2023-04-08:
Hallo,
dieses Topic passt ein stückweit zu meinem aktuellen Problem:
der Solaredge-Inverter hat ModBus 502 voreingestellt. darüber kann dann auch die Varta-Batterie inkl. Grid-meter die PV-Leistung in der Varta-App anzeigen. 
evcc kann aber mit dem Port 502 nix anfangen, auch wenn ich die .yaml mit 502 versorge.
Anpassung des WS-Konfiguration auf 1502 löst dieses Problem für evcc 👍 

Dummerweise ist nun aber die Varta-Anzeige blind, und zurückstellen hilft auch nix mehr, varta-App ist zickig.
Ist der Satz in der evcc-Übesicht "nur ein System darf auf den WR zugreifen" so gemeint, dass es da keine Parallelwelten geben darf? (hüstel)

ohne es besser zu wissen sind 1502 / 502 ja nicht nur Nummern, sondern auch wohl Funktionsmodi, also passive Listener, slave, Relay,... Ich berüchte, meine Idealvorstellung kollidiert daher mit der Realität :)  wäre aber trotzdem über Erleuchtung dankbar.

Die Alternative, dass nur noch evcc meine PV-Leistungen sieht (und ich keine Cloud-Verbindung auf meine Anlagen hab), ist zwar uncool, aber auch kein Beinbruch. PV-Überschussladen ist höher berechtigt.

andig, 2023-04-08:
Neues Thema, das hier ist seit 2 Jahren erledigt.