evcc-io/evcc #729: SEMP: RecommendedPowerConsumption als Float

premultiply, 2021-03-07:
Wie ich gerade durch einen Mitschnitt der Kommunikation zwischen SHM und EVCC feststellen musste sendet der SHM gelegentlich den Wert RecommendedPowerConsumption als Float, was EVCC dann mit einem HTTP/1.1 400 Bad Request beantwortet.

Oft sehen diese Requests so aus:
```
T 192.168.178.90:58516 -> 192.168.178.41:7070 [AP] #280
  POST /semp/ HTTP/1.1..Host: 192.168.178.41:7070..Accept:
   */*..Content-Type: application/xml..Content-Length: 312
  ....<?xml version="1.0" encoding="UTF-8"?>.<EM2Device xm
  lns="http://www.sma.de/communication/schema/SEMP/v1">..<
  DeviceControl>...<DeviceId>F-28081973-xxxxxxxxxxxx-00</D
  eviceId>...<On>true</On>...<RecommendedPowerConsumption>
  4439</RecommendedPowerConsumption>...<Timestamp>0</Times
  tamp>..</DeviceControl>.</EM2Device>

T 192.168.178.41:7070 -> 192.168.178.90:58516 [AP] #282
  HTTP/1.1 200 OK..Date: Sun, 07 Mar 2021 10:39:10 GMT..Co
  ntent-Length: 0....
```

manchmal aber eben auch so:
```
T 192.168.178.90:58412 -> 192.168.178.41:7070 [AP] #37
  POST /semp/ HTTP/1.1..Host: 192.168.178.41:7070..Accept:
   */*..Content-Type: application/xml..Content-Length: 323
  ....<?xml version="1.0" encoding="UTF-8"?>.<EM2Device xm
  lns="http://www.sma.de/communication/schema/SEMP/v1">..<
  DeviceControl>...<DeviceId>F-28081973-xxxxxxxxxxxx-00</D
  eviceId>...<On>true</On>...<RecommendedPowerConsumption>
  4245.7998046875</RecommendedPowerConsumption>...<Timesta
  mp>0</Timestamp>..</DeviceControl>.</EM2Device>

T 192.168.178.41:7070 -> 192.168.178.90:58412 [AP] #39
  HTTP/1.1 400 Bad Request..Date: Sun, 07 Mar 2021 10:35:1
  0 GMT..Content-Length: 0....
```

**Expected behavior**
`RecommendedPowerConsumption` wird auch als Float akzeptiert/verarbeitet.

premultiply, 2021-03-07:
https://github.com/andig/evcc/blob/fb22f47bc11ed2dca3bd555003ec37fda6fb22f8/hems/semp/messages.go#L150

andig, 2021-03-07:
Gerne, widerspricht aber der SMA-eigenen Doku. Machst Du PR?

PR #730 (2021-03-07): SEMP: RecommendedPowerConsumption as float
Fixes #729
closed 2021-03-07 by andig