evcc-io/evcc #731: Vehicle::Mercedes - someone interested in co-creation?

muelmark, 2021-03-07:
Hello togehter,
I am searching for a co-creator and some hints for preparing a first mercedes vehicle connection.
Is there any interfaces used for creating new vehicles?
Bests,
Markus

andig, 2021-03-07:
Please join us on slack, link it in the Wiki

DetMoerk, 2021-03-09:
There is a Mercedes SoC Modul in openWB (Python), you can check the openWB Wiki, there is a description how to set up a Application in the Mercedes Developer Portal https://developer.mercedes-benz.com/

Regards
Det

DetMoerk, 2021-03-11:
It is not necessary to implement the first authentication via the Evcc Frontend. The api provides a Callback URL which has to be implemented in EVCC. You can jump out from the frontend to a Mercedes Me site and then the callbackURL will be called by the Merceses API and provide a auth Token and a Refresh token and an expire Time. Inside the function behind the Callback URL you have to store these three tokens/value and use them afterwards to access the API. Is the expire time reached, you have to ask for new tokens with the refresh token. 
Regards
Det

andig, 2021-03-11:
> The api provides a Callback URL which has to be implemented in EVCC.

evcc can provide a local callback url (localhost) but not an externally available one (no central service). For other plugins we're parsing/handling the redirect responses and extract the necessary data. The Mercedes API- on first glance- defied this as the returned response basically seemed to be PingID js monster.

> You can jump out from the frontend to a Mercedes Me site and then the callbackURL will be called by the Merceses API and provide a auth Token and a Refresh token and an expire Time

The tesla plugin provides a separate auth command to generate/store the tokens. How long-lived are the tokens?

@Det: if you want to support the implementation please join us on Slack. For me personally- not driving Mercedes- it's not a priority.

PS.: I've provided feedback to the API questionnaire but of course never received a response...

DetMoerk, 2021-03-11:
Gor me it s also not a need, as I have a openWB Standard+ as Charger and the module I provided there works without a problem. 

Token lifetime is 2 h at the moment. But as it is delivered in the Callback, that may change in the future.

Regards
Det

andig, 2021-03-11:
> Token lifetime is 2 h at the moment. But as it is delivered in the Callback, that may change in the future.

2hrs until token refresh or until new login? The latter is imho unacceptable and hence wontfix. 

If anybody from Mercedes wants to get in touch for a solution please contact me at cpuidle@gmx.de

DetMoerk, 2021-03-11:
I wouldn't say that it is a working solution, if I would have to login every two hours ;-). I never had to reauthenticate since December 2021. So the token referesh after the token timeout works. You only have to make sure that you request a new auth token when the old one is expired.

andig, 2021-03-12:
Klingt gut. Ich würde mir das anschauen, bräuchte dafür aber Zugangsdaten zu Testzwecken. 

@muelmark oder bist Du da noch dran?

andig, 2021-03-12:
Super. Sind tatsächlich nur ein paar Zeilen Code in go. Mit Zugangsdaten könnte ich etwas testen :)

DetMoerk, 2021-03-12:
Es gibt auch ein Sandbox Vehicle, das du benutzen kannst für die ersten Tests.

andig, 2021-03-12:
> Es gibt auch ein Sandbox Vehicle, das du benutzen kannst für die ersten Tests.

Ich find's tatsächlich nicht- wo genau?

DetMoerk, 2021-03-12:
https://developer.mercedes-benz.com/products/electric_vehicle_status/

Da gibt es Sandbox und BYOCAR. In der Sandbix findest du das Sandbox Auto

VG
Det

andig, 2021-03-12:
Ich finde kein Sandbox Auto:

[image]
[image]
[image]

Habe wohl meinen blonden Tag...

DetMoerk, 2021-03-12:
Auf deinen Bild ist dich die Sandbox zu sehen.

andig, 2021-03-12:
Ja. Und wenn ich da klicke bekomme ich auch keine VIN für Testzwecke? Glaub wir reden aneinander vorbei :O

DetMoerk, 2021-03-12:
Den Authorization Workflow kannst du nicht testen, das Fahrzeug schon. 
Aber ja, du hast Recht, da fehlt ein entscheidendes Stück des Aufwands.

DetMoerk, 2021-03-12:
Für die Authentifizierungsanfrage brauchst du einen Aufruf wie folgt
https://id.mercedes-benz.com/as/authorization.oauth2?response_type=code&state=lp1&client_id=**<ClientID>**&redirect_uri=http://EVCC**/irgendeinpfad/funktionsname**&scope=mb:vehicle:mbdata:evstatus%20offline_access

Die ClientID bekommst du aus der im Developer Portal angelegten Client ID. Die redirect_uri musst du für EVCC festlegen. Die dort hinterlegte Funktion muss 3 Daten speichern können: AccessToken, RefreshToken und die ExpireTime. Diese Funktion wird nach der Authentifizierung mit Mercedes Me Credentials aufgerufen und liefert die drei Werte in einem JSON zurück. 
Im Parameter state kannst du einen Kenner mitgeben, der 1:1 zurück geliefert wird. Darüber mache ich in openWB die Erkennung für welchen der Ladepunkte die zurückgelieferten Daten sind.

andig, 2021-03-12:
> Für die Authentifizierungsanfrage brauchst du einen Aufruf wie folgt

Ja, OAuth ist klar. Aber: wenn ich mit bearer token von https://id.mercedes-benz.com/as/token.oauth2 auf die tryout vehicle url zugreife gibts nen 401:

```
GET /vehicledata_tryout/v2/vehicles/WDB111111ZZZ22222/containers/electricvehicle HTTP/1.1
Host: api.mercedes-benz.com
Authorization: Bearer eyJhbGciOiJkaXI...

{
  "exveErrorId": "101",
  "exveErrorMsg": "Unauthorized",
  "exveErrorRef": "0c00434b-cc01-4583-8f7b-e740abc66fbf"
}
```

Sollte das nicht gehen? Oder was meintest Du mit:

> Den Authorization Workflow kannst du nicht testen, das Fahrzeug schon.

DetMoerk, 2021-03-12:
Ich bekomme mit dem Testaufruf aus der API Steit eine korrekte Rückmeldung
pi@raspi4:~ $ curl -X GET "https://api.mercedes-benz.com/vehicledata_tryout/v2/vehicles/WDB111111ZZZ22222/containers/electricvehicle" -H  "accept: application/json;charset=utf-8" -H  "authorization: Bearer 2c2c222c-e123-4123-v123-2c2c222c2c22"
[
  {
    "soc": {
      "value": "90",
      "timestamp": 1541080800000
    }
  },
  {
    "rangeelectric": {
      "value": "456",
      "timestamp": 1541080800000
    }
  }

Woher hast du den Bearer Token

andig, 2021-03-12:
Erledigt. Ich hatte den scope fürs API nicht gesetzt. Sorry für die Verwirrung. Die Scopes in der Doku sind sehr gut versteckt...

Nächste Hürde: Nutzung des Refresh Tokens geht nur einmal. Damit braucht ich nach Neustart der Anwendung (evcc persistiert nichts) einen neuen Token den der User erst erzeugen muss. Das wäre wohl tatsächlich ein k.o.-Kriterium?

DetMoerk, 2021-03-12:
Ja wenn der Refresh Token verwendet wurde bekommst du wieder einen kompletten Set Token. Also Access Token, Referesh Token und expire_time.

andig, 2021-03-12:
Erledigt. Ich hatte den scope fürs API nicht gesetzt. Sorry für die Verwirrung. Die Scopes in der Doku sind sehr gut versteckt...

Mit scopes gings genau 1x, danach nicht mehr. 

    ❯ curl -X GET https://api.mercedes-benz.com/vehicledata_tryout/v2/vehicles/WDB111111ZZZ22222/containers/electricvehicle -H "accept: application/json;charset=utf-8" -H "authorization: Bearer eyJhbGciOiJkaXIiLCJlbmMiO

Gibt 401. Das Token kommt aus https://developer.mercedes-benz.com/content-page/oauth-documentation

> Ja wenn der Refresh Token verwendet wurde bekommst du wieder einen kompletten Set Token. Also Access Token, Referesh Token und expire_time.

Das ist leider ein Killer. Bei Tesla z.B. bleibt das Refresh Token gültig...

andig, 2021-03-12:
Woher hast Du dieses kurze Token?

DetMoerk, 2021-03-12:
Das Token aus der oauth Beschreibung ist wohl eher allgemein. In der Sandbox Umgebung findest du ein Token, welches zur entsprechenden API passt. 
-H "authorization: Bearer 2c2c222c-e123-4123-v123-2c2c222c2c22"
Aber wie gesagt, wenn die expire_Time des Auth Token abgelaufen ist, musst du mit dem Refresh Token ein neue Token Set abrufen und vermutlich auch persistieren.

DetMoerk, 2021-03-12:
. 
[image]

andig, 2021-03-12:
>Das Token aus der oauth Beschreibung ist wohl eher allgemein. In der Sandbox Umgebung findest du ein Token, welches zur entsprechenden API passt.

Oh mann. Danke. Wäre schön sowas mal klar hinzuschreiben. Kein Mensch erwartet sowas im kleingedruckten "ausprobieren". Ich jdfls nicht...

Fun fact: einmal hat die API mit "richtigem" Token funktioniert :O

DetMoerk, 2021-03-12:
Wo ist denn der Go Code

> Damit braucht ich nach Neustart der Anwendung (evcc persistiert nichts) einen neuen Token den der User erst erzeugen muss. Das wäre wohl tatsächlich ein k.o.-Kriterium?

Jetzt hab ich es kapiert. Ja ohne Persistenz geht das nicht. Sonst muss man immer nach dem Anmelden über den Login Dialog bei Mercedes Me gehen.

muelmark, 2021-03-12:
Hallo zusammen,

maybe that helps? Der Adapter ioBroker.mercedesme hat die Token Authentifizierung und auch den Token Renewal gelöst.
Soweit ich das überblicke auch mit Persistenz.

Wäre es eine einfach Idee, dass ganze Auth "Geraffel" in den iobroker zu verfrachten?
Mann kann über die 2 Adapter:
- simple rest api Adapter und 
- https://github.com/TA2k/ioBroker.mercedesme/blob/master/README.md

über http://iobroker:8087/getPlainValue/mercedesme.0.(VIN).state.rangeelectric.intValue direkt eine Rest-km Anzahl auslesen.

Dann beschränkt sich das ganze EVCC seitig auf ein paar rest api calls und ein paar einfache Berechnungen.
SoC = Rest km *100 / MaxRange. MaxRange liesse sich auch als Datenpunkt in iobroker einfach anlegen.

Markus

andig, 2021-03-13:
Das kannst du gerne machen. Evcc muss aber auch ohne ioBroker funktionieren.

DetMoerk, 2021-03-13:
Wenn der Rangeelectric angegeben ist, muss eigentlich auch der SoC direkt abfragbar sein. Ich hab beim schnellen drüber schauen nicht gesehen welche API(s) in dem Programm angesprochen werden. Es gibt 3 unterschiedliche. 
EVstatus, Fuelstatus und Vehiclestatus.

andig, 2021-03-13:
Ist er auch, jdfls soweit das ohne Testfzg erkennbar ist

muelmark, 2021-03-13:
Würde die entsprechenden Auth Daten VIN, User+Password temporär bereitstellen, wie am Besten?

andig, 2021-03-13:
Slack oder cpuidle@gmx.de?

andig, 2021-03-14:
Mhhm. Diesmal mit echtem Fahrzeug:

    GET /vehicledata/v2/vehicles/W1.../resources/soc HTTP/1.1
    Host: api.mercedes-benz.com
    Accept: application/json
    Authorization: Bearer eyJhbGciOiJkaX...

    HTTP/2.0 204 No Content
    Cache-Control: no-cache, no-store, max-age=0, must-revalidate
    Date: Sun, 14 Mar 2021 12:05:01 GMT
    Expires: 0
    Pragma: no-cache
    Server: istio-envoy
    Strict-Transport-Security: max-age=15724800; includeSubDomains
    X-Content-Type-Options: nosniff
    X-Envoy-Upstream-Service-Time: 587
    X-Frame-Options: DENY
    X-Xss-Protection: 1; mode=block

@DetMoerk hast Du eine Idee wieso HTTP 204? Warum gibts keine Daten?

andig, 2021-03-14:
Also im Prinzip (tm) funktioniert es, ist so aber nicht einsatzfähig:

- refreshte Tokens müssen gespeichert werden- das ist bei evcc momentan nicht vorgesehen
- das Vehicle müsste dem UI anzeigen können dass es ein Login benötigt das der Anwender dann ausführen kann
- richtig "nice" wäre ein Service unter https://evcc.io der eine globale Redirect URI für den OAuth Code Exchange bereitstellen kann um den Code des OAuth Providers and evcc weiter zu leiten

@DetMoerk und letztlich fehlt noch ein `vehicles` API mit dem evcc selbst rausfinden kann, um welches Auto- falls mehrere vorhanden- es sich handelt.

Ist leider in Summe eine Menge Aufwand die sich zu betreiben momentan wohl kaum lohnt, v.a. weil auch auf Anwenderseite immer noch eine Menge fehleranfällig technischer Schritte übrig bleiben.

Daher- schweren Herzens- erstmal wontfix.

DetMoerk, 2021-03-14:
204 kommt dann, wenn die Drittanbieter Freigabe nicht gesetzt ist. Muss für jedes Fahrzeug einzeln vom Eigentümer in Mercedes Me gemacht werden. 
Ich bin mir nicht sicher, wie lange es dauert bis das übernommen wird. Und danach müsste das Fahrzeug auch einmal gestartet werden. 

Das Verhalten kenne ich, konnte es aber nach der Aktivierung nicht mehr beobachten, aber mein Fahrzeug wird dann auch regelmäßig bewegt.

andig, 2021-03-15:
Hat nach geklappt, anscheinend hat die Aktivierung noch auf das Fahrzeug gewartet.

DetMoerk, 2021-03-15:
Ja das hab ich bei den ooenWB Usern auch immer wieder gesehen.  Meistens aber nur beim ersten Mal aktivieren, auch wenn in der Doku steht, dass wenn das Fahrzeug über längere Zeit steht, dann auch ein 204 Return kommt. Aber da hab ich bisher nichts gehört und auch selbst  nie gehabt.

Sputnik24, 2021-04-10:
Gibt es schon was Neues? Ich sehe, dass der PR noch nicht durch ist. Ich bekomme im Mai meinen A250e und Mai/Juni eine PV-Anlage mit go-e charger und würde gern EVCC nutzen.

andig, 2021-04-10:
@DetMoerk wie hast Du das bei OpenWB mit der Redirect URI gelöst? Die müsste ja jeder Anwender individuell für seine Installation erzeugen, richtig?

DetMoerk, 2021-04-10:
Die Redirect URL ist mehr oder weniger statisch. Da die Pfade in der openWB ja auch fix sind. Von daher kann das eigentlich für jeden identisch laufen wenn man den hostname und nicht die IP verwendet.

DetMoerk, 2021-04-10:
@Sputnik24 : in der openWB gibt es bereits ein funktionierendes Mercedes EQ Modul

Sputnik24, 2021-04-10:
@DetMoerk Mir ist openWB aber überdimensioniert und es läuft nicht in Docker. Ich würde gern EVCC in Docker verwenden.

andig, 2021-04-12:
> Die Redirect URL ist mehr oder weniger statisch.

@DetMoerk das ist klar- aber wie hast Du das auf der Seite von Mercedes gelöst? Du kannst ja nicht wissen unter welcher URL die OpenWB erreichbar ist und daher ein Redirect aussehen müsste damit er wieder auf der OpenWB landet? Oder hab ich da einen Denkfehler?

DetMoerk, 2021-04-12:
Naja, die URL wäre entweder die Lokale IP Adresse und der entsprechende Pfad zum Script, oder der Hostname und der Pfad zum Script. 

Pfad und Script sind statische Informationen. Hostname bzw. IP muss der Anwender selbst kennen und daraus die redirect URL zusammen setzen und im Mercedes Developer Portal entsprechend in seiner Application hinterlegen.

andig, 2021-04-16:
Ich habe mir überlegt, dass wir die Redirect URL ggf. auch über die EVCC Cloud schleusen könnten. Dann ließe sich das API für alle Anwender ohne lokale Konfiguration nutzen. 

Mir fehlt allerdings weiter ein Testfahrzeug um das ausprobieren zu können.

onlineoffline, 2021-04-27:
Hallo, ich würde gerne EVCC mit SMA-WR und go-e charger mit A250e und B250e verwenden. Wäre auch bereit zu testen!

muelmark, 2021-06-05:
> Ich habe mir überlegt, dass wir die Redirect URL ggf. auch über die EVCC Cloud schleusen könnten. Dann ließe sich das API für alle Anwender ohne lokale Konfiguration nutzen.
> 
> Mir fehlt allerdings weiter ein Testfahrzeug um das ausprobieren zu können.
Mir fehlt noch das Solaredge Monitoring, dann werde ich wieder aktiv. Fahrzeug ist nach wie vor vorhanden, kann ich wie das letzte Mal weiterhelfen?

Sputnik24, 2021-06-05:
Ich mach das ganze mittlerweile über iobroker, wo ich den mercedesme Adapter laufen hab. Die Daten stelle ich evcc über mqtt zur Verfügung.

Vielleicht könnt ihr mal beim iobroker Adapter schauen, wie der Kollege das gelöst hat: https://github.com/TA2k/ioBroker.mercedesme

muelmark, 2021-06-05:
^^ backup plan, der Adapter läuft bei mir ebenfalls. MQTT, nicht simple-(rest)-API?

Sputnik24, 2021-06-05:
MQTT, ja. Die Daten der PV-Anlage lass ich auch per MQTT holen.

JochenFriedrich, 2021-09-27:
Alternativ geht auch FHEM und MQTT.

https://forum.fhem.de/index.php/topic,113521.0.html

Ich habe nur die Abfragefrequenz von 5/h auf 30/h erhöht. Laut Doku muss man nur 120s zwischen 2 Aufrufen warten.

andig, 2021-09-27:
Ist nicht geplant. Solange Mercedes es so kompliziert macht dass jeder Anwender sich mit REDIRECT_URIs auseinander setzen muss lasse ich das. Es gibt hier schon genug Supportanfragen.

JochenFriedrich, 2021-09-27:
Kein Bedarf, noch iobroker zu installieren, die Lösung über FHEM funktioniert ja auch 😃

andig, 2021-11-08:
Wir haben jetzt einen Weg gefunden, das Token upfront zu generieren analog der Lösung für Tesla. Damit könnte der Rest in die Umsetzung gehen sobald ich ein Testexemplar hätte.

Macrisu, 2021-12-08:
New EQC owner here, so any chance for MB support?

andig, 2021-12-08:
Not really. The MB api requires complicated setup on the user side- nothing I want to support, see https://github.com/evcc-io/evcc/issues/731#issuecomment-950173651.

Jeepersfinest, 2021-12-11:
Hallo,
in meiner Heimautomatisierung FHEM nutze ich auch die Daten aus den Mercedes-APIs.
Siehe hier: https://forum.fhem.de/index.php/topic,113521.0.html
Der Entwickler des Moduls JustMe1968 ist sehr aktiv in der Programmierung und super hilfsbereit.
Such hier auf GH: https://github.com/justme-1968
Vielleicht könntet ihr ihn einmal um Unterstützung bitten?

Ich drücke die Daumen!

VG
Christian

muelmark, 2022-01-10:
Testexemplar wäre vorhanden, gern!

muelmark, 2022-01-10:
@Macrisu: if that was a question then yes, I know a vehicle to test anything.

[6 more replies]

PR #2259 (2022-01-21): Add Mercedes
Fix #731

First draft to handle authorization code flow for the mercedes API. 

[image]

Login button disappears after ~5 seconds after login, could/should be improved.
Vehicle SoC is fetched after Login and switching the mode to "now".

[image]

Rudimentary logout is possible, MB does not provide an SSO logout. Therefore only the backend can forget the current token and the user has to clear the browser cache.

TODO
- [x]  merge master into branch!
- [ ] long-term test to ensure refresh is also working properly
- [x] make the callback dynamic -> remove 'mercedes' - new PR?
- [x]  [... cut]
closed 2022-01-21 by andig