evcc-io/evcc #748: Discovergy Zähler - Bad Request wegen meterId

KarlJep, 2021-03-11:
**Describe the bug**
Hallo! 
zuerst einmal, tolles Projekt! 
Ich habe auch gleich angefangen, meine erste Konfiguration zu erstellen und bin gleich beim Discovergy Zähler auf Probleme gestoßen. [https://github.com/andig/evcc-config#meter-discovergy](evcc-config#meter-discovergy) lässt mich drauf kommen, dass die MeterID eine14stellige Nummer ist. Eine ähnliche sehe ich auch im Discovergy Portal. Trage ich die ein und starte evcc, erhalte ich aber nur Fehlermeldungen "400 Bad Request". Nach etwas probieren, bin ich draufgekommen, dass man an der Stelle eine 32-stellige Zahl eintragen muss. Diese Zahl gibt es nur über die RESTAPI "meters". 
Mach ich hier was falsch ? 

Vielen Dank! 
**To Reproduce**
1) Benutze default-config für discovergy Zähler (demo)
2) evcc meter

**Expected behavior**
Zählerwerte werden angezeigt

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.48 (d4f8d03)

```

- type: discovergy
  user: demo@discovergy.com 
  password: demo # password 
  meter: 1ESY1161229886

[main  ] INFO 2021/03/11 19:45:48 evcc 0.48 (d4f8d03)
[main  ] INFO 2021/03/11 19:45:48 using config file C:\temp\evcc\evcc.yaml
Power: unexpected status: 400

Rufe ich die API per Hand auf , gibt es noch etwas mehr Output:
400 Bad Request: Invalid meterId: 1ESY1161229886

andig, 2021-03-11:
Bitte mit —log trace ausführen

KarlJep, 2021-03-11:
Hier noch eine valide meterId für das demo konto : 774513c8890f4dee8c013878e497ce00

Und hier noch das Log: 
[main  ] INFO 2021/03/11 20:11:04 evcc 0.48 (d4f8d03)
[main  ] INFO 2021/03/11 20:11:04 using config file C:\temp\evcc\evcc.yaml
[discgy] TRACE 2021/03/11 20:11:04 GET /public/v1/last_reading?meterId=2fdfd02cd6314e25bc71c6419db94ad HTTP/1.1
Host: api.discovergy.com
Authorization: Basic a2FybC5qQHdha3VlYW5kbW9yZS5kZTpocFpnaHFiQzczR2R3OVQ=
[discgy] TRACE 2021/03/11 20:11:06 HTTP/1.1 400 Bad Request
Content-Length: 65
Access-Control-Allow-Headers: X-Requested-With, Content-Type
Access-Control-Allow-Methods: GET, POST, DELETE, PUT
Access-Control-Allow-Origin: *
Connection: keep-alive
Content-Type: text/plain
Date: Thu, 11 Mar 2021 19:11:06 GMT
Server: Discovergy 1.0.1

400 Bad Request: Invalid meterId: 2fdfd02cd6314e25bc71c6419db94ad
Power: unexpected status: 400

andig, 2021-03-11:
Nimm mal das meter raus. Vmtl ist die id einfach falsch

KarlJep, 2021-03-11:
Mh welches Meter soll ich rausnehmen ? 
Benutze ich statt der 1ESY1161229886  die 774513c8890f4dee8c013878e497ce00 funktioniert alles wie es soll

andig, 2021-03-11:
Genau das. Raus damit, dann siehst du was falsch ist

KarlJep, 2021-03-11:
Mhh, ja , ganz ohne Angabe einer Id funktioniert es.
Dann hat mich wohl die Beispielconfig auf den falschen Weg gebracht. 

Wenn ich diesen Thread auch gleich noch für eine weitere Frage kapern darf; wie greife ich bei dem Zähler auf das Einspeisezählwerk zu? ich würde den gleichen Zähler auch für die Einspeisung verwenden

andig, 2021-03-11:
Vmtl hat Discovergy einen der beiden Demozähler- und zwar den falschen- aus der Demo entfernt. Deshalb geht das ursprüngliche Beispiel nicht mehr. Ohne Angabe wird der andere, verbliebene, Zähler automatisch gefunden.

closed 2021-03-11 by andig