evcc-io/evcc #751: climater 401 error after token refresh failed: unexpected status: 400

bicaluv, 2021-03-12:
**Describe the bug**
I'm not 100% sure if is a bug, cause I can not reproduce it on demand.

**To Reproduce**
Keep evcc up and running over some time.
Have to restart evcc to get rid of this 401 error

**Expected behavior**
Not getting 401 for climater.

**EVCC details:**
Output from **syslog**:
```
Mar 12 16:38:59 raspberrypi evcc[9094]: [lp-1  ] DEBUG 2021/03/12 16:38:59 targetSoC reached: 84.0 > 80
Mar 12 16:38:59 raspberrypi evcc[9094]: [lp-1  ] DEBUG 2021/03/12 16:38:59 climater active: false, target temp: 19.0°C, outside temp: 8.5°C
Mar 12 16:39:09 raspberrypi evcc[9094]: [lp-1  ] DEBUG 2021/03/12 16:39:09 charge power: 0W
Mar 12 16:39:09 raspberrypi evcc[9094]: [lp-1  ] DEBUG 2021/03/12 16:39:09 charger status: B
Mar 12 16:39:09 raspberrypi evcc[9094]: [lp-1  ] DEBUG 2021/03/12 16:39:09 targetSoC reached: 84.0 > 80
Mar 12 16:39:09 raspberrypi evcc[9094]: [lp-1  ] DEBUG 2021/03/12 16:39:09 climater active: false, target temp: 19.0°C, outside temp: 8.5°C
Mar 12 16:39:19 raspberrypi evcc[9094]: [lp-1  ] DEBUG 2021/03/12 16:39:19 charge power: 0W
Mar 12 16:39:19 raspberrypi evcc[9094]: [lp-1  ] DEBUG 2021/03/12 16:39:19 charger status: B
Mar 12 16:39:19 raspberrypi evcc[9094]: [lp-1  ] DEBUG 2021/03/12 16:39:19 targetSoC reached: 84.0 > 80
Mar 12 16:39:19 raspberrypi evcc[9094]: [skoda ] ERROR 2021/03/12 16:39:19 token refresh failed: unexpected status: 400
Mar 12 16:39:20 raspberrypi evcc[9094]: [lp-1  ] ERROR 2021/03/12 16:39:20 climater: unexpected status: 401
Mar 12 16:39:29 raspberrypi evcc[9094]: [lp-1  ] DEBUG 2021/03/12 16:39:29 charge power: 0W
Mar 12 16:39:29 raspberrypi evcc[9094]: [lp-1  ] DEBUG 2021/03/12 16:39:29 charger status: B
Mar 12 16:39:29 raspberrypi evcc[9094]: [lp-1  ] DEBUG 2021/03/12 16:39:29 targetSoC reached: 84.0 > 80
Mar 12 16:39:29 raspberrypi evcc[9094]: [lp-1  ] ERROR 2021/03/12 16:39:29 climater: unexpected status: 401
Mar 12 16:39:39 raspberrypi evcc[9094]: [lp-1  ] DEBUG 2021/03/12 16:39:39 charge power: 0W

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
[main  ] INFO 2021/03/12 16:46:20 evcc 0.48 (dc67e9b)
[main  ] INFO 2021/03/12 16:46:20 using config file evcc.yaml
config
======

grid: sma-grid
--------------
Power:          -1035W
Current L1..L3: 1.11A 1.93A 2.42A

pv: sma-pv
----------
Power: 1780W

loadpoint 1
===========

charger: go-eCharger
--------------------
Power:          0W
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        false
Charged:        0.1kWh

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
only snipping vehicle config here:

vehicles:
- name: Citigo
  type: skoda
  title: CITIGOe iV
  capacity: 36 # kWh
  user: xxx@yyy.de
  password: pw1245
  vin: vin12345

</pre></details>

<details><summary>Show evcc log output with <code>--log trace</code>:</summary><br><pre>
<- paste here

</pre></details>

andig, 2021-03-12:
Kannst Du das mal mit trace log machen damit man sieht was passiert? Gerne per mail an cpuidle@gmx.de oder anonymisieren.

bicaluv, 2021-03-12:
Gerne.
Hast du einen Tipp, wie man das ggf. forcieren kann? Also dass der Token erneuert werden muss?

andig, 2021-03-12:
Du kannst im Code in vw/identity/token basteln. Vielleicht sollten da einfach mal 10s Sicherheitspuffer rein?

bicaluv, 2021-03-13:
Habe da so einen Verdacht: das Token liefert ja als `ExpiresIn` den Wert 3599 (Sekunden), was 59,983 Minuten sind. Der poll findet aber, wenn nicht geladen wird, alle 60 Minuten statt, also knapp zu spät. Ich teste mal mit 58 Minuten poll interval.
Ansonsten nur her mit Tipps, wo ich hinschauen soll.
Ach ja, habe noch nicht herausgefunden, wie es `climater: unexpected status: 401` ausgelöst wird. Lasse gerade alles im Debugger laufen und habe mal einen Breakpoint im `loadpoint.go` gesetzt, um mir dann im Fehlerfall den Callstack anzusehen.

bicaluv, 2021-03-13:
@andig : So, Callstack zum Fehler (poll Intervall steht auf 60m, Auto wird nicht geladen):

```
github.com/andig/evcc/vehicle/vw.(*Identity).Token (/Users/matthias/Documents/evcc/evcc/vehicle/vw/identity.go:246)
github.com/andig/evcc/vehicle/vw.(*API).getJSON (/Users/matthias/Documents/evcc/evcc/vehicle/vw/api.go:58)
github.com/andig/evcc/vehicle/vw.(*API).Climater (/Users/matthias/Documents/evcc/evcc/vehicle/vw/api.go:174)
github.com/andig/evcc/vehicle/vw.NewProvider.func2 (/Users/matthias/Documents/evcc/evcc/vehicle/vw/provider.go:24)
github.com/andig/evcc/provider.(*Cached).InterfaceGetter.func1 (/Users/matthias/Documents/evcc/evcc/provider/cache.go:142)
github.com/andig/evcc/vehicle/vw.(*Provider).Climater (/Users/matthias/Documents/evcc/evcc/vehicle/vw/provider.go:92)
github.com/andig/evcc/core.(*LoadPoint).climateActive (/Users/matthias/Documents/evcc/evcc/core/loadpoint.go:541)
github.com/andig/evcc/core.(*LoadPoint).Update (/Users/matthias/Documents/evcc/evcc/core/loadpoint.go:985)
github.com/andig/evcc/core.(*Site).update (/Users/matthias/Documents/evcc/evcc/core/site.go:311)
github.com/andig/evcc/core.(*Site).Run (/Users/matthias/Documents/evcc/evcc/core/site.go:364)
github.com/andig/evcc/cmd.run.func1 (/Users/matthias/Documents/evcc/evcc/cmd/root.go:231)
runtime.goexit (/usr/local/go/src/runtime/asm_amd64.s:1371)
```
Was ich leider noch nicht verstehe, was da genau passiert. Aufgrund eines Cache Timeouts wird ein neuer vehicle api provider angelegt.
Weitere Info:
```
[main  ] INFO 2021/03/13 12:37:41 evcc 0.0.1-alpha (HEAD)
...
[lp-1  ] DEBUG 2021/03/13 13:38:50 pv max charge current: 0A
```
--> Fehler, also nach über 60 Minuten.

andig, 2021-03-13:
Mach doch einfach den timeout mal 1min kleiner und schau obs dann geht

bicaluv, 2021-03-14:
Das hat geklappt (58m).
ABER: heute Nacht hatte ich zufällig Probleme mit meinem Router und der Netzverbindung.
Folge (XXXXXXXXXX ist die von mir entferne VIN und 192.168.2.1 ist der Router): 
`climater: Get \"https://msg.volkswagen.de/fs-car/bs/climatisation/v1/VW/CZ/vehicles/XXXXXXXXXX/climater\": dial tcp: lookup msg.volkswagen.de on 192.168.2.1:53: server misbehaving`
Auch, als der Router und die DNS-auflösung wieder funktioniert hat.
Hat der Climater einen eigenen Token oder ein anderes Refreshing? wie schon geschrieben, verstehe den Callstack im Fehlerfall nicht so ganz.

andig, 2021-03-14:
Dein dns server (in der fritzbox) spinnt port 53 ist dns. Sollte ein ausnahmefall sein. Lass uns mal die 2min (oder 1min) Karenz einbauen.

andig, 2021-03-14:
Willst du pr machen?

bicaluv, 2021-03-14:
Dass es mein Server war, der DNS nicht auflösen konnte bzw. gar nicht erreichbar war ist schon klar.
Aber mir geht es darum, dass es unterschiedliche Gründe sind/waren, warum der Refresh nicht zeitgerecht erfolgen konnte und danach erholt sich der Climater nicht mehr. Eine Karenz einzubauen behebt das Problem nicht wirklich, es muss auch der Cache neu aufgebaut werden (zumindest für den Climater).
PR kann ich machen und mein Glück versuchen.

andig, 2021-03-14:
> und danach erholt sich der Climater nicht mehr.

kann ich nicht erkennen. Wenn dns kaputt gehts halt nicht. Der cache merkt sich auch Fehler, expired aber auch die irgendwann, eben nach 15min.

closed 2021-03-14 by andig