evcc-io/evcc #767: Enhance MQTT interface

drivingmx, 2021-03-16:
Due to my own implemented energy manager in nodered, I'm looking for any possibility to limit or command charging power/ current/ energy for all loadpoints as one. (energy allocation for evcc) 

I looked through the API code and found "ChargerEx" . I understand this as an external max current limit (per loadpoint?). 
This would be fine for me. 

Is there any complement for a min current limit? 

How can I address these values via MQTT?

andig, 2021-03-16:
Aktuell ist das nicht möglich. Wir könnten überlegen, ob es nicht- in Ergänzung zu OCPP und SHM- ein "externes" Lastmanagement System geben sollte, dass sich analog der Zähler etc in EVCC integrieren ließe und die Möglichkeit bekäme, in die Ladevorgänge einzugreifen.

Die Umsetzung ist vmtl. trivial, ein sauberes Konzept aber nicht. Was es dafür bräuchte wäre eine Beschreibung des Leistungsumfangs für so ein Lastmanagement und die Arbeitsweise in Kombination mit den EVCC Lademodi. Anregungen gibts im Quellcode der SHM Implementierung.

andig, 2021-03-16:
> Is there any complement for a min current limit?

Das gibts im Configfile

premultiply, 2021-03-16:
Die SHM-Steuerung benötigt für dessen Vorgaben ohnehin noch ein externes "Per-Charger-Limit" welches die weiteren Limits weiter einschränkt. Für ein Charger übergreifendes Limit bräuchte man erst einmal einen Algorithmus oder zumindest eine Logik wie dies über die einzelnen Charger verteilt wird, insbesondere dann es es nicht für alle die wollen reicht.

drivingmx, 2021-03-16:
Man kann auch Vereinfachen: 
Wenn es noch kein Management für die Energieverteilung zwischen Loadpoints gibt, kann man das auch erstmal gedanklich in die HEMS Schnittstelle verlegen. Dann müsste die pro Ladepunkt Zugriff auf die benötigten Parameter ermöglichen.
Man kann noch weiter vereinfachen:
Die dynamische Limitierung auf Leistungsgrenzen (oder auch Stromgrenzen) könnten über API kommandiert werden.

Ich glaube die Instanzierung eines generic hems bringt keine höherwertigen Vorteile. Da die Anbindung eh eine eigens erschaffene Adapterschicht benötigt, kann man in dieser auch die verbleibende Logik unterbringen.

drivingmx, 2021-03-16:
> Was es dafür bräuchte wäre eine Beschreibung des Leistungsumfangs für so ein Lastmanagement und die Arbeitsweise in Kombination mit den EVCC Lademodi.

Nunja, das hängt auch an den Umfängen der Steuerungsmöglichkeiten des hems.
Ich stelle mir vor dass ein hems in Grenzen (Min/Max werten kommendiert). Innerhalb dieser Grenzen darf sich der Client (=evcc) bewegen. Dabei stellt der client sicher, dass er sowohl die Grenzen des hems, als auch die Grenzen des chargers und des vehicles berücksichtigt.

Ich würde diese Begrenzungen nur im MinPV und im PV Modus berücksichtigen.
Im PV Modus ist die Wirkung selbstredend. Im Min + PV Modus würden sich die Grenzen nur auf den PV Pfad auswirken, das Auffüllen mit Netzbezug würde sich dann ergeben.

premultiply, 2021-03-16:
Wozu Min-Werte?
Ich denke Freigabe ja/nein und Max. Leistungsvorgabe sollte reichen.

drivingmx, 2021-03-16:
Stimmt. Min Werte könnte man nutzen um bewusst dynamisch die PV Batterie ladung zu überstimmen. Ist aber eher ein prophylaktischer Anwendungsfall.
Die Freigabe ist in der Leistungsgrenze schon enthalten.

andig, 2021-03-16:
> Ich glaube die Instanzierung eines generic hems bringt keine höherwertigen Vorteile. 

Das kann man so sehen. Allerdings merke ich SHM und OCPP in evcc heute an, dass sie dran geflanscht wurden. Wäre schön, das mal anforderungsgetrieben "richtig" zu machen.

> Da die Anbindung eh eine eigens erschaffene Adapterschicht benötigt, kann man in dieser auch die verbleibende Logik unterbringen.

Die Frage ist genau, welche Informationen/Aktoren so eine Adapterschicht braucht.

drivingmx, 2021-03-16:
Zusätzliche, über die bisherige API hinaus gehende Informationen benötigt die Adapterschicht nicht von "unten" (evcc). Über die Adapterschicht werden auf einer zu vereinbarenden physikalischen Ebene Kontingente zugeteilt. Hier bietet sich Leistung an. 
Das heißt aus meiner Sicht, dass vor allem evcc auf die Kontingente hört.
Die anderen Informationen für eine Leistungskontingentzuteilung sind Teil einer Strategie auf höherer Flugebene. Da kann man Informationen bis hin zu PV Ertragsprognose heranziehen. Das würde ich ungern in evcc integrieren. Das übersteigt den scope von evcc.

Konkret als Anforderung: 
Schnittstelle loadpoint/n/pvmax: maximale zulässige PV Leistung pro Ladepunkt [W] 0..22001 W
Die Schnittstelle soll dynamisch einstellbar sein.
Im PV Modus: evcc soll diese berücksichtigen und die Ladeleistung darauf begrenzen (so als ob die Erzeugungsleistung weniger werden würde).
Im min + PV modus: evcc soll den verwendeten PV Leistungsanteil auf das Kontingent begrenzen und zur Erreichung der Mindestladeleistung den Netzbezug erhöhen. (so als ob die Erzeugungsleistung weniger werden würde)
0 W bedeutet PV Bezug verboten.
22001W bedeutet keine Limitierung des PV Leistungsanteils.

andig, 2021-03-16:
Scheint mir ehr eingeschränkt und passt auch nicht zum SHM. Ich würde da erstmal gar nicht implementieren bis es ein wirklich gutes Konzept gibt.

drivingmx, 2021-03-17:
> Warum PV Leistung begrenzen statt Leistung?

Zu aller erst: ich denke die Entscheidung über die (Ziel)ladeleistung sollte evcc entscheiden. Hier geht es nur um die Verfügbarkeit von PV Energie zur Erreichung des Ziels. (Eine Gesamtleistungssteuerung kann auch Sinn machen, steht hier aber nicht im Fokus).
Im PV Modus ist PV die einzige Energiequelle. Damit hat man also Durchgriff auf die Gesamtleistung.
Im min + PV Modus muss es die PV Leistung sein, weil man ansonsten die Mindestladeleistung torpediert. Netzbezug muss eben immer auffüllen, was an PV nicht zur Verfügung steht. Bei Gesamtleistungsbegrenzung wäre der Modus dann nicht min sondern eher ein manueller Modus: P_Laden = P_PV + (P_Laden - P_PV). Dann kann es aber sein, das mehr PV Leistung aufgebracht wird, als das der Energiemanager vorsehen würde.
Übergreifend ist es so, dass nur die PV Leistung aus Sicht eines Energiemanagements Konflikte auslöst, die durch Verteilung und Zuweisung gelöst werden können. Bspw. der Konflikt zwischen PV Batterie- und Fzg Batterieladung. Wenn man die Energie für die PV Batterie priorisieren möchte hält man der eben ein Kontingent vor. 
Der SOC in der Konfig hilft nicht: denn evcc verdrängt die PV Batterieladung. Die PV Batterie lädt aus der Differenz zwischen Hausverbrauch und Ertrag. EVCC zieht aber die Hausverbrauch hoch, weswegen die Batterie keine Ladeleistung sieht.
Das ist auch die Antwort auf die Frage:
> Wozu?

Ich bin nicht im Bilde was sich SMA mit dem SHM gedacht hat, aber aus meiner laienhaften Sicht ist das von mir geschilderte Konzept genau das was man braucht um zentral entscheiden zu können wo die PV Energie adressiert wird. 

Ich werde mir nun erstmal behelfen in dem ich die PV Power aus einer Zwischenschicht per MQTT bediene. Da kann ich dann die Kontingentierung vornehmen. Somit sieht evcc nicht die gesamte Leistung, sondern nur die vorkontingentierte.

premultiply, 2021-03-17:
EVCC schaut nur auf die Bilanz am Gridmeter (und ggf. der Batterie).
PV-Erzeugung spielt keine Rolle und wird höchstens in der GUI angezeigt.

drivingmx, 2021-03-17:
Danke guter Hinweis, dann nehme ich das Gridmeter.

premultiply, 2021-03-17:
Zurück zum Thema Lastmanagement:
Ich sehe da nur zwei Fälle:
1) Die potenzielle Summenladeleistung ist höher als der Netzanschluss oder die Installation aushält.
2) Ein Energiemanagementsystem schränkt temporär die verfügbare Ladeleistung ein um andere Verbraucher (prognosebasiert) zu bevorzugen (z.B. Batterieladung, Warmwassererzeugung, ...).

drivingmx, 2021-03-17:
Korrekt. 
1. entspricht meiner Vorstellung von Lastmanagement.
2. entspricht meiner Vorstellung von Energiemanagement

drivingmx, 2021-03-17:
perspektivisch kann man auch  noch anführen:
3. Kommandierte Ladung wegen smart grid trigger (günstiger Strom)

premultiply, 2021-03-17:
Kann man mit 2) abbilden.

Man kann keine Ladung und auch keine Leistungsaufnahme erzwingen. Man kann nur anbieten/zulassen und Limits vorgeben.

drivingmx, 2021-03-17:
Stimmt, wenn man die Leistung nicht auf die PV Leistung beschränkt

premultiply, 2021-03-17:
An erster Stelle spielt ja auch die Benutzerwahl des Lademodus und die Intention dahinter eine Rolle.
Während „Sofort“ aus Nutzersicht eher mit „so schnell wie möglich, koste es was es wolle“ zu übersetzen ist bei „PV“ eher „mir doch egal wie lange es dauert solange es nix kostet“ die Maxime.

Trotzdem muss auch bei „Sofort“ immer ein Lastmanagement greifen um die Infrastruktur zu schützen, denn wenn es die Sicherung raus haut dauert es länger als der Nutzer erwartet.

drivingmx, 2021-03-17:
Exakt.
Damit sind wir bei 2 Grenzen:
Die erste aus Sicht Anlagenschutz (Summenanschlussleistungsbegrenzung - schönes Wort).
Die zweite aus Sicht PV Energieverteilung.

andig, 2021-03-20:
I denke hier gibt es keine schnelle Lösung mangels Gesamtkonzept. Erstmal wontfix.

closed 2021-03-20 by andig