evcc-io/evcc #776: Unterstützung für EM-CP-PP-ETH ohne EEM-MA250

jrester, 2021-03-18:
**Is your feature request related to a problem? Please describe.**
Die Integration für die phoenix contact EM-CP-PP-ETH nimmt an, dass das Modul EEM-MA250 für power und energy readings angeschlossen ist. Das ist aber nicht unbedingt immer der Fall. Dann ließt evcc falsche Werte aus und gibt z.B. 42950000.0 kWh für die Leistung an, was ein offensichtlich unmöglicher Wert ist.

**Describe the solution you'd like**
Es sollte möglich sein, für die EM-CP-PP-ETH power und energy readings in der config zu deaktivieren. Ich wäre bereit diese Lösung in einem PR zu implementieren.

**Describe alternatives you've considered**
Man könnte versuchen automatisch zu erkennen, ob das EEM-MA250 Modul angeschlossen ist, indem man die Werte für die power readings überprüft. Ist es nämlich nicht verbunden scheint in den jeweiligen Registern immer `0xFFFF` zu stehen. 

**Additional context**
Ich arbeite mit einer UM DE EV Charge Control 2902802 mit Firmeware version 2.0.5

premultiply, 2021-03-18:
Lass einfach die Zählereinstellungen in der Konfig weg.

```
- type: phoenix-em-eth
  uri: 192.168.0.8:502
```

andig, 2021-03-18:
> Es sollte möglich sein, für die EM-CP-PP-ETH power und energy readings in der config zu deaktivieren.

Das ist bereits so. Du musst sie aktiv einschalten:

	cc := struct {
		URI   string
		ID    uint8
		Meter struct {
			Power, Energy, Currents bool
		}
	}{
		URI: "192.168.0.8:502", // default
		ID:  180,               // default
	}

> Die Integration für die phoenix contact EM-CP-PP-ETH nimmt an, dass das Modul EEM-MA250 für power und energy readings angeschlossen ist. 

Es wird gar nichts angenommen, sondern die Werte aus dem Charger ausgelesen- dort muss die Konfiguration korrekt sein falls Du die Verwendung der Messwerte einschaltest.

Damit wir nicht raten müssen gerne vollständiges Ticket mit Config und `evcc dump`

jrester, 2021-03-18:
Oh, sorry! Zu blöd die Doku zu lesen. Danke!

closed 2021-03-18 by jrester

premultiply, 2023-03-29:
Kannst du mal bitte `evcc charger --log trace` ausführen?

EKona000, 2023-03-30:
[main  ] INFO 2023/03/30 10:52:42 evcc 0.114.1
[main  ] INFO 2023/03/30 10:52:42 using config file: /etc/evcc.yaml
[db    ] INFO 2023/03/30 10:52:42 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2023/03/30 10:52:42 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/03/30 10:52:42 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/03/30 10:52:42 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/03/30 10:52:42 SELECT * FROM `settings` 0 <nil>
[em-eth] TRACE 2023/03/30 10:52:42 modbus: send 00 01 00 00 00 06 b4 04 00 78 00 02
[em-eth] TRACE 2023/03/30 10:52:42 modbus: recv 00 01 00 00 00 07 b4 04 04 ff ff ff ff
[em-eth] TRACE 2023/03/30 10:52:42 modbus: send 00 02 00 00 00 06 b4 04 00 80 00 02
[em-eth] TRACE 2023/03/30 10:52:42 modbus: recv 00 02 00 00 00 07 b4 04 04 ff ff ff ff
[em-eth] TRACE 2023/03/30 10:52:42 modbus: send 00 03 00 00 00 06 b4 04 00 72 00 02
[em-eth] TRACE 2023/03/30 10:52:42 modbus: recv 00 03 00 00 00 07 b4 04 04 ff ff ff ff
[em-eth] TRACE 2023/03/30 10:52:42 modbus: send 00 04 00 00 00 06 b4 04 00 74 00 02
[em-eth] TRACE 2023/03/30 10:52:42 modbus: recv 00 04 00 00 00 07 b4 04 04 ff ff ff ff
[em-eth] TRACE 2023/03/30 10:52:42 modbus: send 00 05 00 00 00 06 b4 04 00 76 00 02
[em-eth] TRACE 2023/03/30 10:52:42 modbus: recv 00 05 00 00 00 07 b4 04 04 ff ff ff ff
[em-eth] TRACE 2023/03/30 10:52:42 modbus: send 00 06 00 00 00 06 b4 04 00 64 00 01
[em-eth] TRACE 2023/03/30 10:52:42 modbus: recv 00 06 00 00 00 05 b4 04 02 00 42
[em-eth] TRACE 2023/03/30 10:52:42 modbus: send 00 07 00 00 00 06 b4 01 01 90 00 01
[em-eth] TRACE 2023/03/30 10:52:42 modbus: recv 00 07 00 00 00 04 b4 01 01 00
[em-eth] TRACE 2023/03/30 10:52:42 modbus: send 00 08 00 00 00 06 b4 04 00 66 00 02
[em-eth] TRACE 2023/03/30 10:52:42 modbus: recv 00 08 00 00 00 07 b4 04 04 0d 9b 00 00
Power:          42949672950W
Energy:         42949673.0kWh
Current L1..L3: 4.29e+06A 4.29e+06A 4.29e+06A
Charge status:  B
Enabled:        false
Duration:       58m3s

Hilft das?
Danke