evcc-io/evcc #779: Puffer für die Einspeisung

gyroslav, 2021-03-19:
Es gibt ja die threshold im ENABLE und DISABLE delay. Diese steuern aber nur Start und Stop. 
Ich würde gerne noch einen Puffer des Überschusses während der Ladezeit haben. 

Beispiel: der Überschuss ist 7,5KW, mein Auto lädt mit 7,2KW, Rest 300 Watt. Dann geht z.B. der Herd an, Zieht 1 KW. 
Macht insgesamt -700 Watt, die ich jetzt aus dem Netz beziehe. EVCC merkt das aber erst verzögert. Und regelt erst später runter.
Der Puffer könnte sowas abfangen, weil EVCC dann schon von forne herein nur 6,5KW frei geben würde. 

Dieser Puffer könnte den DISABLE-threshold ersetzen, weil sobald die Wirkleistung niedriger als der Puffer ist, schaltet er ja sowieso ab. Und wenn man den Puffer negativ eingeben kann, dann wird auch noch Netzbezug erlaubt, wie der aktuelle DISABLE-threshold halt auch. 

Schöne Grüße!
Eugen

andig, 2021-03-19:
Du kannst `site.residualpower` nutzen: https://github.com/andig/evcc/blob/master/core/site.go#L36

closed 2021-03-19 by andig

gyroslav, 2021-03-27:
Wie gebe ich das in der evcc.yaml an?

andig, 2021-05-08:
Siehe Link oben