evcc-io/evcc #782: No bug, but hint: Deprecated: golang.org/x/oauth2: Transport.CancelRequest no longer does anything; use contexts

thierolm, 2021-03-19:
Starting evcc version 0.50 the following message is displayed in the log:

```
tserver evcc[810]: 2021/03/19 03:31:57 deprecated: golang.org/x/oauth2: Transport.CancelRequest no longer does anything; use contexts
```
I searched for evcc sources importing oauth2 package, but found no location where Transport.CancelReques is executed.

So this issue is more a hint and currently no bug!

andig, 2021-03-19:
Bei mir sehe ich das nicht- kannst du das auf einen bestimmten commit zurück führen (git bisect)? Welche Devices hast Du im Einsatz?

thierolm, 2021-03-19:
Ich habe den go-e charger als lp-1 und den fritzdect als lp-2.
Die Meldung kommt nur einmal bei der Initialisierung am Anfang am ende der lp-1 Meldungen dann nicht mehr (siehe log).
Ich hab' das offizielle 0.50 binary im Einsatz. 

Entweder tritt die Meldung auf, am Ende der go-e Initialisierung, oder der es hat etwas mit dem request.InsecureTransport() für die https Verbindung mit der Fritzbox zu tun.
 
```
Mär 19 19:21:25 tserver systemd[1]: Stopped evcc.
Mär 19 19:21:25 tserver systemd[1]: Started evcc.
Mär 19 19:21:29 tserver evcc[108936]: [lp-1  ] INFO 2021/03/19 19:21:29 loadpoint 1:
Mär 19 19:21:29 tserver evcc[108936]: [lp-1  ] INFO 2021/03/19 19:21:29   mode:      pv
Mär 19 19:21:29 tserver evcc[108936]: [lp-1  ] INFO 2021/03/19 19:21:29   charger:   power ✓ energy — currents ✓ timer —
Mär 19 19:21:29 tserver evcc[108936]: [lp-1  ] INFO 2021/03/19 19:21:29   meters:    charge ✓
Mär 19 19:21:29 tserver evcc[108936]: [lp-1  ] INFO 2021/03/19 19:21:29     charge:  power ✓ energy — currents ✓
Mär 19 19:21:29 tserver evcc[108936]: [lp-1  ] INFO 2021/03/19 19:21:29   vehicles:  ✓
Mär 19 19:21:29 tserver evcc[108936]: [lp-1  ] INFO 2021/03/19 19:21:29     car 0:   range ✓ finish ✓ status ✓ climate ✓
Mär 19 19:21:29 tserver evcc[108936]: [lp-1  ] INFO 2021/03/19 19:21:29     car 1:   range ✓ finish ✓ status ✓ climate ✓
Mär 19 19:21:29 tserver evcc[108936]: [lp-2  ] INFO 2021/03/19 19:21:29 loadpoint 2:
Mär 19 19:21:29 tserver evcc[108936]: [lp-2  ] INFO 2021/03/19 19:21:29   mode:      pv
Mär 19 19:21:29 tserver evcc[108936]: [lp-2  ] INFO 2021/03/19 19:21:29   charger:   power ✓ energy — currents — timer —
Mär 19 19:21:29 tserver evcc[108936]: [lp-2  ] INFO 2021/03/19 19:21:29   meters:    charge ✓
Mär 19 19:21:29 tserver evcc[108936]: [lp-2  ] INFO 2021/03/19 19:21:29     charge:  power ✓ energy — currents —
Mär 19 19:21:29 tserver evcc[108936]: [lp-2  ] INFO 2021/03/19 19:21:29   vehicles:  ✓
Mär 19 19:21:29 tserver evcc[108936]: [lp-2  ] INFO 2021/03/19 19:21:29     car 0:   range — finish — status — climate —
Mär 19 19:21:30 tserver evcc[108936]: [lp-1  ] DEBUG 2021/03/19 19:21:30 charge power: 0W
Mär 19 19:21:30 tserver evcc[108936]: [lp-1  ] DEBUG 2021/03/19 19:21:30 charger status: B
Mär 19 19:21:31 tserver evcc[108936]: [lp-1  ] DEBUG 2021/03/19 19:21:31 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 19 19:21:31 tserver evcc[108936]: [lp-1  ] DEBUG 2021/03/19 19:21:31 vehicle soc: 61%
Mär 19 19:21:31 tserver evcc[108936]: [lp-1  ] DEBUG 2021/03/19 19:21:31 vehicle range: 134km
Mär 19 19:21:31 tserver evcc[108936]: [lp-1  ] DEBUG 2021/03/19 19:21:31 max charge current: 0A = 0A + -6.7A (1546W @ 1p)
Mär 19 19:21:31 tserver evcc[108936]: [lp-1  ] DEBUG 2021/03/19 19:21:31 pv max charge current: 0A
Mär 19 19:21:38 tserver evcc[108936]: 2021/03/19 19:21:38 deprecated: golang.org/x/oauth2: Transport.CancelRequest no longer does anything; use contexts
Mär 19 19:22:31 tserver evcc[108936]: [lp-2  ] DEBUG 2021/03/19 19:22:31 charge power: 0W
Mär 19 19:22:31 tserver evcc[108936]: [lp-2  ] DEBUG 2021/03/19 19:22:31 charger status: B
Mär 19 19:22:31 tserver evcc[108936]: [lp-2  ] DEBUG 2021/03/19 19:22:31 vehicle soc: 95%
Mär 19 19:22:31 tserver evcc[108936]: [lp-2  ] DEBUG 2021/03/19 19:22:31 max charge current: 0A = 0A + -6.9A (1595W @ 1p)
Mär 19 19:22:31 tserver evcc[108936]: [lp-2  ] DEBUG 2021/03/19 19:22:31 pv max charge current: 0A
Mär 19 19:23:30 tserver evcc[108936]: [lp-1  ] DEBUG 2021/03/19 19:23:30 charge power: 0W
Mär 19 19:23:30 tserver evcc[108936]: [lp-1  ] DEBUG 2021/03/19 19:23:30 charger status: B
Mär 19 19:23:30 tserver evcc[108936]: [lp-1  ] DEBUG 2021/03/19 19:23:30 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 19 19:23:30 tserver evcc[108936]: [lp-1  ] DEBUG 2021/03/19 19:23:30 max charge current: 0A = 0A + -3.6A (831W @ 1p)
Mär 19 19:23:30 tserver evcc[108936]: [lp-1  ] DEBUG 2021/03/19 19:23:30 pv max charge current: 0A
Mär 19 19:24:30 tserver evcc[108936]: [lp-2  ] DEBUG 2021/03/19 19:24:30 charge power: 0W
Mär 19 19:24:31 tserver evcc[108936]: [lp-2  ] DEBUG 2021/03/19 19:24:31 charger status: B
Mär 19 19:24:31 tserver evcc[108936]: [lp-2  ] DEBUG 2021/03/19 19:24:31 max charge current: 0A = 0A + -3.5A (796W @ 1p)
Mär 19 19:24:31 tserver evcc[108936]: [lp-2  ] DEBUG 2021/03/19 19:24:31 pv max charge current: 0A
Mär 19 19:25:30 tserver evcc[108936]: [lp-1  ] DEBUG 2021/03/19 19:25:30 charge power: 0W
Mär 19 19:25:30 tserver evcc[108936]: [lp-1  ] DEBUG 2021/03/19 19:25:30 charger status: B
Mär 19 19:25:30 tserver evcc[108936]: [lp-1  ] DEBUG 2021/03/19 19:25:30 vehicle status: B (ZOE Experience R135/Z.E. 50)
```

andig, 2021-03-19:
Eher nicht. Welches Auto hast du?

thierolm, 2021-03-19:
Renault ZOE Experience R135/Z.E. 50

andig, 2021-03-20:
> Entweder tritt die Meldung auf, am Ende der go-e Initialisierung, oder der es hat etwas mit dem request.InsecureTransport() für die https Verbindung mit der Fritzbox zu tun.

Kannst Du das rausfinden? Seit welchem Commit/Version tritt das auf?

thierolm, 2021-03-20:
Es muss am audi vehicle liegen (Hatte vergessen, dass ich meinen Q5 E-Tron auch in der Config habe). 
Die Meldung erscheint erst nach evcc Version 0.48 (ich hatte die 0.49 nicht installiert).

Im folgenden die Logs  (mit einer nur auf den go-e und das audi vehicle reduzierten Config):
Zuerst mit dem aktuellen 0.50 Release aus git:
```
xxxxxx@xxxxxx:/usr/local/bin$ sudo ./evcc_0.50 -c /home/thierolm/evcc/git/evcc/evcc_goe.yaml -l debug
[main  ] INFO 2021/03/20 15:55:06 evcc 0.50 (57495de)
[main  ] INFO 2021/03/20 15:55:06 using config file /home/thierolm/evcc/git/evcc/evcc_goe.yaml
[main  ] INFO 2021/03/20 15:55:06 listening at 0.0.0.0:7070
[lp-1  ] DEBUG 2021/03/20 15:55:08 max charge current: 7
[site  ] INFO 2021/03/20 15:55:08 site config:
[site  ] INFO 2021/03/20 15:55:08   meters:    grid ✓ pv ✓ battery —
[site  ] INFO 2021/03/20 15:55:08     grid:    power ✓ energy — currents —
[site  ] INFO 2021/03/20 15:55:08     pv:      power ✓ energy — currents —
[lp-1  ] INFO 2021/03/20 15:55:08 loadpoint 1:
[lp-1  ] INFO 2021/03/20 15:55:08   mode:      pv
[lp-1  ] INFO 2021/03/20 15:55:08   charger:   power ✓ energy — currents ✓ timer —
[lp-1  ] INFO 2021/03/20 15:55:08   meters:    charge ✓
[lp-1  ] INFO 2021/03/20 15:55:08     charge:  power ✓ energy — currents ✓
[lp-1  ] INFO 2021/03/20 15:55:08   vehicles:  ✓
[lp-1  ] INFO 2021/03/20 15:55:08     car 0:   range ✓ finish ✓ status ✓ climate ✓
[site  ] DEBUG 2021/03/20 15:55:08 ----
[site  ] DEBUG 2021/03/20 15:55:08 pv power: 1428W
[site  ] DEBUG 2021/03/20 15:55:09 grid power: -897W
[site  ] DEBUG 2021/03/20 15:55:09 site power: -897W
[lp-1  ] DEBUG 2021/03/20 15:55:09 charge power: 0W
[lp-1  ] DEBUG 2021/03/20 15:55:09 charger status: B
[lp-1  ] DEBUG 2021/03/20 15:55:10 vehicle soc: 16%
[lp-1  ] DEBUG 2021/03/20 15:55:10 vehicle range: 5km
[lp-1  ] DEBUG 2021/03/20 15:55:10 max charge current: 16
2021/03/20 15:55:17 deprecated: golang.org/x/oauth2: Transport.CancelRequest no longer does anything; use contexts
[site  ] DEBUG 2021/03/20 15:55:38 ----
[site  ] DEBUG 2021/03/20 15:55:38 pv power: 1437W
[site  ] DEBUG 2021/03/20 15:55:39 grid power: -918W
[site  ] DEBUG 2021/03/20 15:55:39 site power: -918W
[lp-1  ] DEBUG 2021/03/20 15:55:39 charge power: 0W
[lp-1  ] DEBUG 2021/03/20 15:55:39 charger status: B
```
Und jetzt die evcc Version 0.48 die ich mit meinem damals noch nicht gemergetem fritzdect Modul selbst kompiliert hatte.
Da kommt die Meldung nicht:
```
xxxxx@xxxxx:/usr/local/bin$ sudo ./evcc_0.48_fritzdect -c /home/thierolm/evcc/git/evcc/evcc_goe.yaml -l debug
[main  ] INFO 2021/03/20 15:55:53 evcc 0.48 (479bb1c)
[main  ] INFO 2021/03/20 15:55:53 using config file /home/thierolm/evcc/git/evcc/evcc_goe.yaml
[main  ] INFO 2021/03/20 15:55:53 listening at 0.0.0.0:7070
[lp-1  ] DEBUG 2021/03/20 15:55:55 max charge current: 7
[site  ] INFO 2021/03/20 15:55:55 site config:
[site  ] INFO 2021/03/20 15:55:55   meters:    grid ✓ pv ✓ battery —
[site  ] INFO 2021/03/20 15:55:55     grid:    power ✓ energy — currents —
[site  ] INFO 2021/03/20 15:55:55     pv:      power ✓ energy — currents —
[lp-1  ] INFO 2021/03/20 15:55:55 loadpoint 1:
[lp-1  ] INFO 2021/03/20 15:55:55   mode:      pv
[lp-1  ] INFO 2021/03/20 15:55:55   charger:   power ✓ energy — currents ✓ timer —
[lp-1  ] INFO 2021/03/20 15:55:55   meters:    charge ✓
[lp-1  ] INFO 2021/03/20 15:55:55     charge:  power ✓ energy — currents ✓
[lp-1  ] INFO 2021/03/20 15:55:55   vehicles:  ✓
[lp-1  ] INFO 2021/03/20 15:55:55     car 0:   range ✓ finish ✓ status ✓ climate ✓
[site  ] DEBUG 2021/03/20 15:55:55 ----
[site  ] DEBUG 2021/03/20 15:55:56 pv power: 1438W
[site  ] DEBUG 2021/03/20 15:55:56 grid power: -923W
[site  ] DEBUG 2021/03/20 15:55:56 site power: -923W
[lp-1  ] DEBUG 2021/03/20 15:55:56 charge power: 0W
[lp-1  ] DEBUG 2021/03/20 15:55:57 charger status: B
[lp-1  ] DEBUG 2021/03/20 15:55:57 vehicle soc: 16%
[lp-1  ] DEBUG 2021/03/20 15:55:57 vehicle range: 5km
[lp-1  ] DEBUG 2021/03/20 15:55:57 max charge current: 16
[site  ] DEBUG 2021/03/20 15:56:25 ----
[site  ] DEBUG 2021/03/20 15:56:26 pv power: 1437W
[site  ] DEBUG 2021/03/20 15:56:26 grid power: -887W
[site  ] DEBUG 2021/03/20 15:56:26 site power: -887W
[lp-1  ] DEBUG 2021/03/20 15:56:26 charge power: 0W
[lp-1  ] DEBUG 2021/03/20 15:56:26 charger status: B
```

Die audi vehicle config:
```
- name: audi
  type: audi
  title: Audi Q5 TFSI eTron # display name for UI
  capacity: 14 # kWh
  user: xxxxx # user
  password: fxxxx # password
  vin: WAUZXXXXXX0XXXX3 # optional
  cache: 5m
```

andig, 2021-03-20:
Yo, wirnutzen jetzt oauth2. Scheint ein Go Kompatibilitätsding zu sein. Nicht weiter relevant, passiert nur 1x. Ich machs mal zu, wer will kann die Frage in gonuts verfolgen.

closed 2021-03-20 by andig

andig, 2021-03-24:
See https://github.com/golang/oauth2/issues/487 and https://github.com/andig/evcc/commit/9d2d314b8a62ab813b2e173231275bb209dece34