evcc-io/evcc #788: evse-wifi: geladene Energie verdoppelt sich wenn Ladung beendet wird.

schenlap, 2021-03-20:
**Describe the bug**
Die Anzeige der geladenen Energie (+ influxdb) verdoppelt sich wenn die Ladung beendet wird (Soc Limit erreicht oder auf "Stop" gelickt)

**To Reproduce**
Steps to reproduce the behavior:
1. Auto Laden
2. z.B Energie steht bei 2kWh -> "Stop" drücken
3. Anzeige Energie ghet von 2kWh auf 4kWh

**Expected behavior**
Bei Stop bleibt die Anzeige der Energie gleich

**EVCC details:**
Release V0.5 precompiled

<details><summary>Show evcc log output with <code>--log trace</code>:</summary><br><pre>

{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":16,"actualPower":3.36,"durati
on":113000,"alwaysActive":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0.1,"mileage":0.5,"meterReading"
:3.91,"currentP1":14.81,"currentP2":0,"currentP3":0,"useMeter":true}]}
[lp-1  ] DEBUG 2021/03/20 10:40:49 charge power: 3360W
[influx] TRACE 2021/03/20 10:40:49 write chargePower=3360 (map[loadpoint:main])
[cache ] TRACE 2021/03/20 10:40:49 chargePower: 3360
[cache ] TRACE 2021/03/20 10:40:49 chargeCurrent: 16
[influx] TRACE 2021/03/20 10:40:49 write chargeCurrent=16 (map[loadpoint:main])
[evse  ] TRACE 2021/03/20 10:40:49 GET /getParameters HTTP/1.1
Host: 10.0.0.208
[evse  ] TRACE 2021/03/20 10:40:49 HTTP/1.1 200 OK
Connection: close
Content-Length: 320
Accept-Ranges: none
Content-Type: application/json

{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":16,"actualPower":3.36,"durati
on":113000,"alwaysActive":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0.1,"mileage":0.5,"meterReading"
:3.91,"currentP1":14.81,"currentP2":0,"currentP3":0,"useMeter":true}]}
[evse  ] TRACE 2021/03/20 10:40:49 GET /getParameters HTTP/1.1
Host: 10.0.0.208
[evse  ] TRACE 2021/03/20 10:40:49 HTTP/1.1 200 OK
Connection: close
Content-Length: 320
Accept-Ranges: none
Content-Type: application/json

{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":16,"actualPower":3.36,"durati
on":113000,"alwaysActive":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0.1,"mileage":0.5,"meterReading"
:3.91,"currentP1":14.81,"currentP2":0,"currentP3":0,"useMeter":true}]}
[cache ] TRACE 2021/03/20 10:40:49 chargedEnergy: 100.00000000000009
[influx] TRACE 2021/03/20 10:40:49 write chargedEnergy=100.00000000000009 (map[loadpoint:main])
[cache ] TRACE 2021/03/20 10:40:49 chargeDuration: 1m53s
[evse  ] TRACE 2021/03/20 10:40:49 GET /getParameters HTTP/1.1
Host: 10.0.0.208
[evse  ] TRACE 2021/03/20 10:40:49 HTTP/1.1 200 OK
Connection: close
Content-Length: 320
Accept-Ranges: none
Content-Type: application/json

{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":16,"actualPower":3.36,"durati
on":113000,"alwaysActive":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0.1,"mileage":0.5,"meterReading"
:3.91,"currentP1":14.81,"currentP2":0,"currentP3":0,"useMeter":true}]}
[lp-1  ] DEBUG 2021/03/20 10:40:49 charger status: C
[lp-1  ] TRACE 2021/03/20 10:40:49 soc estimated: 52.29% (vehicle: 51.00%)
[lp-1  ] DEBUG 2021/03/20 10:40:49 vehicle soc: 52%
[cache ] TRACE 2021/03/20 10:40:49 connected: true
[cache ] TRACE 2021/03/20 10:40:49 charging: true
[cache ] TRACE 2021/03/20 10:40:49 enabled: true
[cache ] TRACE 2021/03/20 10:40:49 socCharge: 52
[lp-1  ] DEBUG 2021/03/20 10:40:49 vehicle range: 14km
[evse  ] TRACE 2021/03/20 10:40:49 GET /getParameters HTTP/1.1
[... 193 lines cut ...]
Accept-Ranges: none
Content-Type: application/json

</pre></details>

Der Fehler ist hier schön zu sehen
<pre>
{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":16,"actualPower":3.36,"durati
on":117000,"alwaysActive":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0.1,"mileage":0.6,"meterReading"
:3.91,"currentP1":14.81,"currentP2":0,"currentP3":0,"useMeter":true}]}
[cache ] TRACE 2021/03/20 10:40:53 chargedEnergy: 100.00000000000009   # <--hier stimmt es noch
[cache ] TRACE 2021/03/20 10:40:53 chargeDuration: 1m57s
[evse  ] TRACE 2021/03/20 10:40:53 GET /getParameters HTTP/1.1
Host: 10.0.0.208
[evse  ] TRACE 2021/03/20 10:40:53 HTTP/1.1 200 OK
Connection: close

{"type":"parameters","list":[{"vehicleState":2,"evseState":false,"maxCurrent":16,"actualCurrent":0,"actualPower":0,"duration"
:117000,"alwaysActive":true,"lastActionUser":"API","lastActionUID":"API","energy":0.1,"mileage":0.6,"meterReading":4.02,"curr
entP1":0,"currentP2":0,"currentP3":0,"useMeter":true}]}
[evse  ] TRACE 2021/03/20 10:40:59 GET /getParameters HTTP/1.1
Host: 10.0.0.208

[cache ] TRACE 2021/03/20 10:40:59 chargedEnergy: 209.99999999999952 <-- hier sollte 100 stehen
</pre>

andig, 2021-03-20:
Wie immer: bitte vollständige Config und Logfile. Aus den Schnipseln lässt sich nichts erraten.

schenlap, 2021-03-20:
Hi,

log ist dabei, musst du nur aufklappen.

andig, 2021-03-21:
Ah. Config?

schenlap, 2021-03-21:
<details><summary>config <code></code>:</summary><br><pre>

uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

log: trace
#log: error
levels:
   core: trace
   influx: error
   lp-1: trace
#   #server: trace
   modbus: error
#   #modbus: trace
   ws: error
   #   wifi: trace
   audi: trace

# influx database (v1)
influx:
   url: http://10.0.0.180:8086
   database: evcc
#  user:
#  password:

# meters
meters:
- name: vzpv # pv Ertrag
  type: default
  power:
    type: http
    uri: http://volkszaehler/api/data/aaaaaaaa-bbbb-cccc-dddd-eeeeeeeeeeee.json?from=now
    jq: .data.tuples[0][1] # parse response json
- name: pvse
  type: default
  power:
    type: calc
    add:
     - type: modbus
       uri: 10.0.0.207:502
       model: sunspec
       value: Power
       id: 1
       scale: 1
     - type: modbus
       uri: 10.0.0.207:502
       model: sunspec
       value: Power
       id: 2
       scale: 1
- name: pvse1 # pv Ertrag from solaredge via modbus
  type: default
  power:
    type: modbus
    uri: 10.0.0.207:502
    id: 1
    model: sunspec
    value: Power
- name: pvse2 # pv Ertrag from solaredge via modbus
  type: default
  power:
    type: modbus
    uri: 10.0.0.207:502
    id: 2
    model: sunspec
    value: Power
- name: vzwb # Wallbox
  type: default
  power:
    type: http
    uri: http://volkszaehler/api/data/aaaaaaaa-bbbb-cccc-dddd-eeeeeeeeeeee.json?from=now
    jq: .data.tuples[0][1] # parse response json
- name: vzws # combined with websockets
  type: default
  power:
    type: calc
    add:
      - type: ws
        uri: ws://10.0.0.180:8082/socket #Bezug
        jq: .data | select(.uuid=="aaaaaaaa-bbbb-cccc-dddd-eeeeeeeeeeee") .tuples[0][1] # import
        timeout: 30s
        scale: 1
      - type: ws
        uri: ws://10.0.0.180:8082/socket #Einspeisung
        jq: .data | select(.uuid=="aaaaaaaa-bbbb-cccc-dddd-eeeeeeeeeeee") .tuples[0][1] # export
        timeout: 30s
        scale: -1

chargers:
- name: evsewifi
  type: evsewifi # Charger with Phoenix Contact controller
  uri: http://10.0.0.208 # SimpleEVSE-Wifi address
  meter:
     energy: true
     currents: true
     power: true

vehicles:
- name: audi
  type: audi
  title: A3 e-tron
  capacity: 7.5 # kWh
  user: geheim@geheim.com
  password: pw
  vin: WAUZ
  cache: 30m
- name: configurable
  type: default
  title: Default
  capacity: 8 # kWh
  charge:
    type: script # use script
    cmd: cat /tmp/soc #/bin/sh -c "echo 50.1" # actual command
    timeout: 3s # kill script after 3 seconds
  cache: 1m

site:
  title: Home
  meters:
     grid: vzws
     pv: pvse
     #pv:
     #- pvse1
     #- pvse2

loadpoints:
- title: main # name for logging
  #vehicle: configurable
  vehicle: audi
  mode: pv
  soc:
    min: 20 # immediately charge to 0% regardless of mode unless "off" (disabled)
    estimate: true
    target: 100
    levels: # target soc
    - 30
    - 50
    - 95
    - 100
    poll:
      mode: charging
      interval: 180m # intervall when not charging
  onDisconnect:
    mode: now
    targetSoC: 95
  charger: evsewifi # charger
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long, 2m
    threshold: -400 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 2m # threshold must be exceeded for this long, 2m
    threshold: 1100 # maximum import power (W # MIN = 1380-enable(600) = 780)
  guardduration: 20s # switch charger contactor not more often than this (default 10m)
  maxcurrent: 16 # maximum charge current (default 16A)
  phases: 1 # ev phases

</pre>

andig, 2021-03-21:
Könntest Du das bitte formatieren? Github Markdown. Irgendwie kommt mir der Fehler bekannt vor- hatten wir das nicht schonmal?

schenlap, 2021-03-21:
Nein, nicht dass ich wüsste. Wir hatten schon viele andere Problem, aber das, meines Wissens nach, noch nicht. Mittlerweile wird die Energie nicht beim Ladeende zurückgesetzt, ev. merkt sich evcc die Energie beim Ladeende und zählt sie dann hinzu?

andig, 2021-03-21:
Ich kann mit dem Log nichts anfangen da der Ladevorgang nicht erkennbar ist (start..stop) :(

schenlap, 2021-03-23:
Anbei ein Log von heute.
[evcc_20210323.log](https://github.com/andig/evcc/files/6191783/evcc_20210323.log)
und jeweils ein  Screenshots kurz vor und kurz nach dem Beenden des Ladevorgangs:
[image]
[image]

andig, 2021-03-23:
Laut Logfile sieht der Wert nach Stop richtig aus:

    [cache ] TRACE 2021/03/23 18:47:57 chargedEnergy: 380.0000000000008

schenlap, 2021-03-23:
nein, das ist schon der falsche Wert! Anzeige 0.2 kWh ist richtig, 0.4 ist schon der falsche verdoppelte Wert.

schenlap, 2021-03-23:
Das ganze verhält sich auch mit einer längernen Ladung so, dann geht der Wert von z.V 4kWh auf 8 kWh. Das Log ist dann aber schon sehr groß

schenlap, 2021-03-23:
`[cache ] TRACE 2021/03/23 18:47:51 chargedEnergy: 190.00000000000128`
6 sek später habe ich dann off gedrückt
`[cache ] TRACE 2021/03/23 18:47:57 mode: off`
und dann kommt
`[cache ] TRACE 2021/03/23 18:47:57 chargedEnergy: 380.0000000000008`

andig, 2021-03-23:
> nein, das ist schon der falsche Wert! Anzeige 0.2 kWh ist richtig, 0.4 ist schon der falsche verdoppelte Wert.

@schenlap ich kann Dich nur bitten, mal in core/wrapper/chargerater.go ein paar Debugstatements einzubauen. Ich vermute, dass in Deinem Fall eine Doppelzählung passiert, wobei das Endergebnis eigentlich zu passen scheint:

    "meterReading":14.29

bis 

    "meterReading":14.67

das sind 0.4!

andig, 2021-03-23:
> 6 sek später habe ich dann off gedrückt ... und dann kommt

Ja, aber schau Dir bitte mal die API-Werte der EVSE an. Dann weisst Du wo der Fehler liegt ;). Die verdoppelt anscheinend den Verbrauch. Hast Du einen S0 Zähler da drin oder einen richtigen?

schenlap, 2021-03-23:
S0. Ich hatte bisher nur energy angeschaut, das paßt. Die Anzeige bei evse stimmt auch, meterReading aber nicht.
OK, wird dann ein Ticket bei evse

closed 2021-03-23 by schenlap

andig, 2021-03-23:
Ich hab evse schon in Slack benachrichtigt, aber gerne Ticket hier verlinken...