evcc-io/evcc #793: For 2 loadpoints with SEMP only first loadpoint shows in Sunny Portal

profigriller, 2021-03-20:
**Describe the bug**
In case 2load points are configured only the first load point shows in Sunny Portal.
semp.xml is shown below (last 6 digits of ID have been replaxed with XXXXXX )

**To Reproduce**
- build loadpoints as shown below
- both loadpoints show in evcc
- both loadpoints show in /semp url of evcc
- only first loadpoint is shown in Sunny Portal

It is likely that device ID needs to be incremented and that the sub device ID stay at 00

loadpoints:
- title: Garage Outlander # display name for UI
  charger: goalt
  vehicle: OutlanderPHEV
  maxcurrent: 16 # maximum charge current (default 16A)
  phases: 1 # ev phases (default 3)
  mode: pv
  onDisconnect: # set defaults when vehicle disconnects
    #mode: pv # switch back to pv mode
    mode: off # switch back to pv mode
    targetSoC: 90 # charge to 100%
  enable: # pv mode enable behavior
    delay: 2m
    threshold: 3000 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 2m
    threshold: 500 # maximum import power (W)
  guardduration: 10m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)
  soc:
    levels: # target soc levels for UI
    - 40
    - 60
    - 80
    - 90
    - 100
- title: Stellplatz Leaf # display name for UI
  charger: gonew
  vehicle: LeafEV
  maxcurrent: 16 # maximum charge current (default 16A)
  phases: 1 # ev phases (default 3)
  mode: pv
  onDisconnect: # set defaults when vehicle disconnects
    #mode: pv # switch back to pv mode
    mode: off # switch back to pv mode
    targetSoC: 90 # charge to 100%
  enable: # pv mode enable behavior
    delay: 2m
    threshold: 3000 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 2m
    threshold: 500 # maximum import power (W)
  guardduration: 10m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)
  soc:
    levels: # target soc levels for UI
    - 40
    - 60
    - 80
    - 90
    - 100

******* semp.xml *****************
<Device2EM>
<DeviceInfo>
<Identification>
<DeviceId>F-28081973-48af3eXXXXXX-00</DeviceId>
<DeviceName>Garage Outlander</DeviceName>
<DeviceType>EVCharger</DeviceType>
<DeviceSerial>48af3eXXXXXX-0</DeviceSerial>
<DeviceVendor>github.com/andig/evcc</DeviceVendor>
</Identification>
<Characteristics>
<MinPowerConsumption>1380</MinPowerConsumption>
<MaxPowerConsumption>3680</MaxPowerConsumption>
</Characteristics>
<Capabilities>
<CurrentPower>
<Method>Measurement</Method>
</CurrentPower>
<Timestamps>
<AbsoluteTimestamps>false</AbsoluteTimestamps>
</Timestamps>
<Interruptions>
<InterruptionsAllowed>true</InterruptionsAllowed>
</Interruptions>
<Requests>
<OptionalEnergy>true</OptionalEnergy>
</Requests>
</Capabilities>
</DeviceInfo>
<DeviceInfo>
<Identification>
<DeviceId>F-28081973-48af3eXXXXXX-01</DeviceId>
<DeviceName>Stellplatz Leaf</DeviceName>
<DeviceType>EVCharger</DeviceType>
<DeviceSerial>48af3eXXXXXX-1</DeviceSerial>
<DeviceVendor>github.com/andig/evcc</DeviceVendor>
</Identification>
<Characteristics>
<MinPowerConsumption>1380</MinPowerConsumption>
<MaxPowerConsumption>3680</MaxPowerConsumption>
</Characteristics>
<Capabilities>
<CurrentPower>
<Method>Measurement</Method>
</CurrentPower>
<Timestamps>
<AbsoluteTimestamps>false</AbsoluteTimestamps>
</Timestamps>
<Interruptions>
<InterruptionsAllowed>true</InterruptionsAllowed>
</Interruptions>
<Requests>
<OptionalEnergy>true</OptionalEnergy>
</Requests>
</Capabilities>
</DeviceInfo>
<DeviceStatus>
<DeviceId>F-28081973-48af3eXXXXXX-00</DeviceId>
<EMSignalsAccepted>true</EMSignalsAccepted>
<Status>Off</Status>
<PowerConsumption>
<PowerInfo>
<AveragePower>0</AveragePower>
<Timestamp>0</Timestamp>
<AveragingInterval>60</AveragingInterval>
</PowerInfo>
</PowerConsumption>
</DeviceStatus>
<DeviceStatus>
<DeviceId>F-28081973-48af3eXXXXXX-01</DeviceId>
<EMSignalsAccepted>true</EMSignalsAccepted>
<Status>Off</Status>
<PowerConsumption>
<PowerInfo>
<AveragePower>0</AveragePower>
<Timestamp>0</Timestamp>
<AveragingInterval>60</AveragingInterval>
</PowerInfo>
</PowerConsumption>
</DeviceStatus>
</Device2EM>

andig, 2021-03-21:
Bitte formatieren, psoting unleserlich

premultiply, 2021-03-21:
https://github.com/andig/evcc/discussions/786

andig, 2021-03-21:
Dann mache ich hier mal zu. Eine Diskussion reicht.

premultiply, 2021-03-21:
Ist ein Problem/Bug keine Diskussion.

premultiply, 2021-03-25:
@andig Kannst du hier bitte nochmal aufmachen?

andig, 2021-03-25:
Vielleicht möchte der Daumen nach unten ja auch noch seine Meinungsäußerung erläutern?

premultiply, 2021-03-25:
Die letzten beiden Ziffern der DeviceId müssen immer -00 sein. SMA beschreibt das zwar in der SEMP-Doku als SubDev-ID erwartet aber technisch dass diese immer -00 ist.
Zur Unterscheidung muss stattdessen die vorstehende Serial hochgezählt bzw. einmalig sein damit der SHM die Geräte unterscheiden kann.
Axel Müller beschreibt dies auch hier: https://github.com/camueller/SmartApplianceEnabler/blob/master/doc/Appliance_DE.md

Nicht hauen! Ich bin ja auch nur der Bote...
Ich hab das auch so verstanden und hätte es auch so implementiert wie es jetzt ist.

andig, 2021-03-26:
> SubDevices fasst diese also immer als ein Gerät zusammen, insbesondere für die Messwerte und Statistiken. Das ist das Problem hier. Die einzelnen LPs tauchen nicht als eigenständige Geräte auf, sondern werden zusammengefasst.

Ahh. Das Ding stellt keine Subdevices dar? Also hier ändern:

	sempLocalDevice  = "F-28081973-%s-%.02d"

closed 2021-03-29 by andig

DerAndereAndi, 2021-03-30:
@profigriller thanks a lot for that report! You should now be able to use the latest code from the master branch oder the latest docker container build and see both chargers appearing in the SMA Portal.