evcc-io/evcc #794: Charging not starting in pv mode

thierolm, 2021-03-20:
This morning evcc version 0.50 showed strange behavior:
There was enough pv power, car (Zoe) was connected but charging didn't start. 
In Go-E App all the necessary settings were set correct (by evcc) but charging didn't start.
Zoe load mode was set correctly to immediate start. 

In evcc slack chat B. Busche explained the same behaviour:(https://evccgroup.slack.com/archives/C01321PUJAD/p1616244731039200)

Loadpoint enable threshold is set to 2m:
```
  enable: # pv mode enable behavior
    delay: 2m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
``` 
But charging didn't start after 2 minutes in pv mode.
I activated now mode, but starting didn' t started.
Finally I activated charging by using my MY Renault App (Car was shown as connected).

<details><summary>See evcc log output:</summary><br><pre>
Mär 20 10:39:23 tserver systemd[1]: Started evcc.
Mär 20 10:39:27 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:39:27 max charge current: 7
Mär 20 10:39:28 tserver evcc[32218]: [lp-1  ] INFO 2021/03/20 10:39:28 loadpoint 1:
Mär 20 10:39:28 tserver evcc[32218]: [lp-1  ] INFO 2021/03/20 10:39:28   mode:      pv
Mär 20 10:39:28 tserver evcc[32218]: [lp-1  ] INFO 2021/03/20 10:39:28   charger:   power ✓ energy — currents ✓ timer —
Mär 20 10:39:28 tserver evcc[32218]: [lp-1  ] INFO 2021/03/20 10:39:28   meters:    charge ✓
Mär 20 10:39:28 tserver evcc[32218]: [lp-1  ] INFO 2021/03/20 10:39:28     charge:  power ✓ energy — currents ✓
Mär 20 10:39:28 tserver evcc[32218]: [lp-1  ] INFO 2021/03/20 10:39:28   vehicles:  ✓
Mär 20 10:39:28 tserver evcc[32218]: [lp-1  ] INFO 2021/03/20 10:39:28     car 0:   range ✓ finish ✓ status ✓ climate ✓
Mär 20 10:39:28 tserver evcc[32218]: [lp-1  ] INFO 2021/03/20 10:39:28     car 1:   range ✓ finish ✓ status ✓ climate ✓
Mär 20 10:39:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:39:29 charge power: 0W
Mär 20 10:39:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:39:29 charger status: B
Mär 20 10:39:30 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:39:30 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 20 10:39:30 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:39:30 vehicle soc: 60%
Mär 20 10:39:30 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:39:30 vehicle range: 134km
Mär 20 10:39:31 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:39:31 max charge current: 11A = 0A + 11A (-2564W @ 1p)
Mär 20 10:39:31 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:39:31 pv timer reset
Mär 20 10:39:31 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:39:31 pv max charge current: 11A
Mär 20 10:39:31 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:39:31 max charge current: 11
Mär 20 10:39:36 tserver evcc[32218]: 2021/03/20 10:39:36 deprecated: golang.org/x/oauth2: Transport.CancelRequest no longer does anything; use contexts
Mär 20 10:40:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:40:29 charge power: 0W
Mär 20 10:40:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:40:29 charger status: B
Mär 20 10:40:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:40:29 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 20 10:40:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:40:29 max charge current: 11A = 0A + 11A (-2572W @ 1p)
Mär 20 10:40:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:40:29 pv timer reset
Mär 20 10:40:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:40:29 pv max charge current: 11A
Mär 20 10:40:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:40:29 max charge current: 11
Mär 20 10:41:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:41:29 charge power: 0W
Mär 20 10:41:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:41:29 charger status: B
Mär 20 10:41:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:41:29 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 20 10:41:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:41:29 max charge current: 12A = 0A + 12A (-2661W @ 1p)
Mär 20 10:41:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:41:29 pv timer reset
Mär 20 10:41:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:41:29 pv max charge current: 12A
Mär 20 10:41:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:41:29 max charge current: 11
Mär 20 10:42:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:42:29 charge power: 0W
Mär 20 10:42:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:42:29 charger status: B
Mär 20 10:42:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:42:29 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 20 10:42:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:42:29 max charge current: 12A = 0A + 12A (-2711W @ 1p)
Mär 20 10:42:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:42:29 pv timer reset
Mär 20 10:42:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:42:29 pv max charge current: 12A
Mär 20 10:42:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:42:29 max charge current: 11
Mär 20 10:43:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:43:29 charge power: 0W
Mär 20 10:43:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:43:29 charger status: B
Mär 20 10:43:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:43:29 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 20 10:43:30 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:43:30 max charge current: 12A = 0A + 12A (-2719W @ 1p)
Mär 20 10:43:30 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:43:30 pv timer reset
Mär 20 10:43:30 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:43:30 pv max charge current: 12A
Mär 20 10:43:30 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:43:30 max charge current: 11
Mär 20 10:44:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:44:29 charge power: 0W
Mär 20 10:44:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:44:29 charger status: B
Mär 20 10:44:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:44:29 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 20 10:44:29 tserver evcc[32218]: [lp-1  ] DEBUG 2021/03/20 10:44:29 max charge current: 12A = 0A + 12A (-2738W @ 1p)
[... 54 lines cut ...]
Mär 20 10:57:09 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 10:57:09 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 20 10:58:08 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 10:58:08 charge power: 0W
Mär 20 10:58:08 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 10:58:08 charger status: B
Mär 20 10:58:08 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 10:58:08 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 20 10:59:08 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 10:59:08 charge power: 0W
Mär 20 10:59:08 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 10:59:08 charger status: B
Mär 20 10:59:08 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 10:59:08 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 20 11:00:08 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 11:00:08 charge power: 3440W
Mär 20 11:00:08 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 11:00:08 charger status: C
Mär 20 11:00:08 tserver evcc[34435]: [lp-1  ] INFO 2021/03/20 11:00:08 start charging ->
Mär 20 11:00:08 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 11:00:08 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 20 11:00:08 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 11:00:08 vehicle soc: 60%
Mär 20 11:00:08 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 11:00:08 vehicle range: 134km
Mär 20 11:00:08 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 11:00:08 detected phases: 1p [15.2 0 0]A
Mär 20 11:01:08 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 11:01:08 charge power: 3440W
Mär 20 11:01:09 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 11:01:09 charger status: C
Mär 20 11:01:09 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 11:01:09 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 20 11:01:09 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 11:01:09 vehicle soc: 60%
Mär 20 11:01:09 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 11:01:09 vehicle range: 134km
Mär 20 11:01:09 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 11:01:09 detected phases: 1p [15.2 0 0]A
Mär 20 11:02:08 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 11:02:08 charge power: 3440W
Mär 20 11:02:08 tserver evcc[34435]: [lp-1  ] DEBUG 2021/03/20 11:02:08 charger status: C
</pre></details>

A possible reason could also be the fact, that my Zoe was parked outside having 0°C . Is it possible that Zoe internal charger decided to not start automatic charging because of this low temp?

andig, 2021-03-21:
Hast Du ein Trace log? Debug reicht leider nicht :(

thierolm, 2021-03-21:
> Hast Du ein Trace log

Sobald mal wieder die Sonne ausreichend scheint, erstelle ich eins. :-)

andig, 2021-03-21:
Vllt müsste man auch den Timer eher als Integral implementieren denn als unterbrechbaren Countdown. Hier ist die Frage aber einfach warum Dein Timer immer wieder zurück gesetzt wurde. Vmtl. Wechselhafte Bewölkung und Einspeisung nur knapp über Limit- dann reichts halt nie wirklich...

thierolm, 2021-03-21:
Ich habe in meinem fork mal in die loadpoint.go etwas mehr debugging für die enable und disable sequence eingebaut (siehe log unten).
Ich hatte mein minCurrent auf 7A stehen, das scheint für meinen Zoe aber zu wenig zu sein, um einphasig mit dem Laden zu beginnen. Ich habe es jetzt mal auf 9A erhöht und er scheint wie gewollt zu starten (siehe auch log).

Soll ich die etwas mehr beim debugging blubbernde loadpoint.go als PR einstellen?   

<details><summary>See evcc log output:</summary><br><pre>
Mär 21 16:49:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:49:11 charge power: 0W
Mär 21 16:49:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:49:11 charger status: B
Mär 21 16:49:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:49:11 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 21 16:49:12 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:49:12 max charge current: 0.19A = 0A + 0.19A (-43W @ 1p)
Mär 21 16:49:12 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:49:12 enable: start
Mär 21 16:49:12 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:49:12 enable: check targetCurrent 0A >= MinCurrent 9A or (enable threshold 0W != 0W and s>
Mär 21 16:49:12 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:49:12 enable: pv timer reset 2021-03-21 16:49:12 +0100 CET
Mär 21 16:49:12 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:49:12 enable: end 0A
Mär 21 16:49:12 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:49:12 pv max charge current: 0A
Mär 21 16:50:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:50:11 charge power: 0W
Mär 21 16:50:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:50:11 charger status: B
Mär 21 16:50:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:50:11 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 21 16:50:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:50:11 max charge current: 11A = 0A + 11A (-2432W @ 1p)
Mär 21 16:50:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:50:11 enable: start
Mär 21 16:50:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:50:11 enable: check targetCurrent 11A >= MinCurrent 9A or (enable threshold 0W != 0W and >
Mär 21 16:50:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:50:11 enable: site power -2432W < enable threshold 0W
Mär 21 16:50:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:50:11 enable: pv timer remaining: 1m0s
Mär 21 16:50:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:50:11 enable: end 0A
Mär 21 16:50:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:50:11 pv max charge current: 0A
Mär 21 16:51:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:51:11 charge power: 0W
Mär 21 16:51:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:51:11 charger status: B
Mär 21 16:51:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:51:11 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 21 16:51:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:51:11 max charge current: 12A = 0A + 12A (-2795W @ 1p)
Mär 21 16:51:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:51:11 enable: start
Mär 21 16:51:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:51:11 enable: check targetCurrent 12A >= MinCurrent 9A or (enable threshold 0W != 0W and >
Mär 21 16:51:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:51:11 enable: site power -2795W < enable threshold 0W
Mär 21 16:51:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:51:11 enable: pv timer remaining: 1s
Mär 21 16:51:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:51:11 enable: end 0A
Mär 21 16:51:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:51:11 pv max charge current: 0A
Mär 21 16:52:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:52:11 charge power: 0W
Mär 21 16:52:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:52:11 charger status: B
Mär 21 16:52:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:52:11 vehicle status: B (ZOE Experience R135/Z.E. 50)
Mär 21 16:52:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:52:11 max charge current: 12A = 0A + 12A (-2673W @ 1p)
Mär 21 16:52:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:52:11 enable: start
Mär 21 16:52:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:52:11 enable: check targetCurrent 12A >= MinCurrent 9A or (enable threshold 0W != 0W and >
Mär 21 16:52:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:52:11 enable: site power -2673W < enable threshold 0W
Mär 21 16:52:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:52:11 enable: end 9A - pv timer elapsed
Mär 21 16:52:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:52:11 pv max charge current: 9A
Mär 21 16:52:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:52:11 max charge current: 9
Mär 21 16:52:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:52:11 charger enable
Mär 21 16:53:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:53:11 charge power: 1690W
Mär 21 16:53:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:53:11 charger status: C
Mär 21 16:53:11 tserver evcc[71723]: [lp-1  ] INFO 2021/03/21 16:53:11 start charging ->
Mär 21 16:53:12 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:53:12 vehicle status: C (ZOE Experience R135/Z.E. 50)
Mär 21 16:53:12 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:53:12 vehicle soc: 66%
Mär 21 16:53:12 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:53:12 vehicle range: 147km
Mär 21 16:53:13 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 16:53:13 detected phases: 1p [7.8 0 0]A
[... cut ...]
Mär 21 17:01:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 17:01:11 enable: start
Mär 21 17:01:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 17:01:11 enable: check targetCurrent 0A >= MinCurrent 9A or (enable threshold 0W != 0W and s>
Mär 21 17:01:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 17:01:11 enable: pv timer reset 2021-03-21 17:01:12 +0100 CET
Mär 21 17:01:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 17:01:11 enable: end 0A
Mär 21 17:01:11 tserver evcc[71723]: [lp-1  ] DEBUG 2021/03/21 17:01:11 pv max charge current: 0A
<pre><details>

andig, 2021-03-21:
Also hier erstmal erledigt?

thierolm, 2021-03-22:
Jepp alles erledigt.

closed 2021-03-22 by thierolm

gyroslav, 2021-03-28:
Hab das gleiche Verhalten eben erlebt. Timer wird permanent neu gestartet, obwohl genug Leistung über längere Zeit zur Verfügung stand. In der Go-E App wurde der Status "Abgeschlossen" angezeigt, und die LEDs leuchten hell grün (= Standard für Laden fertig). 
EVCC hat aber den korrekten Ladestand von 16% angezeigt. Start aus der Skoda-App heraus klappte auch nicht, weil der Go-E kein Strom lieferte. Lösung war, das Auto ein mal auf und dann wieder zu zuschließen.

premultiply, 2021-03-28:
Den Status "abgeschlossen" gibt es via Typ2 nicht. Was reimt sich die WB denn da zusammen?
Ist da irgendwo ein erreichtes Energielimit in der Box definiert?

gyroslav, 2021-03-29:
Nein, hab ich auch schon kontrolliert. Keine limits oder ziele in der WB-App aktiv.

gyroslav, 2021-03-29:
Heute das selbe Spiel. PV liefert genug Überschuss, Ladebalken im EVCC ist grau. Dann springt er auf Blau und schluss. 
GO-E App zeigt wieder "Fertiggestellt" an. Habe dann Auto wieder geöffnet und geschlossen, dann sprang die Ladung direkt an. 
Was mir dann aufgefallen ist: in EVCC gab es keine Angabe für Restzeit. Nach paar Minuten ging die Farbe wieder auf Blau (SoC 27%) und Go-E hat wieder "Fertiggestellt" angezeigt. 
Habe dann gedrückt STOP -> MIN und dann lief die Ladung wieder an, dieses Mal mit einer Restzeit. 
Diese hat sich aber nicht geändert, sobald die Ladeleistung sich geändert hatte. Egal ob mit 6kW oder 2,7kW geladen wurde, er zählte gemütlich von 3:37 runter. 

Jetzt ist der Ladevorgang wieder bei SoC 65% mit "Fertiggestellt" stehen geblieben. Ich kann 80% oder 100% drücken, passiert nichts. Farbe ich Blau - halt eben "fertig". 

Kann es sein, dass die Version 0.5 ein Problem mit dem Polling und dem "ESTIMATE" hat?
Also dass EVCC denkt, die Ziel-SoC ist bereits erreicht?

andig, 2021-03-29:
Logfile Level Trace?

gyroslav, 2021-03-29:
wenn du mir sagsen kannst, wie, dann gerne! 

so?
```
log: trace
levels:
  core: debug
  lp-1: debug
  lp-2: debug

```