evcc-io/evcc #801: Messaging variable title publishes site and loadpoint titles

thierolm, 2021-03-24:
I extended the default messaging messages in my evcc.yaml to become more meaninful.
In a first step I modified the charging start and finish charging messages, to show my loadpoints in order to see whether my go-e or my fritzdect charger started / finished charging.

Message definitions in evcc.yaml
```
# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: ${title} started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: ${title} finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
```
Checking my today's messages I realized, that the first strat/finish messages are created propperly using my configured loadpoint titles, but after the second loadpoint start charging messages the title variable was overwritten with the site title:

[image]

In my synced evcc repo I found two sources in which  ${title} variable is published:
- loadpoint.go - function Prepare - lp.publish("title", lp.Title) and
- site.go - function DumpConfig - site.publish("title", site.Title)

Couldn't this also lead to some UI confusion?

andig, 2021-03-25:
Wenn Du magst kannst Du das wie bei Slack diskutiert schnell selbst fixen. Muss in Site.vue auch passieren.

thierolm, 2021-03-25:
OK
In der feature/messages oder in master branch?

thierolm, 2021-03-25:
Hm, was du in der feature/messages branch alles schon gemacht hast, ist ziemlich umfangreich. Ich mach in master die Änderung in der site.go und Site.vue :-)

thierolm, 2021-03-25:
Ich hab' die beiden kleinen Änderungen in meinem fork gemacht.
Allerdings scheint er beim make die aktualisierte Site.vue nicht zu berücksichtigen.
In Chrome in der Entwickleransicht zeigt er mir immer noch title , obwohl ich in der Site.vue das auf siteTitle geändert habe.
Muss ich den make Befehl noch irgendwie mit einem Parameter laufen lassen, um die *.vue Änderungen mit einzubeziehen?

[image]

[image]

thierolm, 2021-03-25:
Lasse gerade ein make all laufen, das dauert etwas ...

andig, 2021-03-26:
Für das UI brauchst Du 

    npm run watch

...dann wird im Hintergrund immer gebaut wenn Du was änderst.

thierolm, 2021-03-26:
Hab's vor ein paar Minuten mit einem make assets && make build hinbekommen. Siehe PR.

closed 2021-03-26 by andig