evcc-io/evcc #812: Threshold und Ampere in der gui live ändern

gyroslav, 2021-03-28:
Wunsch:
Threshold, residualPower und max Stromstärke in der GUI live ändern. 

Warum:
Extrem flexibel an Wettersituation und Energiebedarf anpassen ohne yaml edit.

Beispiel:
Ich möchte mein auto nur mit PV laden. Ich habe nur wenige tägliche km zu fahren. Lade das auto also nicht regelmäßig. Eher nur, wenn richtig Sonne scheint und meine PV in den 70% Begrenzer des Netzbetreiber läuft. 
Manchmal fahre ich aber richtig lange strecken. Dann komme ich mit leerer Batterie zurück und brauche für den nächsten Tag nur paar kWh mehr. 

Wenn das Wetter durchwachsen ist, muss ich 
- meinen Rechner starten
- putty an
- login im PI
- yaml edit
- evcc Container restart

Morgen fahre ich dann paar km und es folgen nur sonnige Tage.
Also darf ich morgen mittag wieder
- meinen Rechner starten
- putty an
- login im PI
- yaml edit
- evcc Container restart

Man müsste also die Werte für Stromstärke, enable threshold und residualPower in der GUI live ändern können. 
Sei es als Eingabefeld mit speichern button. Oder per Schieberegler. 
Oder z.B. wie die Ziel-SoC. Mögliche Werte in der yaml definieren, und dann in der GUI nur noch drauf klicken. 
Oder Szenarien in der yaml zusammen stellen.
 Szene:
  -title: Feuer frei
     MaxA: 16
     Enable threshold: 0
     Residualpower: 0
  -title: Grenze
     MaxA: 7
     Enable threshold: 6400
     Residualpower: 500

DerAndereAndi, 2021-03-28:
Du schreibst hier von sehr stark mit dem Hintergrund eine Lösung die du dir vorstellst. Wir möchten dass man so wenig wie nur möglich einstellen/ändern muss und mit der gewählten Konfiguration der Rest automatisch passiert. Ansonsten wird auch die Bedienung viel zu komplex und kompliziert. Die Konfiguration ist jetzt schon kompliziert genug (oder zu kompliziert).

Leider habe ich das Grundproblem noch nicht ganz verstanden. Könntest du es evtl. versuchen nochmal zu Beschreiben ohne mit einer möglichen Lösung gleich in der Beschreibung? Das wäre super!

premultiply, 2021-03-28:
Ich habe das Problem auch nicht verstanden.
Über die Web-GUI kann man doch die unterschiedlichen Lademodi und das jeweilige Limit je nach Bedarf jederzeit per Klick festlegen. Was genau fehlt dir da?

andig, 2021-03-28:
Meinst Du sowas https://github.com/andig/evcc/pull/538?

gyroslav, 2021-03-28:
Ich versuche das zu erklären. 
Meine PV-Anlage hat 9,18 kWp. Ausrichtung 45° nach Süden. An guten Tagen habe ich also von 10 - 16 Uhr theoretisch die volle Leistung. 
Unser Netzbetreiber hier hat die maximale Einspeisung aber bei 70% der Anlagenleistung pauschal abgeriegelt. Wenn also nichts da ist, was Leistung will, regelt der WR hart bei 6,4 KW ab. Es werden also 2,7KW verschenkt. Bei 6h am Tag sind das also 16kWh. 

Mein kranker Kopf weigert sich, diese 16kWh zu vergeuden. Also habe ich die Max-Ampere (6) und "enable threshold" (-6400) so eingerichtet, dass EVCC genau diese 2,7kW ins Auto lädt. Dank Corona steht der Wagen jetzt fast nur noch. 

Ich fahre aber auch mal eine längere Strecke und muss den sehr leeren Akku dann mit jedem bisschen, was die PV Anlage liefert, wieder aufladen. Damit ich am nächsten Tag zumindest meine tägliche Strecke zum Kindergarten hin bekomme. Also muss ich jedes Mal, wenn der Wagen restlos leer ist, die yaml editieren. 
Also jedes Mal wenn sich das Wetter ändert, und der SoC bis zum nächsten sonnigen Tag nicht ausreicht. 

Die Modi aktuell sind STOP, SOFORT, MIN+PV, PV. Ich möchte halt noch einen individuell-modus haben - nennen wir ihn CUSTOM - , wo ich die Stromstärke, die enable threshold und ggf den residualPower ändern kann. 
Für das "normale" Laden wäre "PV". Für den Notfall "SOFORT". Und für meinen geizigen Arsch dann der "CUSTOM".

gyroslav, 2021-03-28:
> 
> 
> Meinst Du sowas #538?

Nein, ich möchte wirklich nur die drei Werte 
 - enable threshold
 - maxcurrent
 - residualPower

per Knopfdurck ändern können

andig, 2021-03-28:
Ja, aber WARUM?

andig, 2021-03-28:
Du beschreibst eine Lösung, Dein Problem wird aber nicht klar. Auch Min+PV nutzt die PV Anlage immer voll aus?

gyroslav, 2021-03-28:
min+PV ist genau das Gegenteil, da nimmt sich EVCC doch erst aus dem Netz und dann erst aus der PV. 
Ich will, dass EVCC nur die letzten 2,7kW aus der PV nimmt. Also eher sowas wie PV+MAX

gyroslav, 2021-03-28:
> 
> 
> Ja, aber WARUM?

ich will nur dann laden, wenn die PV-Anlage in den 70%-Begrenzer läuft. siehe  https://github.com/andig/evcc/issues/812#issuecomment-808921014

andig, 2021-03-28:
> min+PV ist genau das Gegenteil, da nimmt sich EVCC doch erst aus dem Netz und dann erst aus der PV.

Falsch und physikalisch unmöglich.

DerAndereAndi, 2021-03-28:
Mal ne ganz andere Frage: Wo genau wurde die Einspeisebegrenzung denn gesetzt? Was für eine Anlage ist das? Normalerweise wird die maximale Erzeugungsleistung (!!) auf 70% des theoretischen Maximums fest im Wechselrichter gesetzt. Die Alternative ist z.B. der SMA Home Manager der dynamische diese Begrenzung im Wechselrichter verändern kann.

Ich behaupt mal der Netzbetreiber hat da gar nichts gesetzt, sondern der Elektriker der alles installiert hat, hat dem Netzbetreiber ein Bild gemacht wie er die Begrenzung im Wechselrichter gesetzt hat.

Und damit wird es deinen gewünschten Fall nie geben, außer eben du hast eine Anlage die das dynamisch ändert.

Also, wo wurde denn was genau eingestellt?

gyroslav, 2021-03-28:
min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)

> Mal ne ganz andere Frage: Wo genau wurde die Einspeisebegrenzung denn gesetzt? Was für eine Anlage ist das? Normalerweise wird die maximale Erzeugungsleistung (!!) auf 70% des theoretischen Maximums [...]

Im Homemanager dynamisch auf 70% vom Installateur. Westnetz sieht entweder ein Rundsteuerempfänger vor oder eine Begrenzung auf 70%. Der SMA Home Manager regelt das dann dynamisch. Wenn ich also im Begrenzer bin, gibt der WR nur 6.4kW raus. Wenn dann noch ein Verbraucher hinzuschaltet (Backofen mit 2kW), dann regelt der SHM den WR auf 8,4kW hoch.
Mein EXPORT bleibt ebi 6,4, die GESAMT-WIRKLEISTUNG steigt aber von 6,4 auf 8,4.

gyroslav, 2021-03-28:
Mein Problem: ich muss mehrmals pro Woche die evcc.yaml editieren und den docker-container neustarten.
Meine Wunschlösung wäre also ein Knopfdruck auf der GUI.

Ist das hilfreicher? :-)

DerAndereAndi, 2021-03-28:
Ok, danke. Jetzt kommen wir der Sache schon näher. So wie der SMA Home Manager "prognosebasiertes Laden" der Hausbatterie ermöglicht, wäre das hier etwas ähnliches. Auch dieser bewirkt dass 70% weich nicht greift und in einer Hausbatterie zu den Zeiten noch genug "Platz" ist.

Das ist aber jetzt schon extrem spezifisch für eine sehr spezifische Konfiguration. Und dann noch der sehr spezifische Wunsche nur den Teil über dem 70% Wert ins EV zu laden. Uff. Also in diesem Szenario sehe ich keine Chance dass wir so etwas umsetzen. Vor allem musst du ja aktiv dann davor sitzen und das schalten. Das kann ja auch nicht Sinn der Sache sein.

Wir behalten da Szenario im Hinterkopf, ich hab ja selbst den SMA Home Manager 2.0, aber eben ne Hausbatterie wo das "abgefangen" wird.

Sorry.

gyroslav, 2021-03-28:
> 
> 
> > min+PV ist genau das Gegenteil, da nimmt sich EVCC doch erst aus dem Netz und dann erst aus der PV.
> 
> Falsch und physikalisch unmöglich.

    min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
    
lade bis min sofort, also wenn kein PV da, dann auch aus dem Netz.

DerAndereAndi, 2021-03-28:
nicht ganz, sondern: lade MIT der Mindestleistung sofort AUCH/SELBST wenn kein PV da ist ;)

gyroslav, 2021-03-28:
> Ok, danke. Jetzt kommen wir der Sache schon näher. So wie der SMA Home Manager "prognosebasiertes Laden" der Hausbatterie ermöglicht, wäre das hier etwas ähnliches. Auch dieser bewirkt dass 70% [...]

Wenn die beiden Werte enable threshold und maxcurrent in der gui verstellbar wären, würde das ja schon etliche Konstellationen abdecke.

DerAndereAndi, 2021-03-28:
Schon möglich. Aber wir werden die UI so einfach wie nur möglich halten. Man sollte am Besten diese so gut wie nie benutzen müssen.

Du kannst aber gerne eine Fork des Projektes machen und deine eigene Version mit diesen Möglichkeiten bauen.

closed 2021-03-28 by DerAndereAndi

gyroslav, 2021-03-28:
> 
> 
> nicht ganz, sondern: lade MIT der Mindestleistung sofort AUCH/SELBST wenn kein PV da ist ;)

PV+MAX = Lade mit PV bis XX% und danach nur noch wenn MAX-threshold überschritten wurde...

gyroslav, 2021-03-28:
> 
> 
> Schon möglich. Aber wir werden die UI so einfach wie nur möglich halten. Man sollte am Besten diese so gut wie nie benutzen müssen.
> 
> Du kannst aber gerne eine Fork des Projektes machen und deine eigene Version mit diesen Möglichkeiten bauen.

Wenn ich nur programmieren könnte ... Ich bin eher so in der JCL und REXX-Welt unterwegs. 

Trotzdem danke für die Aufmerksamkeit!

premultiply, 2021-03-28:
So, alles was du brauchst ist schon drin.
Du musst in der Konfig allowControl: true beim hems-Eintrag für den SHM setzen.
Dann im Sunny Portal den Regler für den Ladepunkt ganz auf Rechtsanschlag schieben. In EVCC stellst du dann auf Min+PV oder PV (sollte dafür egal sein) und der SHM übersteuert dann EVCC.

gyroslav, 2021-03-28:
> So, alles was du brauchst ist schon drin.
> Du musst in der Konfig allowControl: true beim hems-Eintrag für den SHM setzen.
> Dann im Sunny Portal den Regler für den Ladepunkt ganz auf Rechtsanschlag [...]

Danke für den Hinweis! Werde ich mir anschauen.