evcc-io/evcc #825: SoC-Abfrage Nissan Leaf: "authentication failed" bei neuem Ladevorgang

junibart, 2021-03-31:
**Describe the bug**

Es geht um die SoC-Abfrage bei neueren Nissan Leaf, ab Modelljahr 2019 (MY19). 
Nach dem Start von EVCC (als Service auf einem Raspberry Pi) wird der SoC des Fahrzeugs zunächst erfolgreich bei Nissan abgefragt. Wird das Fahrzeug nach beendeter Ladung abgesteckt, gefahren und wieder angeschlossen, führt eine erneute SoC-Abfrage zu einem "authentication failed" Fehler, der SoC wird nicht aktualisiert. 
Die Nissan-Server sind jedoch verfügbar, ein zeitgleicher Aufruf von "./evcc vehicle" führt zu einem gültigen Ergebnis. 
Im Anhang sind zwei TraceLog-Dateien, einmal das gekürzte Log des als Service laufenden EVCC (evcc.log.err_unauthorized) sowie das Log eines Aufrufs von `./evcc/evcc vehicle --log trace > evcc-command.log`. VIN unf Bearer ID sind anonymisiert. 

Hier eine zeitliche Abfolge mit Verweisen auf die Zeilennummern des evcc.log.err_unauthorized. 

EVCC als Service gestartet: 

Z00001 18:04:47: Start von EVCC, Fahrzeug ist angesteckt. 
Z00234 18:04:53: SoC-Abfrage bei Nissan liefert den korrekten Wert (86%)
Z00386 18:05:48: SoC wird im WebIf auf 100% gesetzt, Ladung beginnt
Z00439 18:15:23: SoC-Abfrage bei Nissan liefert 86%, EVCC weist in der nächsten Zeile 87% aus. 
Z00615 18:25:53: SoC-Abfrage bei Nissan liefert 88%
Z00769 18:35:54: SoC-Abfrage bei Nissan liefert 90%
Z00861: 18:36:25 Ladung wird beendet. 

Z00925 05:55:53 Fahrzeug getrennt und wird weggefahren. 

Z 00995 7:19:31 Fahrzeug kehrt zurück und wird angeschlossen, SoC 85% abgelesen
Z01002  7:19:53: SoC-Abfrage bei Nissan ergibt "authentication failed" (Zeile 1069)

 (keine weitere Abfrage nach 10 bzw. 15 Minuten, um 7:43 wurde auf der Kommandozeile ./evcc vehicle mit gültigem Ergebnis aufgerufen)

Z01139 7:45:27: SoC wird im WebIf auf 100% gesetzt, Ladung beginnt
Z01204  7:45:53: SoC-Abfrage bei Nissan ergibt "authentication failed" (Zeile 1271)

Der Fehler tritt in V0.50 auf, ist aber nicht neu. Er zeigte sich schon bei den Versionen um 0.42, ich hatte leider zwischenzeitlich keine Zeit, mich hier zu beteiligen. Vermutlich ist er auch Ursache des Issue #661. 

**To Reproduce**

Fahrzeug (Nissan Leaf ab Modelljahr 2019) ist angeschlossen, EVCC als starten. (sudo systemctl start evcc)
Webinterface beobachten oder Logfile mit tail -f beobachten. 
Nach beendetem Ladevorgang das Fahrzeug abstecken, (eine Tour fahren) und wieder anschließen. 
Webinterface beobachten oder Logfile mit tail -f beobachten. 

**Expected behavior**

Die SoC-Abfragen sollten jederzeit zu einem gültigen Ergebnis führen, zumindest, wenn die Nissan-Server funktionieren. 

**EVCC details:**
Show output of <code>evcc -v</code>:
```
pi@EVCC:~ $ ./evcc/evcc -v
evcc version 0.50 (57495de)
```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
pi@EVCC:~ $ ./evcc/evcc dump -c evcc.yaml
[lp-1  ] WARN 2021/03/31 09:32:59 poll interval '15m0s' is lower than 1h0m0s and may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
config
======

grid: grid
----------
Power:  0W
Energy: 1793.9kWh

loadpoint 1
===========

charge: charge
--------------
Power:  0W
Energy: 1793.9kWh

charger: simpleevse
-------------------
Charge status: B
Enabled:       false

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
uri: 0.0.0.0:7070 # uri for ui
interval: 30s # control cycle interval

# log settings
log: error
levels:
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
- name: charge
  type: modbus
  model: sdm # SDM630
  uri: 192.168.xxx.xx:26
  rtu: true # rs485 device connected using ethernet adapter
  id: 3
  power: Power # default value, optionally override
  energy: Sum # default value, optionally override
- name: grid
  type: modbus
  model: sdm # SDM630
  uri: 192.168.xxx.xx:26
  rtu: true # rs485 device connected using ethernet adapter
  id: 3
  power: Power # default value, optionally override
  energy: Sum # default value, optionally override

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/andig/evcc-config#chargers
chargers:
- name: openwb
  type: openwb
  broker: 192.168.xxx.xx # openWB IP
  id: 1 # loadpoint id

chargers:
- name: simpleevse
  type: simpleevse
  uri: 192.168.xxx.xx:26
  id: 1

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/andig/evcc-config#vehicles
vehicles:
- name: leaf
  type: nissan
  title: Leaf
  capacity: 40 # kWh
  user: xxxxxxxxxxxxx@gmail.com
  password: xxxxxxxxxx
  region: World
  cache: 10m

# site describes the EVU connection, PV and home battery
site:
  title: LC11 # display name for UI
  meters:
    grid: grid # grid meter
#    pv: pv # pv meter
#    battery: battery # battery meter
#  prioritySoC: 60 # give home battery priority up to this soc (0 to disable)

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
- title: Carport # display name for UI
  charger: simpleevse #openwb # charger
  meters:
    charge: charge # charge meter
  vehicle: leaf
  mode: now
  soc:
    poll:
      mode: charging
      interval: 15m
    min: 40 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 85 # always charge to 100%
    estimate: true # set true to interpolate between api updates
    levels: # target soc levels for UI
    - 60
    - 70
    - 80
    - 90
    - 100
  onDisconnect: # set defaults when vehicle disconnects
    mode: now # switch back to now mode
    targetSoC: 80 # charge to 100%
  phases: 3 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 5m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 19 # maximum charge current (default 16A)

# mqtt message broker
mqtt:
  broker: 192.168.xxx.xxx:1883
  topic: evcc 
  # user:
  # password:

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# push messages
#chargeEstimate socCharge
#bei ${connectedDuration} Prozent gestartet. Erwartete Ladezeit: ${connectedDuration} 
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Ladung im Mode "${mode}" bei ${socCharge}% SoC gestartet. Erwartete Ladezeit ${chargeEstimate}.
    stop: # charge stop event
      title: Charge finished
      msg: Ladung beendet mit ${chargedEnergy:%.1fk}kWh in ${chargeDuration}, SoC von ${socCharge} erreicht.
    connect: # vehicle connect event
      title: Car connected
      msg: "Fahrzeug angeschlossen."
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Fahrzeug getrennt! ${connectedDuration}
  services:
      type: telegram
      token: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
      chats: xxxxxxxxx
  
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>

</pre></details>

<details><summary>Show evcc log output with <code>--log trace</code>:</summary><br><pre>
siehe Dateien
[evcc-command.log](https://github.com/andig/evcc/files/6235225/evcc-command.log)
[evcc.err_unauthorized.log](https://github.com/andig/evcc/files/6235227/evcc.err_unauthorized.log)

</pre></details>

**If using Docker:**
Show output of <code>docker run andig/evcc -v</code>:
```
<- paste here

```

<details><summary>Show evcc log output with <code>docker logs <container id></code>:</summary><br><pre>
<- paste here

</pre></details>

DerAndereAndi, 2021-03-31:
Das Problem scheint folgendes zu sein:

- Um 7:19:53 wird der nächste Request mit dem Access Token vom Vorabend benutzt, der Request endet mit einem Fehler
- Nun wird `refreshToken()` aufgerufen, d.h. mit dem `refreshToken` vom Vorabend wird versucht ein neues Auth Token zu bekommen
- Das liefert KEINEN Fehler, sondern HTTP 200, Erfolg und liefert ein neues `access_token` zurück. D.h. das `refresh_token` war noch gültig.
- Das neue `access_token` wird aber nicht gespeichert und das alte damit überschrieben.
- Der nächste Request verwendet weiter das alte `access_token`

Das Problem liegt hier: https://github.com/andig/evcc/blob/master/internal/vehicle/nissan.go#L256-L257
Im Auth Flow sieht das etwas anders aus: https://github.com/andig/evcc/blob/master/internal/vehicle/nissan.go#L219

@andig Das neue `access_token` wird nur in einer lokalen Variable gespeichert. Zeile 256 löschen und in 256 `..., &v.tokens); err ...` rein?

DerAndereAndi, 2021-03-31:
@junibart falls du dir evcc selbst bauen kannst, probier mal bitte das hier aus: https://github.com/andig/evcc/pull/830

closed 2021-03-31 by andig

junibart, 2021-04-03:
Hi, danke für die Anpassung. Habe mir EVCC gebaut und bei den zwei bisherigen Tests hat es den "authentication failed"-Fehler nicht mehr gegeben. Dankeschön!

Gunnar

andig, 2021-05-13:
@junibart ich würde gerne den Refresh asynchron machen, dafür bräuchte ich Zugangsdaten zum Testen. Falls Du helfen willst gerne per Mail melden (cpuidle@gmx.de).

junibart, 2021-05-13:
Du kannst gerne meine Zugangsdaten haben. Aber ich sehe auch, dass Du einen PR vorbereitet hast. Ich seh mal zu, dass ich den PR heute abend teste. Brauchst Du die Zugangsdaten trotzdem?

andig, 2021-05-13:
Bin versorgt, danke :)