evcc-io/evcc #833: Ford API - SoC is fetched, but not updated

MrJayC, 2021-04-02:
**Describe the bug**
I have a fairly new Kuga PHEV (built Feb 2021), which I added as a vehicle to EVCC. I do get the SoC, but it does not update. If I look in the FordPass-App, I get the same behavior, but in the App I can "pull down" to initiate an update request. If I do this, EVCC will get the updated SoC with the next API call. Looks like the Ford API does cache the vehicle data until an update request is made.

**Expected behavior**
Initiate update of SoC instead of getting the cached SoC.

**Proposed Solution**
I have no experience in Go, so I would appreciate some help. You also did the implementation of the Ford API, so you can check if I might be on a blind alley.

I had a look into the API calls of the FordPass-App. When I "pull down" to get an update, the following API call is made:
1) a **PUT request** is sent to the following URL: https://usapi.cv.ford.com/api/vehicles/v2/%s/status (with VIN as %s); I tried in Postman with the same headers as already implemented in EVCC -> works
2) the JSON response to this PUT request looks as follows:
```
{
    "$id": "1",
    "commandId": "35d74860-992e-4375-9cbb-373a19017411",
    "status": 200,
    "version": "1.0.0"
}
```
3) The commandId is important as it is used to track the request by sending a cyclic GET request to the following URL
https://usapi.cv.ford.com/api/vehicles/v3/%s/statusrefresh/35d74860-992e-4375-9cbb-373a19017411 (notice the command id at the end of the URL; still the same headers as implemented). The App seems to send a request roughly 1-1.5s after the previous one was complete - I guess you should add a timeout, just in case. In my trials, the update was complete after about 10s, but this may vary depending on the vehicle's mobile connection? I'm pasting a full JSON response to give you a full picture of the included data.
Note that this is the response after the update is complete - before that all data is "nulled", but most importantly the status is 552 instead of 200 (the JSON attribute "status", not the request status).
After getting the updated data, you could send the request to v4 of the API, but I guess it's easier to get the information directly from the response to the v3 call as it also includes the battery fill level (SoC).
```
{
    "vehiclestatus": {
        "vin": "xxx",
        "lockStatus": {
            "value": "UNLOCKED",
            "status": "CURRENT",
            "timestamp": "04-02-2021 13:21:53"
        },
        "alarm": {
            "value": "NOTSET",
            "status": "CURRENT",
            "timestamp": "04-02-2021 13:21:53"
        },
        "PrmtAlarmEvent": {
            "value": "Null",
            "status": "CURRENT",
            "timestamp": "04-02-2021 13:21:53"
        },
        "odometer": {
            "value": 69.0,
            "status": "CURRENT",
            "timestamp": "04-02-2021 13:21:53"
        },
        "fuel": {
            "fuelLevel": 49.347984,
            "distanceToEmpty": 324.8,
            "status": "CURRENT",
            "timestamp": "04-02-2021 13:21:53"
        },
        "gps": {
            "latitude": "49.5835650",
            "longitude": "10.5973200",
            "gpsState": "UNSHIFTED",
            "status": "LAST_KNOWN",
            "timestamp": "04-01-2021 14:19:37"
        },
        "remoteStart": {
            "remoteStartDuration": 10,
            "remoteStartTime": 0,
            "status": "CURRENT",
            "timestamp": "04-02-2021 13:21:53"
        },
        "remoteStartStatus": {
            "value": 0,
            "status": "CURRENT",
[... 245 lines cut ...]
        "lifeCycMode": {
            "value": "Normal",
            "status": "CURRENT",
            "timestamp": "04-02-2021 13:21:53"
        },
        "dieselSystemStatus": {
            "exhaustFluidLevel": null,
            "filterSoot": null,
            "ureaRange": null,
            "metricType": {
                "value": "Active",
                "status": "LAST_KNOWN",
                "timestamp": "04-01-2021 13:48:25"
            },
            "filterRegenerationStatus": {
                "value": "DPF_Normal_Operation",
                "status": "CURRENT",
                "timestamp": "04-02-2021 13:21:53"
            }
        }
    },
    "version": "3.0.0",
    "status": 200
}
```

DerAndereAndi, 2021-04-02:
Hi @MrJayC,

did you check the log file to see if the implementation tried the call for requesting an updated SoC? You can set the log level to trace. The current implementation uses a "simple" call to a V4 API, see https://github.com/andig/evcc/blob/master/internal/vehicle/ford.go#L137-L164

To really understand the issue, we'd need more (trace log) data. I am also not sure if we still have a Ford account available that we could use for testing. I didn't have Go know how before I joined and contributed some code, it's not (that) bad at all 😄  It's always better to have someone with an actual car at hand to check and fix some things.

Thanks,
Andreas

andig, 2021-04-02:
@MrJayC kannst du den erwarteten Flow mal einfach auf Deutsch beschreiben? Was kommt nach dem PUT, woher kommt die Command Id, welcher Request muss wann ausgeführt werden?

MrJayC, 2021-04-02:
@DerAndereAndi Genau. EVCC holt sich den SoC vom Server, der Server cached die Daten aber über einen sehr langen Zeitraum (mindestens über die Dauer eines halben Ladevorgangs, evtl. kommt überhaupt nur ein Update wenn das Fahrzeug in Betrieb war). Man muss dem Server sagen, dass er neue Daten vom Auto anfordern soll, was während des Ladevorgangs nötig ist.

@andig 
1) Auth-Token besorgen (schon implementiert)
2) PUT-Request senden, um neue Daten vom Fahrzeug zu holen; der Server antwortet mit einem JSON-Objekt (Beispiel siehe oben), relevant ist dort die Eigenschaft "commandId"
URL: https://usapi.cv.ford.com/api/vehicles/v2/%s/status (mit VIN anstelle von %s)
3) GET-Request senden. Der Server antwortet mit einem JSON-Objekt.
https://usapi.cv.ford.com/api/vehicles/v3/%s/statusrefresh/%s (mit VIN, commandId für %s)
4) JSON-Objekt auswerten, abhängig vom Wert der Eigenschaft "status" im JSON-Objekt wie folgt fortfahren:
a) status = 200 -> SoC zurückgeben
b) status <> 200 (oder =552, müsste man beobachten) -> Schritt 3 wiederholen
c) nach x Versuchen abbrechen, Warnung "SoC kann nicht aktualisiert werden", ggf. den zwischengespeicherten Wert verwenden der heute bereits ermittelt wird

Wenn man es weiter treiben möchte, könnte man das wie folgt tun:
1.1) zuerst nach der heutigen Methode den SoC ermitteln
1.2) zunächst neben batteryFillLevel.value auch batteryFillLevel.timestamp auswerten. Das ist die UTC-Zeit der letzten Aktualisierung (Format: "04-02-2021 17:22:49")
1.3) ist der Wert aktuell?
a) ja -> direkt verwenden
b) nein -> weiter bei (2)
Der Aufwand würde sich m.E. nur lohnen, wenn es Fahrzeuge gibt die sich anders verhalten. Sehen wir erst, wenn die ersten Mach-E-Fahrer kommen.

andig, 2021-04-02:
Hast Du ein Gefühl, wie lange das Update dauert? Lohnt sich warten vs. evcc weiter laufen zu lassen?

andig, 2021-04-02:
Und: willst Du Dich an einem PR versuchen?

MrJayC, 2021-04-02:
Das Update dauert ca. 10s - wäre eleganter das asynchron zu machen, hier fehlt mir aber noch Verständnis für den grundsätzlichen Ablauf. Ich versuche mich an einem quick & dirty "Proof of Concept" (mit warten) und werde am Wochenende mal in den Slack kommen. Wenn es grundsätzlich funktioniert, brauche ich etwas Hilfe von euch um das "schön" zu machen.

andig, 2021-04-03:
Mach mal den Geradeausweg. Die „schöne“ Variante könnte so aussehen:
- Auto gibt ErrMustRetry zurück falls es ein Hintergrundupdate angestossen hat- das muss unter Umgehung des Caches stattfinden
- Wert wird entweder im Hintergrund (go Routine) oder beim nächsten Aufruf (dem Retry) abgeholt
- Der Ladepunkt sorgt dafür, dass er bei ErrMustRetry eine erneute Abfrage startet
- Wenn das Auto dann mal einen neuen Wert hat, sollte es den allerdings wie üblich cachen da sonst alle 10s eine Abfrage erfolgen würde

PR #860 (2021-04-18): Ford-API: Request status update if required, Get Range & Charge State
- if server returns an outdated vehicle status (older than 5 minutes), a refresh is requested (Fixes #833)
- vehicle range added for Ford vehicles
- charge state added for Ford vehicles

Tested with Kuga PHEV, no Explorer PHEV and Mach-E available for testing

```me@Johanness-MBP evcc % ./evcc vehicle                           
[main  ] INFO 2021/04/10 20:25:21 evcc 0.50 (b629d07)
[main  ] INFO 2021/04/10 20:25:21 using config file /Users/me/evcc/evcc.yaml
[ford  ] DEBUG 2021/04/10 20:25:23 Vehicle Status is outdated (age 20 [... cut]
closed 2021-04-18 by andig