evcc-io/evcc #883: Add JQ to MQTT and Shell plugins

andig, 2021-04-17:
Schöne Aufgabe für Einsteiger- das HTTP Plugin zeigt wie das geht ;)

Alestrix, 2021-04-19:
Sehe ich es richtig, dass hier https://github.com/andig/evcc/blob/49450cf07983ce4f4aa4926420b805d9b34f49ec/provider/mqtt/client.go#L125 durch alle listener des selben Topics mit der einmal empfangenen Payload durchiteriert wird und somit keine Probleme entstehen, wenn mehrmals das selbe Topic subscribiert wird? Wenn ja, sollte die jq Implementierung für MQTT schnell gemacht sein, würde mich die Tage mal daran versuchen, dieses mal ordentlicher als beim Script Provider.

andig, 2021-04-19:
Was hat das mit jq zu tun? Wenns mehrere Listener gibt bekommen natürlich alle die Nachricht.

Alestrix, 2021-04-19:
> Was hat das mit jq zu tun? 

Bei der Nutzung von JQ ist es wahrscheinlich, dass mehrere Werte in der selben Nachricht enthalten sind und somit das gleiche Topic mehrmals subscribed werden muss. Hat also nur indirekt was mit einander zu tun.

PR #943 (2021-05-01): Mqtt: add jq parsing
Fix #883, Replace #906

@Alestrix magst Du's mal testen? Ich hab den Ansatz etwas geändert und angehübscht.
closed 2021-05-01 by andig