evcc-io/evcc #908: Timeout für http provider

StefanSchoof, 2021-04-23:
**Is your feature request related to a problem? Please describe.**
Ich versuche gerade einen Charger über http zu steuern. Der Charger braucht relative lange um eine Erfolgsmeldung zurück zu liefern.

**Describe the solution you'd like**
Eine timeout Option in dem http provider.
```yaml
  enable:
    type: http
    uri: http://charger:8080/enable
    method: post
    timeout: 60000
```

**Describe alternatives you've considered**
wget im sctript provider mit timeout

**Additional context**
Ich bekomme folge Meldung:
```
evcc_1        | [lp-1  ] ERROR 2021/04/23 13:34:52 charger enable: Post "http://charger:8080/enable": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
```

andig, 2021-04-23:
Gute Idee. Versuchst Du PR? Was für einen charger hast Du?

StefanSchoof, 2021-04-24:
Ich kann einen PR versuchen, wenn jemand anderes das machen will passt das auch für mich. PR kann eine Weile dauern.
Ich habe eine ID.Charger (die Entscheidung habe ich getroffen bevor ich evcc gefunden habe :( )
Ich versuche gerade prototypische, ob ich das Laden starten und stoppen über den ID.3 machen kann. Mir ist klar, dass das keine optimale Lösung ist, aber vielleicht reicht sie für meine Zwecke.

andig, 2021-04-24:
Yep, das wär ja besser als nichts. Falls jemals eebus kommen sollte werden wir das auch bedienen können.

andig, 2021-04-25:
Ich mach mal solange zu.

PR #931 (2021-04-29): Add timeout for http provider
Adding a timeout parameter to the http provider.

Sadly this PR is putting the default timeout of 10 seconds to more places then before. Any advice how this can be done better?

Resolves #908
closed 2021-04-25 by andig