evcc-io/evcc #909: OCPP charger

swaminirbhay, 2021-04-23:
I want to control a charger, that is using OCPP over WLAN/LAN.
Is there any work already done on this topic?

I still did not buy the device, but it is immediatelly available in the shop of daheimladen.de. They are a start-up, so I guess, they will be cooperative, if anyone needs access to a charger for test purpose.

[DaheimLaden Handbuch V.1 V.2.pdf](https://github.com/andig/evcc/files/6365661/DaheimLaden.Handbuch.V.1.V.2.pdf)

andig, 2021-04-24:
Ìn der Doku kann ich von OCPP nicht finden, auch nichts zu Smart Charging Profile. Wenn Interesse an einer Zusammenarbeit besteht, kann sich der Anbieter gerne hier bei uns melden.

swaminirbhay, 2021-04-24:
OK, ich werde es weiter geben. Die haben sich im Handbuch bisher nicht mal dazu geäußert, wie man den Ladepunkt bei der Central Station anmelden kann. Interessant ist vor allem, dass sie die Wallboxen zu einem erschwinglichen Preis auf Lager haben. 

Das einzige an OpenSource zu dem Thema OCPP, was ich finden konnte, ist SteVe ("SteckdosenVerwaltung") von der Uni Aachen. Aber da fehlt noch eine API, damit man es in ein Lademanagement integrieren kann.

andig, 2021-04-24:
Ich mache mal zu bis es Feedback gibt.  Gerne hier melden wenn Interesse an einer Zusammenarbeit besteht.

closed 2021-04-24 by andig

andig, 2022-01-05:
https://github.com/evcc-io/evcc/pull/1963 nimmt langsam Gestalt an. Wer mag und sich traut kann schonmal einen Blick drauf werfen.