evcc-io/evcc #910: Maximaler Ladeleistung etwas niedriger als verfügbare PV-Leistung

Nick-s77, 2021-04-23:
Im Praxisbetrieb ist mir aufgefallen, dass aufgrund von kleinen Schwankungen meiner Hausverbraucher, d.h. mehrere Wechsel pro Sekunde mt jeweils ca. +/- 20 W, von evcc stets der Ladestrom angepasst werden muss und dadurch jeweils geringe Einspeisungen und aber auch Bezüge vom Stromnetz entstehen, die ja eigentlich vermieden werden sollten. 

Anregung: Wert der Ladestromstärke variabel, z.B. um 1 % bis 10 % niedriger berechnen lassen. Verbleibende Rest-Einspeisung  festzulegen entweder mit einer Variablen, z.B. 1 - 100 W oder mit einem Prozentsatz, z.B.  1 - 10 %.

Konfiguration über einen Parameter in der evcc.yaml-Datei bei den charger-Definitionen und Berechnung des aktuell vrogegebenen Ladestroms entsprechend verringert um den variablen Wert oder Prozentsatz.

Da ich leider überhaupt keine Ahnung von der hier verwendeten Programmiersprache habe und schon froh bin, wenn ich eine laufende evcc.yaml erstellt bekomme, gebe ich diese Anregung mal hier im Forum.

andig, 2021-04-23:
Du kannst residualpower angeben, in Deinem Fall negativ

andig, 2021-04-24:
Ich mach mal zu- denke die Frage ist beantwortet.

closed 2021-04-24 by andig