evcc-io/evcc #926: Make generic/plugin configuration more accessible

andig, 2021-04-26:
Ich könnte mir vorschlagen, die `default` Typen in `plugin` umzubenennen und den jeweils konkreten Plugin `type:` durch `plugin:` zu ersetzen:

    - type: plugin
      status: # charger status A..F
        plugin: mqtt
        topic: some/topic1
      enabled: # charger enabled state (true/false or 0/1)
        plugin: mqtt
        topic: some/topic2
      enable: # set charger enabled state
        plugin: script
        cmd: /bin/sh -c "echo ${enable}"
      maxcurrent: # set charger max current
        plugin: script
        cmd: /bin/sh -c "echo ${maxcurrent}"

Würde das helfen, die Konfigurationen besser zu verstehen?

_Originally posted by @andig in https://github.com/andig/evcc/discussions/914#discussioncomment-659227_

naltatis, 2021-04-26:
Das macht die Konfiguration auf jeden Fall einfacher zu verstehen.

andig, 2021-04-26:
@premultiply würdest Du evtl. einen (Monster)PR fertig machen um alle Configs umzustellen? Das muss einmal komplett durchgetestet werden, dazu würde ich es gerne- for the time being- abwärtskompatibel halten, muss aber schauen ob das geht.

premultiply, 2021-04-26:
Ja, die configs kann ich übernehmen.

bicaluv, 2021-04-26:
Prima Idee.
Ich wäre jedoch für type `custom`. (wurde ja an andere Stelle schon mal vorgeschlagen)
Plug-in ist eine Technik und kein Typ. Der Typ ist eben nicht einer der bekannten, sondern "angepasst" oder "anpassbar" oder "selbst gemacht".
Zumal der Begriff "plugin" oben als Typ vorkommt und bei der Angabe von plugins die genutzt werden sollen.
Jetzt klingt es nach rekursiv, ein Plugin nutzt andere Plug-ins, aber eigentlich ist es ein eigener, neuer, selbst konfigurierter Typ.

andig, 2021-04-26:
Die Dinger hiessen auch mal `generic`. Mir gefiel `plugin` weil damit die offensichtliche Beziehung zu den eingesetzten plugins da ist.

bicaluv, 2021-04-26:
generic = allgemein. Was ist allgemein? Alles und nichts.
custom = selbstgemacht/angepasst
plugin hat halt dann eine doppelte Bedeutung, was es nicht verständlicher macht.
Nur meine Gedanken dazu.

andig, 2021-04-26:
> generic = allgemein. 

generisch- für alle Fälle anwendbar

> custom = selbstgemacht/angepasst

`customizable`? `configurable`? `flexible`?

Der PR ist da und kann asap mit neuem evcc-config getestet werden. Im aktuellen Stand auf die "harte tour", final würde ich die ursprünglichen Bezeichner für eine Übergangszeit wieder erlauben.

Damit @premultiply was ändern kann brauchen wir Konsens zum Naming. `plugin` als key für den Plugintyp habe ich schonmal eingeloggt- das scheint erstmal konsensfähig (@naltatis: würde natürlich wieder einen Sonderfall für die Config UI bedeuten, aber das ist vielleicht erstmal sekundär...).

naltatis, 2021-04-26:
> (@naltatis: würde natürlich wieder einen Sonderfall für die Config UI bedeuten, aber das ist vielleicht erstmal sekundär...).

Ja, würde ich auch nicht als Gegenargument sehen. Die Tatsache, dass die Konfiguration dadurch einfacher verstehbar wird, wird sich ja positiv auf die UI auswirken.

andig, 2021-04-27:
Uns fehlt noch der Konsens zum Naming. Irgendwann wird das Issue stale ;)

bicaluv, 2021-04-27:
Ich würde es halt vermeiden einen Begriff für unterschiedliche Sachen verwenden. 
ist in der evcc.dist.yaml leider auch so.

andig, 2021-04-29:
`misc`, `any`, `multi` und `pluggable` wären weitere Varianten

premultiply, 2021-04-29:
Erster zugehöriger Aufschlag: https://github.com/andig/evcc-config/pull/34
Lässt sich problemlos bei Bedarf noch umbenennen.
Hoffe ich habe alle relevanten Stellen erwischt.

andig, 2021-05-01:
@premultiply hat fast alle Änderungen fertig. Letzter Diskussionspunkt: nennen wir die einzelnen Werte in einem `custom` Device `plugin` oder `source` (Vorschlag von Andi)? Letzteres wäre aus Anwendersicht vielleicht noch klarer (Herkunft der Daten), hätte aber keinen Bezug mehr zu den Plugins?

bicaluv, 2021-05-01:
Source ist eine prima Idee.

andig, 2021-05-04:
Finaler call for comments: `plugin` oder `source`?

DerAndereAndi, 2021-05-04:
Source, ist ja von mir 😁

andig, 2021-05-04:
Voten für `source`

andig, 2021-05-04:
Voten für `plugin`

PR #927 (2021-05-05): Rename default plugin and plugin type key
Fix #926
closed 2021-05-05 by andig