evcc-io/evcc #940: cannot create type 'discovergy': certificate signed by unknown authority

lLarryLaffer, 2021-04-29:
**Describe the bug**
Meter kann nicht angelegt werden. Es kommt folgende Meldung:
[main  ] FATAL 2021/04/29 13:25:06 cannot create meter 'grid': cannot create type 'discovergy': Get "https://api.discovergy.com/public/v1/meters": x509: certificate signed by unknown authority
Benutze Docker auf Synology: andig/evcc:latest

**To Reproduce**
Steps to reproduce the behavior:
meters:
- name: grid
  type: discovergy
  user: my@email.adress
  password: "mypassword"
  meter: 1ESY1111111111

andig, 2021-04-29:
Interesting. I cannot repro this with OSX. Could you check what certificates Discovergy sends? If you want to share your credentials please mail me at cpuidle@gmx.de

andig, 2021-04-29:
Cannot reproduce with given credentials either.

andig, 2021-04-29:
Du müsstest jetzt mal in Deinem Container mit curl debuggen, was da passiert. Bei mir- auch mir Docker- kein Problem.

lLarryLaffer, 2021-04-29:
oh je. ich war so dumm und habe als Volume komplett /etc gemountet und
nicht nur die evcc.yaml 

sorry 

mea culpa 

nun hängts beim Tesla:

while parsing config: yaml: line 56: did not find expected tag URI

ein URI Tag ist nicht in den Example Configs enthalten. Ich vermute
aber, dass es aktuell eh nicht geht, weil Tesla die Authentifizierung
umgestellt hat?!

- name: Tesla
type: tesla
title: Model3 # display name for UI
capacity: 80 # kWh
user: xxxxx # email
password: xxxx # password
vin: 5YJ3E7EB9MF111111 

VG 

Andreas 

 

Links:
------
[1] https://github.com/andig/evcc/issues/940#issuecomment-829214968
[2]
https://github.com/

andig, 2021-04-29:
Ich mach das issue mal zu- Hilfsanfragen bitte in den Discussions.