evcc-io/evcc #962: [Frage] Charger Plugin Enabled state

StefanSchoof, 2021-05-03:
Ich versuche gerade zu verstehen, wann Enabled in einem Charger Plugin true und wann false liefern soll.
Sowohl bei https://github.com/andig/evcc/blob/380cd461b935c5042262b62a79173e28ce8d1fd5/internal/charger/charger.go#L92 als auch in https://github.com/andig/evcc-config#charger-generic wird das nicht genauer erklärt.
Ist das wenn der Charger an ist oder wenn er lädt? Wenn es laden ist, wie unterscheidet es sich dann von Status C? Gibt es irgendwo eine Doku die ich übersehen habe?

premultiply, 2021-05-03:
Dabei geht es um das Feedback zu enable.
Das ist die Ladefreigabe durch evcc an den Controller.
Mit dem Ladestatus hat das nichts zu tun.

StefanSchoof, 2021-05-03:
Das heißt, wenn evc enable sendet aber der Charger noch dabei ist die Ladung zu starten, soll das plugin enable true, Status B liefern? Sobald er am Laden ist Status C, enabled true. Dann kommt ein enable false. Dann liefert der Charger enable, false Status C, bis die Ladung gestoppt ist und dann kommt enable false, Status B?

premultiply, 2021-05-03:
Nein. Das ist die Ladefreigabe.
Wenn evcc enable auf false setzt darf der Controller gar nix mehr machen (keine Ladefreigabe) und muss enabled = false zurückliefern.
Bei enable = true muss er enabled = true zurückgeben und DARF er ein Fahrzeug laden.

Wenn evcc auf Modus Stop steht oder nicht genug Überschuss im PV-Modus vorhanden ist wird enable auf false gesetzt.

StefanSchoof, 2021-05-03:
Okay, danke. Jetzt habe ich es verstanden.

closed 2021-05-03 by StefanSchoof