evcc-io/evcc #966: User interface hickup in current master

thierolm, 2021-05-05:
Ich habe einen build des aktuellen Masters heute installiert und kann für meine beiden Loadpoints den Mode nicht mehr im UI für meine beiden loadpoints gleichzeitig setzten:
[image]
Wenn ich den Nur PV Mode Button beim Tasmo Loadpoint setze, verschwindet er beim go-e und umgekehrt ... Soll ich einen Issue aufmachen? Im Release 0.53 ist das noch kein Problem ...
Den master Build habe ich mit einem `make all` gebaut.

andig, 2021-05-05:
@thierolm was wird im backend aufgerufen? Wir mal bitte Blick in die Browser Konssole bzgl. AJAX Requests

thierolm, 2021-05-05:
> Wir mal bitte Blick in die Browser Konssole bzgl. AJAX Requests

Hatte mich mit Michael im Slack schon abgestimmt und er hatte auch schon die Ursache identifiziert. (Neue bootstrap Version).

PR #967 (2021-05-05): fixes mode-select for multiple loadpoints (fixes #966)
Fix #966 

The button-group in bootstrap 5 required a different markup structure for label+input areas, which would require unique id's. The strange behaviour was a result from the id's not being unique.
I've switched from label+input to a simpler button implementation which is completely sufficient for our use-case.
closed 2021-05-05 by andig