evcc-io/evcc #971: Tesla startet Ladevorgang nach mehreren Unterbrechungen nicht mehr, da Ladesäule als defekt angesehen wird

otiedemann, 2021-05-06:
**Is your feature request related to a problem? Please describe.**
Abgesehen von dem Sleep Issue das ich eben re-opened habe funktioniert evcc soweit super bei mir. Leider startet Tesla den Ladevorgang nach mehreren (ca. 10) Unterbrechungen nicht mehr selbst, da er davon ausgeht, daß die Ladesäule defekt ist. Abhilfe schafft hier nur ein Abstecken und erneutes Anstecken des Ladekabels. Gleichzeitig wird bei jeder Ladeunterbrechung durch evcc eine Push-Nachricht generiert, daß das Laden interrupted wurde.

**Describe the solution you'd like**
Über die Tesla API erst das Laden im Fahrzeug sauber stoppen (wie über die App Starten/Stoppen), erst danach an der Go-E den Ladevorgang unterbrechen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
gelegentlich, wenn der Tesla wie in Version 0.50 10 Minuten nach Unterbrechen der Ladung schläft wacht er nicht automatisch wieder auf, wenn wieder genügend Überschuss vorhanden ist. Der go-E aktiviert zwar die Ladung, der Tesla selbst schläft aber weiter. Hier würde evtl. auch ein API Call zum Aufwecken und Starten der Ladung helfen, wenn die Ladung nach längerer Pause wieder aktiviert wird.

andig, 2021-05-06:
/cc @mark-sch siehst Du das Verhalten auch?

@otiedemann ein Stop API war in der Diskussion, führt aber auch zum Entriegeln des Steckers- ist das gewünscht?

andig, 2021-05-06:
> Abgesehen von dem Sleep Issue das ich eben re-opened habe funktioniert evcc soweit super bei mir. 

Das kannst Du Dir sparen- solange die angefragten Informationen nicht kommen werde ich die Issues kommentarlos schließen. Für Ratespiele habe ich keine Zeit und bitte darauf Rücksicht zu nehmen.

otiedemann, 2021-05-07:
> > Abgesehen von dem Sleep Issue das ich eben re-opened habe funktioniert evcc soweit super bei mir.
> 
> Das kannst Du Dir sparen- solange die angefragten Informationen nicht kommen werde ich die Issues kommentarlos schließen. Für Ratespiele habe ich keine Zeit und bitte darauf Rücksicht zu nehmen.

Sorry - ich hatte natürlich alle Daten zusammengestellt, beim Abschicken aber die Fehlermeldung von Github übersehen, daß die Zeichenanzahl überschritten sei, so daß die Daten zwar in der Preview waren aber nicht im abgeschickten Comment. Ich versuche das heute abend mal zu kürzen bzw. die Logs in Dateianhänge zu packen und dann den Issue nochmal ordentlich mit allen Daten zu submitten. Deine Zeit will ich natürlich nicht unnötig beanspruchen - das Projekt ist wirklich toll und hilfreich, in jedem Fall vielen Dank dafür

otiedemann, 2021-05-07:
> /cc @mark-sch siehst Du das Verhalten auch?
> 
> @otiedemann ein Stop API war in der Diskussion, führt aber auch zum Entriegeln des Steckers- ist das gewünscht?

Wäre in meinem Fall beim Laden zu Hause schon OK (Wallbox ist in der geschlossenen Garage und Kabel führt raus). In der Tesla App gibt es bei aktivem Ladevorgang aber auch 2 separate Buttons - Stop Charging und Unlock Charge Port. Wenn man in der App nur Stop Charging wählt bleibt der Port verriegelt und man muss explizit Unlock Charge Port in einem zweiten Schritt drücken.

naltatis, 2021-05-07:
> gelegentlich, wenn der Tesla wie in Version 0.50 10 Minuten nach Unterbrechen der Ladung schläft wacht er nicht automatisch wieder auf, wenn wieder genügend Überschuss vorhanden ist. Der go-E [...]

Die Vermutung hatte ich auch schon bei mir. Allerdings konnte ich das noch nicht wirklich reproduzieren wann das Fahrzeug nicht wieder aufwacht. Bin mir noch nicht ganz sicher, was hier eine gute Aufwecken-Logik wäre.

naltatis, 2021-05-07:
> Leider startet Tesla den Ladevorgang nach mehreren (ca. 10) Unterbrechungen nicht mehr selbst, da er davon ausgeht, daß die Ladesäule defekt ist. 

Das Problem kenne ich. Hatte ich gerade in den ersten Tagen mit evcc auch viel. Lag aber bei mir vor allem an manuellen Steuerung (Wechsel zwischen Stop, Sofort, PV durch die UI). Wenn evcc im PV Modus ist funktioniert das einwandfrei. Auch bei vielen Start/Stops über längere Zeit. Das Flackerverhalten kannst du per `guardduration` ja einstellen. Steht bei mir auf 15m.
Vielleicht machts auch Sinn die `guardduration` in der UI zu respektieren. Also beim Moduswechsel. Das machts aber leider etwas erklärungsbedürftiger weil der Modus dann ggf. erst Zeitversetzt ausgeführt wird.

premultiply, 2021-05-07:
> Vielleicht machts auch Sinn die `guardduration` in der UI zu respektieren. Also beim Moduswechsel.

Bitte nicht. Ich bin gerade froh wenn das System meinen Anordnungen umgehend folgt.

>Das machts aber leider etwas erklärungsbedürftiger weil der Modus dann ggf. erst Zeitversetzt ausgeführt wird.

Eben. Das auch.

andig, 2021-05-08:
> Das Problem kenne ich.

Dann lasst uns doch mal ein STOP API probieren: erst API aufrufen, dann Charger abschalten. Ich schau mal was geht.

andig, 2021-05-08:
Bitte mal testen

closed 2021-05-09 by andig