evcc-io/evcc #985: If vehicle API fails, request is repeated in every EVCC cycle

MrJayC, 2021-05-09:
**Describe the bug**
If an SoC and/or range update from the vehicle API fails due to a server-side problem, the function returns an error and the cached value is not updated. In the next cycle of EVCC, the vehicle API sends a new request.
So in this case, a request to the manufacturer’s API is sent with every cycle (e.g. 10s) instead of the API poll interval (e.g. 15m).

**Expected behavior**
If the API request fails, loadpoint should wait for the poll interval (e.g. 10m) to elapse before making a new API request. SoC estimator covers in the meantime. Server-side problems usually don‘t disappear within 10s and there is a risk that the „fair use policy“ of the vehicle manufacturer is violated by sending requests every 10s.

**Comment**
Ich hatte diesen Fall, als ich kein SoC-Update bekommen konnte. Die API hat reagiert, aber das Update hat nicht funktioniert (vielleicht hatte das Fahrzeug keine Mobilfunkverbindung). Mit der Ford-App ging auch nichts, d.h. kein EVCC-Problem. Aktuell habe ich das lokal behoben, indem ich in diesem Fall keinen Fehler sondern den letzten Wert zurückgebe - der SoC-Estimator rechnet dann weiter hoch. Wenn ich keinen Denkfehler habe, tritt das bei allen serverseitigen Problemen auf, die zur Folge haben dass das Vehicle-API einen Fehler zurückgibt. Daher fände ich eine „globale“ Lösung über den Loadpoint erstrebenswert, möchte aber gerne zur Diskussion stellen wie das sinnvoll umgesetzt werden kann.
Am elegantesten wäre wohl 2-3x zu probieren und dann zu warten. Denke aber, dass das den Aufwand nicht lohnt.

andig, 2021-05-09:
Bitte vollständige Tickets mit Logfile.

closed 2021-05-09 by andig

andig, 2021-05-10:
Eigentlich nicht. Der Commit ist noch nicht gemerged und das von Dir beschriebene Verhalten existiert trotzdem nciht bei mir. Wir können das ignorieren- wenn Du das Thema noch hast bitte mit Logfile.

MrJayC, 2021-05-13:
Ich habe nochmal getestet - ich hatte den Cache nicht verstanden bzw. nicht bedacht, dass er auch Fehler speichert. Hat dann für mich so ausgesehen, als ob in jedem Zyklus die API angefragt wurde, dabei hat nur der Cache immer wieder den Fehler „refresh failed“ zurückgegeben.

-> alles gut, sorry für die Verwirrung