evcc-io/evcc #996: Update to Wiki page: 1.1 Manual installation - Setup Autostart on Linux

gosmarty, 2021-05-10:
**Is your feature request related to a problem? Please describe.**
If using the described content for evcc.service file I'll get following error on my raspberry after reboot:
```
● evcc.service - evcc
   Loaded: loaded (/etc/systemd/system/evcc.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-05-10 11:50:41 CEST; 47s ago
  Process: 478 ExecStart=/usr/local/bin/evcc --log error (code=exited, status=1/FAILURE)
 Main PID: 478 (code=exited, status=1/FAILURE)

Mai 10 11:50:41 raspberrypi systemd[1]: evcc.service: Service RestartSec=100ms expired, scheduling restart.
Mai 10 11:50:41 raspberrypi systemd[1]: evcc.service: Scheduled restart job, restart counter is at 5.
Mai 10 11:50:41 raspberrypi systemd[1]: Stopped evcc.
Mai 10 11:50:41 raspberrypi systemd[1]: evcc.service: Start request repeated too quickly.
Mai 10 11:50:41 raspberrypi systemd[1]: evcc.service: Failed with result 'exit-code'.
Mai 10 11:50:41 raspberrypi systemd[1]: Failed to start evcc.
```

**Describe the solution you'd like**
In my case it helped to change the line `After=syslog.target` to `After=syslog.target network.target`

andig, 2021-05-10:
That‘s how its documented: https://github.com/andig/evcc#installation

The problem is repeated docs covering same content and getting it wrong :/

premultiply, 2021-05-10:
`After=syslog.target network-online.target` would be even better.

premultiply, 2021-05-10:
My Unit-File looks like this (using BT connection for the NRGkick charger):
```
[Unit]
Description=evcc
Requires=bluetooth.target network-online.target
After=bluetooth.target syslog.target network-online.target

[Service]
WorkingDirectory=/home/pi/evcc
ExecStart=/home/pi/evcc/evcc
Restart=always

[Install]
WantedBy=multi-user.target
```

andig, 2021-05-10:
Ich denke, der Kernpunkt ist, dass die duplizierte Doku im Wiki raus muss.

naltatis, 2021-05-10:
Ja, wir sollten gucken, dass wir die README, das Wiki und evcc-config zu einem Punkt zusammenführen. Ich hab da schon ne Idee :). @DerAndereAndi hatte das Thema auch schon mal angeregt. Bräuchten dafür dann aber sicher ein paar motivierte Freiwillige, die daran mit arbeiten.

premultiply, 2021-05-10:
Da wird schon dran gearbeitet, allerdings in einem anderen Repository.
Das ist vielleicht nicht so günstig.

andig, 2021-05-14:
Gibts nier noch neue Aspekte? Sonst würde ich das Issue mangels weiterem Fortschritt zu machen?

naltatis, 2021-05-14:
FunFact: Bin gestern in genau dieses Problem selbst reingelaufen. 🙈

closed 2021-05-24 by andig

blenni, 2021-06-09:
Ich würde gerne hierzu nochmal etwas fragen, ich habe genau den gleichen Fehler.

Beim ersten Mal hat die Lösung im ersten Beitrag funktioniert und die Änderung im Unit-File dazu geführt, dass ich EVCCstarten konnte. Es sieht jetzt so aus wie hier beschrieben: https://github.com/andig/evcc#installation

Gestern musste ich dann meine Raspberry neustarten, seitdem bekomme ich EVCC nicht mehr als Service zum Laufen, genau der Fehler wie im ersten Beitrag.

Starte ich EVCC über die Konsole mit `evcc - l debug`, läuft es und ich habe keinen Fehler.

Hat dazu noch jemand eine Idee? Danke

andig, 2021-06-09:
Wo liegt deine configdatei und wo liegt evcc?

blenni, 2021-06-09:
evcc in `/usr/local/bin` und die Konfig in `/etc/evcc.yaml`

andig, 2021-06-09:
Sieht gut aus. Zeig bitte auch nochmal Dein Unitfile und den Startbefehlt.

blenni, 2021-06-09:
Unit-File:
```[Unit]
Description=evcc
After=syslog.target network.target
[Service]
ExecStart=/usr/local/bin/evcc --log error
Restart=always
[Install]
WantedBy=multi-user.target
```

Startbefehl:
`sudo systemctl start evcc`

Status auslesen:
`sudo systemctl status evcc`

Rückmeldung:
```
● evcc.service - evcc
   Loaded: loaded (/etc/systemd/system/evcc.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2021-06-09 09:15:22 CEST; 3s ago
  Process: 27811 ExecStart=/usr/local/bin/evcc --log error (code=exited, status=1/FAILURE) Main PID: 27811 (code=exited, status=1/FAILURE)

Jun 09 09:15:22 raspberrypi systemd[1]: evcc.service: Service RestartSec=100ms expired, scJun 09 09:15:22 raspberrypi systemd[1]: evcc.service: Scheduled restart job, restart countJun 09 09:15:22 raspberrypi systemd[1]: Stopped evcc.
Jun 09 09:15:22 raspberrypi systemd[1]: evcc.service: Start request repeated too quickly.
Jun 09 09:15:22 raspberrypi systemd[1]: evcc.service: Failed with result 'exit-code'.
Jun 09 09:15:22 raspberrypi systemd[1]: Failed to start evcc.
```

Vielen Dank für die Unterstützung! :)

hijacker123, 2021-09-01:
Zusätzliche Info, was bei mir geholfen hat:
raspi-config öffnen --> Boot-Optionen --> Wait for network at boot (ENABLE)

Der Autostart entsprechend der Anleitung funktionierte nie, weil das Script gestartet wurde, bevor der Raspi seine IP per DHCP erhalten und konnte somit keine ModBus over TCP Verbindung zum Wechselrichter erstellen.

pablopalec, 2021-11-25:
vielleicht könnt ihr mir auch weiter helfen.
Ich habe evcc erfolgreich installiert und konfiguriert.
Lediglich der autostart nach reboot funktioniert nicht.

Wenn ich nach reboot "sudo systemctl status evcc"  ausführe kommt folgende Meldung: 

```
pi@steuermann:~ $ sudo systemctl status evcc
● evcc.service - evcc
   Loaded: loaded (/etc/systemd/system/evcc.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2021-11-25 13:42:49 GMT; 18s ago
  Process: 505 ExecStart=/usr/local/bin/evcc --log error -c /etc/evcc.yaml (code=exited, status=1/FAILURE)
 Main PID: 505 (code=exited, status=1/FAILURE)

Nov 25 13:42:49 steuermann systemd[1]: evcc.service: Service RestartSec=100ms expired, scheduling restart.
Nov 25 13:42:49 steuermann systemd[1]: evcc.service: Scheduled restart job, restart counter is at 5.
Nov 25 13:42:49 steuermann systemd[1]: Stopped evcc.
Nov 25 13:42:49 steuermann systemd[1]: evcc.service: Start request repeated too quickly.
Nov 25 13:42:49 steuermann systemd[1]: evcc.service: Failed with result 'exit-code'.
Nov 25 13:42:49 steuermann systemd[1]: Failed to start evcc.
```

evcc lässt sich manuell starten:

```
pi@steuermann:~ $ sudo systemctl start evcc
pi@steuermann:~ $ sudo systemctl status evcc
● evcc.service - evcc
   Loaded: loaded (/etc/systemd/system/evcc.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2021-11-25 13:44:44 GMT; 4s ago
 Main PID: 784 (evcc)
    Tasks: 9 (limit: 4915)
   CGroup: /system.slice/evcc.service
           └─784 /usr/local/bin/evcc --log error -c /etc/evcc.yaml

Nov 25 13:44:47 steuermann evcc[784]: [lp-1  ] INFO 2021/11/25 13:44:47 loadpoint 1:
Nov 25 13:44:47 steuermann evcc[784]: [lp-1  ] INFO 2021/11/25 13:44:47   mode:      pv
Nov 25 13:44:47 steuermann evcc[784]: [lp-1  ] INFO 2021/11/25 13:44:47   charger:   power ✓ energy ✓ currents ✓ timer ✗
Nov 25 13:44:47 steuermann evcc[784]: [lp-1  ] INFO 2021/11/25 13:44:47   meters:    charge ✓
Nov 25 13:44:47 steuermann evcc[784]: [lp-1  ] INFO 2021/11/25 13:44:47     charge:  power ✓ energy ✓ currents ✓
Nov 25 13:44:47 steuermann evcc[784]: [lp-1  ] INFO 2021/11/25 13:44:47   vehicles:  ✓
Nov 25 13:44:47 steuermann evcc[784]: [lp-1  ] INFO 2021/11/25 13:44:47     car 0:   range ✓ finish ✓ status ✓ climate ✓
Nov 25 13:44:47 steuermann evcc[784]: [lp-1  ] DEBUG 2021/11/25 13:44:47 charge power: 0W
Nov 25 13:44:47 steuermann evcc[784]: [lp-1  ] DEBUG 2021/11/25 13:44:47 charge currents: [0 0 0]A
Nov 25 13:44:47 steuermann evcc[784]: [lp-1  ] DEBUG 2021/11/25 13:44:47 charger status: A
```

Diese Option habe ich auch gesetzt, leider ohne Änderung:
> raspi-config öffnen --> Boot-Optionen --> Wait for network at boot (ENABLE)

premultiply, 2021-11-25:
Ich hab die Doku hinsichtlich der system-Unit mal gefixt: https://docs.evcc.io/docs/installation/manual/#autostart
Bitte nochmal `evcc.service` dahingehend ändern und nochmal prüfen.

pablopalec, 2021-11-25:
Vielen Dank premultiply, jetzt läuft es!