evcc-io/evcc #1020: vehicle remote charge start: invalid character '<' looking for beginning of value

Chris591, 2021-05-14:
**Describe the bug**
if I move the slider for the SoC I receive an error message on EVCC and I get an error from the Car App that the Car couldn't be loaded.
[lp-1  ] ERROR 2021/05/14 20:50:36 vehicle remote charge start: invalid character '<' looking for beginning of value
this happens on the Mobile and Notebook. Browsers tested Mobile Microsoft Edge and Notebook Safari.
I'm not 100% sure if I tested it then the car wasn't loaded to 100% but I think so.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect the Car
2. move the Slider to a different % value
3. you will see this error : vehicle remote charge stop: invalid character '<' looking for beginning of value

**Expected behavior**
shouldn't be an error and the Car shouldn't be updated if it's already loaded above the value or?

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.54 (36582bf)
```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
config
======

grid: grid
----------
Power:          2346W
Current L1..L3: 0.917A 0.352A 9.16A

loadpoint 1
===========

charger: Einfahrt
-----------------
Power:          0W
Energy:         34.1kWh
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        true
Duration:       26s

loadpoint 2
===========

charger: Garage
---------------
Power:          0W
Energy:         0.0kWh
Current L1..L3: 0A 0A 0A
Charge status:  A
Enabled:        false
Duration:       0s
</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
interval: 10s # control cycle interval
uri: 0.0.0.0:7070 # uri for ui

# log settings
log: error
levels:
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
[... 389 lines cut ...]
[lp-1  ] DEBUG 2021/05/14 21:37:22 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2021/05/14 21:37:22 charger status: B
[lp-2  ] DEBUG 2021/05/14 21:37:32 charge power: 0W
[lp-2  ] DEBUG 2021/05/14 21:37:32 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2021/05/14 21:37:32 charger status: A
[lp-1  ] DEBUG 2021/05/14 21:37:42 charge power: 0W
[lp-1  ] DEBUG 2021/05/14 21:37:42 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2021/05/14 21:37:42 charger status: B
[lp-2  ] DEBUG 2021/05/14 21:37:52 charge power: 0W
[lp-2  ] DEBUG 2021/05/14 21:37:52 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2021/05/14 21:37:52 charger status: A
[lp-1  ] DEBUG 2021/05/14 21:38:02 charge power: 0W
[lp-1  ] DEBUG 2021/05/14 21:38:02 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2021/05/14 21:38:02 charger status: B
[lp-2  ] DEBUG 2021/05/14 21:38:12 charge power: 0W
[lp-2  ] DEBUG 2021/05/14 21:38:12 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2021/05/14 21:38:12 charger status: A
[lp-1  ] DEBUG 2021/05/14 21:38:22 charge power: 0W
[lp-1  ] DEBUG 2021/05/14 21:38:22 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2021/05/14 21:38:22 charger status: B
[lp-2  ] DEBUG 2021/05/14 21:38:32 charge power: 0W
[lp-2  ] DEBUG 2021/05/14 21:38:32 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2021/05/14 21:38:32 charger status: A

</pre></details>

andig, 2021-05-15:
@Chris591 das hat nichts mit dem Slider zu tun, sondern mit Deinem Auto. Kannst Du Dich bitte unter cpuidle@gmx.de oder auch Slack per PM mit Zugangsdaten melden damit ich schauen kann, was da los ist? Anscheinend wird ein Startbefehl geschickt, das Auto antwortet aber nicht wie erwartet.

Chris591, 2021-05-15:
Oh dann ist es wohl eher ein VW-SEAT-Cubra problem. Habe vom Händler die Auskunft das die große Probleme mit dem Service im Augenblick haben. Die App läuft auch nicht rund und gibt öfters Fehler aus das was schief gelaufen ist.
Danke

closed 2021-05-15 by Chris591

andig, 2021-05-15:
Ich vermute ein Problem in evcc. Ist ohne Zugangsdaten aber nicht zu analysieren.

Chris591, 2021-05-15:
Ich bekomme auch über die App von Cubra die Meldung das er das Laden nicht starten könnte da ein Problem aufgetreten ist.
Jetzt meckert er sogar das die 12V Batterie fast leer ist und der Motor bald gestartet wird, wenn ich noch 3 weitere Aktionen mache.
Da stimmt noch so einiges am Auto nicht. Aber da er ja eigentlich voll ist sollte EVCC da ja nichts tun, oder?
Hab ja in der config stehen unter poll – mode : charging
    poll:
      # poll mode defines under which condition the vehicle API is called:
      #   charging: update vehicle ONLY when charging (this is the recommended default)
      #   connected: update vehicle when connected (not only charging), interval defines how often
      #   always: always update vehicle regardless of connection state, interval defines how often
      mode: charging
      # poll interval defines how often the vehicle API may be polled if NOT charging
      interval: 60m
Will nicht nochmal mein Passwort ändern. Musste schon die Accounts 2 * löschen im Auto und auch meinen in der Cloud weil er ihn einfach nicht mehr laden wollte oder nur gesponnen hat.
Hoffe das es wenigstens jetzt soweit läuft das wir ordentlich fahren können. 😉
Danke

andig, 2021-05-15:
> Da stimmt noch so einiges am Auto nicht. Aber da er ja eigentlich voll ist sollte EVCC da ja nichts tun, oder?

Mir gefällt das gar nicht. Ist aber ohne Logfile nicht zu analysieren. Der poll mode ist erstmal gut, das sollte ungefährlich sein. Mir wäre trotzdem wohler, wenn Du mir mal ein vollständiges `--log trace` and cpuidle@gmx.de schicken könntest.

> Ich vermute ein Problem in evcc. Ist ohne Zugangsdaten aber nicht zu analysieren.

Ohne kann ich nur raten und auch nicht weiter helfen. An der Stelle ist das Thema für mich leider auch erledigt. Ohne Logfile Sackgasse.

andig, 2021-05-15:
**UPDATE** ich hab den Fehler gefunden. Bitte schalte evcc ab, in max. 15 Minuten kannst Du die neue Version installieren mit der das Problem nicht mehr auftreten sollte!

Chris591, 2021-05-15:
So waren nochmal weg und nun hab ich einen neuen Fehler im Auto – ACC fehler – ist wohl der Radarsensor der ein Problem hat.

Aber hab dann mal mit dem Regler grade ausprobiert den SoC wert zu setzen und es hat geklappt aber diesmal war er ja auch bei 40%.
Hab jetzt die neue Version denke ich:

evcc version 0.55 (f43c204)
hab auch mal den SoC auf unter geladene % eingestellt und er hat aufgehört und auch wieder angefangen als ich ihn wieder auf 100% gestellt haben. Also nochmal vielen Dank!

Wäre vielleicht super, wenn man beim --log trace die Passwörter und so draußen lässt und einen zusätzlichen Parameter, um dann doch alles zu sehen oder andersrum.

wird es nachher nochmal testen, wenn er wieder voll ist.

Vielen Danke das du drangeblieben bist und den Fehler gefunden hast ohne zusätzliche Log files.