evcc-io/evcc #1053: Wrong MQTT Output on evcc/loadpoints/1/chargePower

MehrCurry, 2021-05-19:
I have to vehicles configured for one loadpoint.

```
loadpoints:
  - title: Wallbox # display name for UI
    charger: wallbox # charger reference
    vehicles:
      - Kona1
      - Kona2
    mode: pv
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # poll mode defines under which condition the vehicle API is called:
        #   charging: update vehicle ONLY when charging (this is the recommended default)
        #   connected: update vehicle when connected (not only charging), interval defines how often
        #   always: always update vehicle regardless of connection state, interval defines how often
        mode: always
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 5m
      min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
      # target: 90 # always charge to 100%
      estimate: false # set true to interpolate between api updates
    onDisconnect: # set defaults when vehicle disconnects
      mode: pv # switch back to pv mode
      targetSoC: 80 # charge to 100%
    maxcurrent: 16 # maximum charge current (default 16A)
    mincurrent: 6
    phases: 3
```
One vehicle is currently charging:

```
Mai 19 14:09:32 evcc evcc[14540]: [site  ] DEBUG 2021/05/19 14:09:32 ----
Mai 19 14:09:32 evcc evcc[14540]: [site  ] DEBUG 2021/05/19 14:09:32 pv power: 2423W
Mai 19 14:09:32 evcc evcc[14540]: [site  ] DEBUG 2021/05/19 14:09:32 grid power: -1105W
Mai 19 14:09:32 evcc evcc[14540]: [site  ] DEBUG 2021/05/19 14:09:32 site power: -1105W
Mai 19 14:09:32 evcc evcc[14540]: [lp-1  ] DEBUG 2021/05/19 14:09:32 charge power: 4140W
Mai 19 14:09:32 evcc evcc[14540]: [lp-1  ] DEBUG 2021/05/19 14:09:32 charger status: C
Mai 19 14:09:32 evcc evcc[14540]: [lp-1  ] DEBUG 2021/05/19 14:09:32 vehicle status: C (Schwarzfuß)
Mai 19 14:09:32 evcc evcc[14540]: [lp-1  ] DEBUG 2021/05/19 14:09:32 next soc poll remaining time: 4m50s
Mai 19 14:09:32 evcc evcc[14540]: [lp-1  ] DEBUG 2021/05/19 14:09:32 vehicle soc: 77%
Mai 19 14:09:32 evcc evcc[14540]: [lp-1  ] DEBUG 2021/05/19 14:09:32 max charge current: 7.6A = 6.0A + 1.6A (-1105W @ 3p)
Mai 19 14:09:32 evcc evcc[14540]: [lp-1  ] DEBUG 2021/05/19 14:09:32 pv timer reset
Mai 19 14:09:32 evcc evcc[14540]: [lp-1  ] DEBUG 2021/05/19 14:09:32 pv max charge current: 7.6A
Mai 19 14:09:32 evcc evcc[14540]: [lp-1  ] DEBUG 2021/05/19 14:09:32 max charge current: 7
```

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.55 (f43c204)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
[main  ] INFO 2021/05/19 14:11:30 evcc 0.55 (f43c204)
[main  ] INFO 2021/05/19 14:11:30 using config file evcc.yaml
[mqtt  ] INFO 2021/05/19 14:11:30 connecting evcc-448296847 at 192.168.187.36:1883
[mqtt  ] DEBUG 2021/05/19 14:11:30 192.168.187.36:1883 connected
[lp-1  ] WARN 2021/05/19 14:11:31 poll mode '{always 5m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] WARN 2021/05/19 14:11:31 poll interval '5m0s' is lower than 1h0m0s and may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
config
======

grid: ace3000
-------------
Power: -1121W

pv: solarlog
------------
Power: 4549W

loadpoint 1
===========

charger: wallbox
----------------
Charge status: C
Enabled:       true
Duration:      15m1s
Diagnostic dump:

vehicle 0
---------
SoC:           78%
Charge status: C
Capacity:      64kWh

vehicle 1
---------
SoC:           69%
Charge status: A
Capacity:      64kWh
</pre></details>

When I am observe MQTT topic `evcc/loadpoints/1/chargePower` the value switches between the real charging value seen in the log and 0.

[image]

I suppose that EVCC is sending the chargePower for both vehicles but without any information to see which vehicle the value belongs to.

andig, 2021-05-19:
Es gibt nur einen einzigen Punkt in evcc wo chargePower geschrieben wird und auch das nur 1x pro Durchlauf. Ich vermute dass Deine Werte im Sekundentakt aus anderen Quellen kommen.
Im Ticket fehlt zudem die Konfiguration, daher nur zu erraten.

MehrCurry, 2021-05-19:
Du hast Recht. Das Problem saß vor dem Bildschirm.

closed 2021-05-19 by MehrCurry