evcc-io/evcc #1066: Getting modbus exceptions after upgrade to 0.55

Mootown68, 2021-05-22:
**Describe the bug**
Hi, im running EVCC Release 0.53 on Docker without any problem.
When updating to latest Release 0.55 with the same configuration i get modbus exceptions (see Log).
Back to 0.53 everything is fine again.

Inverter is Fronius Symo 4.5-3-M

Running volkszaehler/mbmd:latest 
run -a 192.168.2.106:502 -d FRONIUS:1 -d FRONIUS:240

works also without any problem.

Hopefully i provided sufficient information on the issue, if you need more information, please tell me.

Thanks

**EVCC details:**

Show evcc configuration file <code>evcc.yaml</code>:
```
# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
meters:
- name: WR
  type: modbus
  model: sdm
  uri: 192.168.2.106:502
  rtu: false # rs485 device connected using ethernet adapter
  id: 1
  power: Power # default values, optionally override
  
- name: SM
  type: modbus
  model: sdm
  uri: 192.168.2.106:502
  rtu: false # rs485 device connected using ethernet adapter
  id: 240
  power: Power # default values, optionally override
 ```
</pre></details>

**If using Docker:**

Show evcc log output with <code>docker logs <container id></code>:
```
[site  ] ERROR 2021/05/21 10:29:39 updating pv meter: read failed: modbus: exception '2' (illegal data address), function '132'
[site  ] ERROR 2021/05/21 10:29:39 updating grid meter: read failed: modbus: exception '2' (illegal data address), function '132'
```
</pre></details>

andig, 2021-05-22:
Könntest Du bitte `modbus: trace` mal auf 0.53 und 0.55 loggen damit ich sehen kann was sich geändert hat?

andig, 2021-05-22:
@Mootown68 warum `rtu: false`? Die SDM sind RTU Zähler. Hast Du *wirklich* einen RTU<->TCP Konverter oder nur RS485<->Ethernet?

Mootown68, 2021-05-22:
Danke für den Hinweis. Ich habe mir meine Konfiguration daraufhin nochmal angeschaut, das hat wohl bisher nur zufällig funktioniert.
Ich habe jetzt `model: sdm` durch `model: FRONIUS1` ersetzt, jetzt funktioniert es sowohl mit 0.53 als auch mit 0.55

closed 2021-05-22 by Mootown68