evcc-io/evcc #1069: Fronius GEN24 SM TS65-3 config not working

thecem, 2021-05-22:
**Describe the bug**
Fronius GEN24  with SM TS65-3 not working.

**To Reproduce**
Config:
```
meters:
- name: grid
  type: modbus
  model: sunspec
  uri: 192.168.1.132:502
  id: 200
  power: 213:W # sunspec meter
```

**Expected behavior**
Reading the sunspec smart meter values out of the inverter.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
<- paste here

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc log output with <code>--log trace</code>:</summary><br><pre>
<- paste here
[site  ] ERROR 2021/05/22 21:02:18 updating grid meter: sunspec: 213:0:W not found

</pre></details>

**If using Docker:**
Show output of <code>docker run andig/evcc -v</code>:
```
<- paste here

```

<details><summary>Show evcc log output with <code>docker logs <container id></code>:</summary><br><pre>
<- paste here

</pre></details>

thecem, 2021-05-22:
Additional info:

evcc detect:

```
+----------------+---------------------------------------+----------+----------------------------------------------------------------------------------------------------------+
| 192.168.6.132  | PV.fritz.box                          | fronius  | {"IP":"192.168.6.132","Port":80}                                                                         |
+                +                                       +----------+----------------------------------------------------------------------------------------------------------+
|                |                                       | sunspec  | {"IP":"192.168.6.132","Port":502,"ModbusResult":{"SlaveID":1,"Model":1,"Point":"Mn","Value":"Fronius"}}  |
+                +                                       +----------+----------------------------------------------------------------------------------------------------------+
|                |                                       | inverter | {"IP":"192.168.6.132","Port":502,"ModbusResult":{"SlaveID":1,"Model":103,"Point":"W","Value":0}}         |
+                +                                       +----------+----------------------------------------------------------------------------------------------------------+
|                |                                       | strings  | {"IP":"192.168.6.132","Port":502,"ModbusResult":{"SlaveID":1,"Model":160,"Point":"N","Value":2}}         |
+----------------+---------------------------------------+----------+----------------------------------------------------------------------------------------------------------+
```

thecem, 2021-05-22:
```
- name: pv
  type: custom
  power:
    source: calc
    add:
    - source: modbus
      model: sunspec
      uri: pv.fritz.box:502
      id: 1
      value: 160:1:DCW # mpp 1 pv
    - source: modbus
      model: sunspec
      uri: pv.fritz.box:502
      id: 1
      value: 160:2:DCW # mpp 2 pv
```

is working for PV, but for grid i don't get it to work.

andig, 2021-05-22:
Not sure what you‘re asking. Please use the discussions when not reporting an issue or improvement.

closed 2021-05-22 by andig

thecem, 2021-05-22:
I configured the Fronius Smart Meter via Gen24 Inverter. 

and get in the log following: 

[site  ] ERROR 2021/05/22 21:02:18 updating grid meter: sunspec: 213:0:W not found

So maybe you could help?

thecem, 2021-05-22:
The right config for a Fronius GEN24 with a Fronius Smart Meter TS65A-3 (Modbus SM readed via inverter IP):

```
- name: grid
  type: modbus
  model: sunspec
  uri: pv.fritz.box:502
  id: 200
  power: Power
```
NOT:

```
- name: grid
  type: modbus
  model: sunspec
  uri: pv.fritz.box:502
  id: 200
  power:  213:W
```

premultiply, 2021-05-23:
@andig detect sollte dann wohl am besten auch noch die Modbus-Id 200 scannen.

andig, 2021-05-23:
Ist die Standard für Fronius?

thierolm, 2021-05-23:
Aus https://www.fronius.com/~/downloads/Solar%20Energy/Operating%20Instructions/42%2C0410%2C2649.pdf  : 

[image]

andig, 2021-05-23:
Also 200-204 und 240-244? Würdest Du PR machen? Bin heute unterwegs...

thierolm, 2021-05-23:
Kann ich, ich bin allerdings nicht der modbus Spezi und kann es nicht testen ...

Wenn ich die taskMeter Task als Template nehme, könnte ich eine froniusIDs Liste definieren, aber passen dann auch die anderen Parameter (Models/Point) ? Nachdem, was ich im funktionierenden Beispiel von @thecem oben sehe, müsste point auf "Power" filter und models wird gar nicht benötigt, richtig?
 
```go
	taskList.Add(tasks.Task{
		ID:      taskMeter,
		Type:    tasks.Modbus,
		Depends: TaskSunspec,
		Config: map[string]interface{}{
			"ids":    sunspecIDs,
			"models": []int{201, 203},
			"point":  "W",
		},
	})
```

thecem, 2021-05-23:
Meter-ID via GEN24: 200 - 204
Meter-ID via Hybrid: 240 - 244

Für die PV Werte solten nicht die DC Werte genutzt und summiert werden sondern die  AC Wandlerwerte:

Smart Meter Register Map with Integer+SF AC-Meter Model:

[image]

Die PV Werte sind leider ohne den WR Verlust, dass heisst die AC Werte sind ca. 30 Watt unter den DCW Werten (160:1:DCW).

Somit sollte hier das Inverter (ID:1) Register 40084 genutzt werden (scale 0.01). 

Ein Hinweis hierzu wäre evtl. eine gute Idee.

thecem, 2021-05-23:
Ich kann das gerne testen und die Ausgaben bereitstellen. Einfach Bescheid geben.... ;-)

thecem, 2021-05-23:
Bei dem detect gibt er den Wert Sunspec Model 103 raus (GEN24 10 PLUS).

thierolm, 2021-05-23:
@thecem du hattest deine `detect` Ergebnisse oben schon gepostet. Kannst du die mal einfach nehmen und ein Mockup für ein `detect` Ergebnis incl. deines grid Meters erstellen, so wie du es erwarten würdest?

thecem, 2021-05-23:
Kostal Smart Meter:
```

+----------------+---------------------------------------+----------+----------------------------------------------------------------------------------------------------------+
| 192.168.76.70  | KSEM.fritz.box                        | meter    | {"IP":"192.168.76.70","Port":502,"ModbusResult":{"SlaveID":71,"Model":203,"Point":"W","Value":71}}       |
+                +                                       +          +----------------------------------------------------------------------------------------------------------+
|                |                                       |          | {"IP":"192.168.76.70","Port":502,"ModbusResult":{"SlaveID":2,"Model":203,"Point":"W","Value":70}}        |
+                +                                       +          +----------------------------------------------------------------------------------------------------------+
|                |                                       |          | {"IP":"192.168.76.70","Port":502,"ModbusResult":{"SlaveID":3,"Model":203,"Point":"W","Value":71}}        |
+                +                                       +          +----------------------------------------------------------------------------------------------------------+
|                |                                       |          | {"IP":"192.168.76.70","Port":502,"ModbusResult":{"SlaveID":71,"Model":203,"Point":"W","Value":71}}       |
+                +                                       +          +----------------------------------------------------------------------------------------------------------+
|                |                                       |          | {"IP":"192.168.76.70","Port":502,"ModbusResult":{"SlaveID":126,"Model":203,"Point":"W","Value":70}}      |
+                +                                       +          +----------------------------------------------------------------------------------------------------------+
|                |                                       |          | {"IP":"192.168.76.70","Port":502,"ModbusResult":{"SlaveID":1,"Model":203,"Point":"W","Value":70}}        |
+                +                                       +          +----------------------------------------------------------------------------------------------------------+
|                |                                       |          | {"IP":"192.168.76.70","Port":502,"ModbusResult":{"SlaveID":2,"Model":203,"Point":"W","Value":70}}        |
+                +                                       +          +----------------------------------------------------------------------------------------------------------+
|                |                                       |          | {"IP":"192.168.76.70","Port":502,"ModbusResult":{"SlaveID":1,"Model":203,"Point":"W","Value":70}}        |
+                +                                       +          +----------------------------------------------------------------------------------------------------------+
|                |                                       |          | {"IP":"192.168.76.70","Port":502,"ModbusResult":{"SlaveID":3,"Model":203,"Point":"W","Value":70}}        |
+                +                                       +          +----------------------------------------------------------------------------------------------------------+
|                |                                       |          | {"IP":"192.168.76.70","Port":502,"ModbusResult":{"SlaveID":126,"Model":203,"Point":"W","Value":69}}      |
+                +                                       +          +----------------------------------------------------------------------------------------------------------+
|                |                                       |          | {"IP":"192.168.76.70","Port":502,"ModbusResult":{"SlaveID":71,"Model":203,"Point":"W","Value":70}}       |
+                +                                       +----------+----------------------------------------------------------------------------------------------------------+
|                |                                       | sunspec  | {"IP":"192.168.76.70","Port":502,"ModbusResult":{"SlaveID":2,"Model":1,"Point":"Mn","Value":"KOSTAL"}}   |
+                +                                       +----------+----------------------------------------------------------------------------------------------------------+
|                |                                       | meter    | {"IP":"192.168.76.70","Port":502,"ModbusResult":{"SlaveID":71,"Model":203,"Point":"W","Value":70}}       |
+                +                                       +          +----------------------------------------------------------------------------------------------------------+
|                |                                       |          | {"IP":"192.168.76.70","Port":502,"ModbusResult":{"SlaveID":126,"Model":203,"Point":"W","Value":70}}      |
+                +                                       +          +----------------------------------------------------------------------------------------------------------+
|                |                                       |          | {"IP":"192.168.76.70","Port":502,"ModbusResult":{"SlaveID":3,"Model":203,"Point":"W","Value":71}}        |
[... cut ...]
+                +                                       +----------+----------------------------------------------------------------------------------------------------------+
|                |                                       | strings  | {"IP":"192.168.76.132","Port":502,"ModbusResult":{"SlaveID":1,"Model":160,"Point":"N","Value":2}}        |
+----------------+---------------------------------------+----------+----------------------------------------------------------------------------------------------------------+
```

Folgend der letzten Fronius Spec:

Smart_Meter_Register_Map_Int&SF.xlsx
und
Gen24_Primo_Symo_Inverter_Register_Map_Int&SF.xlsx

premultiply, 2021-05-23:
Jetzt nicht alles durcheinander werfen hier.
`detect` muss nur zusätzlich um die SunSpec-Suche auf Modbus-ID 200 erweitert werden. Diese Änderung sollte trivial sein.
Die Models sind alle schon drin und es werden auch die relevanten Parameter abgefragt.
Im allergrößten Notfall könnte man noch 211 und 213 (und 801/802) ergänzen, dann sind wir aber wirklich durch.

thierolm, 2021-05-23:
Lass uns bitte im PR https://github.com/andig/evcc/pull/1073 weiter kommunizieren.

thierolm, 2021-05-23:
@thecem kannst du den PR oben testen?

thecem, 2021-05-23:
Mach ich gerne, allerdings habe ich nur eine HA Addon installation. Wenn Du mir kurz beschreibst wie ich die Änderungen in evcc/detect/definitions.go im Addon bekomme?

Auf die shell komme ich..

andig, 2021-05-23:
Wie aktualisiert sich denn generell aktuell das HA addon? Da hab ich den Überblick verloren :(

thierolm, 2021-05-23:
Helft mir kurz, was bedeutet HA addon? :-)
Btw, meine Lady wird morgen 50 ... bin erst ab Mittwoch wieder am Start ... ;-)

thecem, 2021-05-24:
HA: Home Assistant Addon
@andig : Ich kann es nicht genau sagen aber da die addon config latest enthält wird es immer mit der letzten Version im HA als Addon erstellt. Was ich nicht sagen kann ist wenn diese nun von 055 auf 056 springt passiert. Mal sehen ob dann ein update im HA des evcc addon angeboten wird. Ich denke hier wird die commit id verglichen sobald du eine "neues" latest im git erstellst.

andig, 2021-05-24:
Laut der Diskussion bei HA war das wohl nicht so- falls doch wäre das natürlich Klasse!

Caibuk, 2021-05-24:
Nein, Updates werden mit dem HA-Add-on nicht im Supervisor angezeigt. Man muss, um die neueste Version von evcc zu ziehen, das Add-on deinstallieren und gleich wieder installieren.
Da ich durch github-Benachrichtigungen und Slack weiß, wann ich updaten will, ist das für mich kein großes Problem. Der Vorteil der Integration in die HA-Oberfläche und die Einbindung in das Backup-Management ist deutlich größer.

thecem, 2021-05-24:
Sehe ich auch so, dass es kein Problem ist. Evtl. kann der eine oder Andere hier aushelfen wie das versioning gemacht wird.

thecem, 2021-06-06:
Anbei mal die Ausgabe mit .56

```
/app # evcc detect

Auto detection will now start to scan the network for available devices.
Scanning focuses on devices that are commonly used that are detectable with reasonable efforts.
On successful detection, suggestions for EVCC configuration can be made. The suggestions should simplify
configuring EVCC but are probably not sufficient for fully automatic configuration.

+----------------+---------------------------------------+-------------+------------------------------------------------------------------------------------------------------------+
|       IP       |               HOSTNAME                |    TASK     |                                                  DETAILS                                                   |
+----------------+---------------------------------------+-------------+------------------------------------------------------------------------------------------------------------+
| 192.168.2.132  | FG2410AK3L.fritz.box                  | fronius-web | {"IP":"192.168.2.132","Port":80}                                                                           |
+                +                                       +-------------+------------------------------------------------------------------------------------------------------------+
|                |                                       | inverter    | {"IP":"192.168.2.132","Port":502,"ModbusResult":{"SlaveID":1,"Model":103,"Point":"W","Value":5234}}        |
+                +                                       +-------------+------------------------------------------------------------------------------------------------------------+
|                |                                       | strings     | {"IP":"192.168.2.132","Port":502,"ModbusResult":{"SlaveID":1,"Model":160,"Point":"N","Value":2}}           |
+                +                                       +-------------+------------------------------------------------------------------------------------------------------------+
|                |                                       | inverter    | {"IP":"192.168.2.132","Port":502,"ModbusResult":{"SlaveID":1,"Model":103,"Point":"W","Value":5200}}        |
+                +                                       +-------------+------------------------------------------------------------------------------------------------------------+
|                |                                       | strings     | {"IP":"192.168.2.132","Port":502,"ModbusResult":{"SlaveID":1,"Model":160,"Point":"N","Value":2}}           |
+                +                                       +-------------+------------------------------------------------------------------------------------------------------------+
|                |                                       | sunspec     | {"IP":"192.168.2.132","Port":502,"ModbusResult":{"SlaveID":200,"Model":1,"Point":"Mn","Value":"Fronius"}}  |
+                +                                       +             +------------------------------------------------------------------------------------------------------------+
|                |                                       |             | {"IP":"192.168.2.132","Port":502,"ModbusResult":{"SlaveID":1,"Model":1,"Point":"Mn","Value":"Fronius"}}    |
+                +                                       +-------------+------------------------------------------------------------------------------------------------------------+
|                |                                       | meter       | {"IP":"192.168.2.132","Port":502,"ModbusResult":{"SlaveID":200,"Model":203,"Point":"W","Value":23030}}     |
+                +                                       +             +------------------------------------------------------------------------------------------------------------+
|                |                                       |             | {"IP":"192.168.2.132","Port":502,"ModbusResult":{"SlaveID":200,"Model":203,"Point":"W","Value":22960}}     |
+----------------+---------------------------------------+-------------+------------------------------------------------------------------------------------------------------------+
| 192.168.2.70   | KSEM-73016247.fritz.box               | meter       | {"IP":"192.168.2.70","Port":502,"ModbusResult":{"SlaveID":203,"Model":203,"Point":"W","Value":24}}         |
+                +                                       +             +------------------------------------------------------------------------------------------------------------+
|                |                                       |             | {"IP":"192.168.2.70","Port":502,"ModbusResult":{"SlaveID":200,"Model":203,"Point":"W","Value":23}}         |
+                +                                       +             +------------------------------------------------------------------------------------------------------------+
|                |                                       |             | {"IP":"192.168.2.70","Port":502,"ModbusResult":{"SlaveID":201,"Model":203,"Point":"W","Value":23}}         |
+                +                                       +             +------------------------------------------------------------------------------------------------------------+
[... cut ...]
|                |                                       |             | {"IP":"192.168.2.70","Port":502,"ModbusResult":{"SlaveID":1,"Model":203,"Point":"W","Value":23}}           |
+----------------+---------------------------------------+-------------+------------------------------------------------------------------------------------------------------------+

Please open https://github.com/andig/evcc/issues/new in your browser and copy the
results above into a new issue. Please tell us:

        1. Is the scan result correct?
        2. If not correct: please describe your hardware setup.
/app #
```

thecem, 2021-06-06:
Folgende konfig funktioniert nun auch (getestet!):

```
- name: grid
  type: modbus
  model: sunspec
  uri: pv.fritz.box:502
  id: 200
  power:  203:W
```

andig, 2021-06-06:
Super. Fehlt uns da in evcc-config noch ein Beispiel?

/cc @premultiply

thecem, 2021-06-10:
Kann ich Dir nicht sagen ob es fehlt, aber in der config doku macht es sich bestimmt gut! :)
THX