evcc-io/evcc #1082: HASS integration (preferably through HACS)

stepir, 2021-05-25:
I know that this has been already discussed and there's a HASS add-on ( https://github.com/andig/evcc-hassio-addon/blob/main/DOCS.md ) - but rather than trying running EVCC inside HASS, why not looking to have an integration being able to control an external instance of EVCC? pretty much being able to set EVCC parameter from HASS entities rather than using EVCC web interface.

**Is your feature request related to a problem? Please describe.**
Main reason of the request is that, while I simply love using EVCC for PV charging - overnight I would prefer a manual setup based on certain conditions, like how much PV battery I've left, how much house loads are enabled and so on. Things that would be reasonably easy to control and script/automate in HASS.

**Describe the solution you'd like**
The minimal requirement would be being able to disable EVCC control of my charger (so that I can easily control it from HASS where it's already integrated). A nice add-on would being able to have entities switch for controlling the web parameters of EVCC (basically conditional charging and level of charging).

**Describe alternatives you've considered**
As I run EVCC in a container and it's currently integrate with HASS simply with a webpage card ( https://www.home-assistant.io/lovelace/iframe/ ), I thought about using a script to stop and start the EVCC container and configure it as an entity switch in HASS.

[image]

stepir, 2021-05-25:
forgot to mention in the alternatives:

while I can build a ssh script to shutdown the container running evcc the problem is that also my HASS installation run in a container - anytime I'm refreshing the images of either EVCC or HASS I would have to regenerate the ssh keys needed to automate a password less process.

andig, 2021-05-25:
I have no idea about hassio addons, but evcc has a documented api. Feel free to build your own!

Caibuk, 2021-05-25:
Why not using some REST sensors, input_selects, automations and binary sensors?
Then you have everything at your hands to control evcc using automations or the GUI. 

My config as an example:

_sensors_
```
- platform: rest
  name: evcc_state
  resource: http://10.0.1.97:7070/api/state
  json_attributes:
    - mode
    - socTitle
    - socCharge
    - range
    - connected
    - enabled
    - charging
    - climater
  json_attributes_path: loadpoints.0
  value_template: "{{ value_json.loadpoints[0].socCharge }}"
  device_class: battery
  unit_of_measurement: "%"
- platform: template
  sensors:
    evcc_soc_connected:
      friendly_name_template: '{{ states.sensor.evcc_state.attributes["socTitle"] }}'
      value_template: '{{ states.sensor.evcc_state.attributes["socCharge"] }}'
      device_class: battery
      unit_of_measurement: "%"
    evcc_mode:
      value_template: '{{ states.sensor.evcc_state.attributes["mode"] }}'
```

_Input_select_
[image]

_automations_
```
- id: '1593765135350'
  alias: 'EV: EVCC set mode'
  description: Bei Änderung des input_selects den mode an evcc senden.
  trigger:
  - event_data:
      entity_id: input_select.evcc_mode
    event_type: state_changed
    platform: event
  condition: []
  action:
  - data: {}
    service: rest_command.evcc_set_mode
- id: '1593770342483'
  alias: 'EV: EVCC get mode'
  description: Wenn der charge mode in evcc geändert wird, wird der input_select aktualisiert.
  trigger:
  - entity_id: sensor.evcc_mode
    platform: state
  condition: []
  action:
  - data_template:
      entity_id: input_select.evcc_mode
      option: '{{ trigger.to_state.state }}'
    service: input_select.select_option
```

_Binary sensors (just for info)_ 
```
- platform: template
  sensors:
    evcc_connected:
      value_template: '{{ states.sensor.evcc_state.attributes["connected"] }}'
      device_class: plug
    evcc_charging:
      value_template: '{{ states.sensor.evcc_state.attributes["charging"] }}'
      device_class: battery_charging

```

stepir, 2021-05-25:
thank you @Caibuk !!!! while looking at it I realised I would need a way to disable EVCC control of my charger so that I could manually set using its own HA integration. I don't think this is currently possible right @andig ?

closed 2021-05-26 by andig