evcc-io/evcc #1092: tplink "Cahrged: unexpected end of JSON input"

xpcone, 2021-05-28:
Hi,

i got an issue with the tplink chager, that it will most of the time won't report charged kWh, due to not complete JSON response from HS110.
For me it seems like a timeout issue.

`pi@evcc:~ $ sudo evcc -l trace charger
[main  ] INFO 2021/05/28 11:16:46 evcc 0.55 (f43c204)
[main  ] INFO 2021/05/28 11:16:46 using config file /etc/evcc.yaml
[tplink] TRACE 2021/05/28 11:16:46 recv: {"emeter":{"get_realtime":{"voltage_mv":227702,"current_ma":9002,"power_mw":2023560,"total_wh":31449,"err_code":0}}}
[tplink] TRACE 2021/05/28 11:16:46 recv: {"emeter":{"get_realtime":{"voltage_mv":227702,"current_ma":9002,"power_mw":2023560,"total_wh":31449,"err_code":0}}}
[tplink] TRACE 2021/05/28 11:16:46 recv: {"system":{"get_sysinfo":{"sw_ver":"1.5.6 Build 191125 Rel.083657","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS110(EU)","mac":"98:DA:C4:4F:D9:4B","dev_name":"Smart Wi-Fi Plug With Energy Monitoring","alias":"Ladedose","relay_state":1,"on_time":1926,"active_mode":"none","feature":"TIM:ENE","updating":0,"icon_hash":"","rssi":-66,"led_off":0,"longitude_i":73021,"latitude_i":501744,"hwId":"044A516EE63C875F9458DA25C2CCC5A0","fwId":"00000000000000000000000000000000","deviceId":"80065FC78B6728626C07587397F2FA1E1B9E5C8B","oemId":"1998A14DAA86E4E001FD7CAF42868B5E","next_action":{"type":-1},"err_code":0}}}
[tplink] TRACE 2021/05/28 11:16:46 recv: {"emeter":{"get_daystat":{"day_list":[{"year":2021,"month":5,"day":16,"energy_wh":20238},{"year":2021,"month":5,"day":17,"energy_wh":2360},{"year":2021,"month":5,"day":18,"energy_wh":11},{"year":2021,"month":5,"day":19,"energy_wh":0},{"year":2021,"month":5,"day":20,"energy_wh":0},{"year":2021,"month":5,"day":21,"energy_wh":0},{"year":2021,"month":5,"day":22,"energy_wh":0},{"year":2021,"month":5,"day":23,"energy_wh":8405},{"year":2021,"month":5,"day":24,"energy_wh":9175},{"year":2021,"month":5,"day":25,"energy_wh":0},{"year":2021,"month":5,"day":26,"energy_wh":840},{"year":2021,"month":5,"day":27,"energy_wh":0},{"year":2021,"month":5,"day":1,"energy_wh":0},{"year":2021,"month":5,"day":2,"energy_wh":0},{"year":2021,"month":5,"day":3,"energy_wh":0},{"year":2021,"month":5,"day":4,"energy_wh":0},{"year":2021,"month":5,"day":5,"energy_wh":0},{"year":2021,"month":5,"day":6,"energy_wh":0},{"year":2021,"month":5,"day":7,"energy_wh":15447},{"year":2021,"month":5,"day":8,"energy_wh":17338},{"year":2021,"month":5,"day":9,"energy_wh":16878},{"year":2021,"month":5,"day":10,"energy_wh":16351},{"year":2021,"month":5,"day":11,"energy_wh":0},{"year":2021,"month":5,"day":12,"energy_wh":0},{"year":2021,"month":5,"day":13,"energy_wh":0},{"year":2021,"month":5,"day":14,"energy_wh":0},{"year":2021,"month":5,"day":15,"energy_wh":0},{"year":2021,"month":5,"day":28,"energy_wh":1434}],"err_code":0}}}
Power:         2024W
Charge status: C
Enabled:       true
Charged:       1.4kWh
`
`
pi@evcc:~ $ sudo evcc -l trace charger
[main  ] INFO 2021/05/28 11:17:14 evcc 0.55 (f43c204)
[main  ] INFO 2021/05/28 11:17:14 using config file /etc/evcc.yaml
[tplink] TRACE 2021/05/28 11:17:14 recv: {"emeter":{"get_realtime":{"voltage_mv":227768,"current_ma":9000,"power_mw":2024086,"total_wh":31466,"err_code":0}}}
[tplink] TRACE 2021/05/28 11:17:15 recv: {"emeter":{"get_realtime":{"voltage_mv":227768,"current_ma":9000,"power_mw":2024086,"total_wh":31466,"err_code":0}}}
[tplink] TRACE 2021/05/28 11:17:15 recv: {"system":{"get_sysinfo":{"sw_ver":"1.5.6 Build 191125 Rel.083657","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS110(EU)","mac":"98:DA:C4:4F:D9:4B","dev_name":"Smart Wi-Fi Plug With Energy Monitoring","alias":"Ladedose","relay_state":1,"on_time":1954,"active_mode":"none","feature":"TIM:ENE","updating":0,"icon_hash":"","rssi":-66,"led_off":0,"longitude_i":73021,"latitude_i":501744,"hwId":"044A516EE63C875F9458DA25C2CCC5A0","fwId":"00000000000000000000000000000000","deviceId":"80065FC78B6728626C07587397F2FA1E1B9E5C8B","oemId":"1998A14DAA86E4E001FD7CAF42868B5E","next_action":{"type":-1},"err_code":0}}}
[tplink] TRACE 2021/05/28 11:17:15 recv: {"emeter":{"get_daystat":{"day_list":[{"year":2021,"month":5,"day":16,"energy_wh":20238},{"year":2021,"month":5,"day":17,"energy_wh":2360},{"year":2021,"month":5,"day":18,"energy_wh":11},{"year":2021,"month":5,"day":19,"energy_wh":0},{"year":2021,"month":5,"day":20,"energy_wh":0},{"year":2021,"month":5,"day":21,"energy_wh":0},{"year":2021,"month":5,"day":22,"energy_wh":0},{"year":2021,"month":5,"day":23,"energy_wh":8405},{"year":2021,"month":5,"day":24,"energy_wh":9175},{"year":2021,"month":5,"day":25,"energy_wh":0},{"year":2021,"month":5,"day":26,"energy_wh":840},{"year":2021,"month":5,"day":27,"energy_wh":0},{"year":2021,"month":5,"day":1,"energy_wh":0},{"year":2021,"month":5,"day":2,"energy_wh":0},{"year":2021,"month":5,"day":3,"energy_wh":0},{"year":2021,"month":5,"day":4,"energy_wh":0},{"year":2021,"month":5,"day":5,"energy_wh":0},{"year":2021,"month":5,"day":6,"energy_wh":0},{"year":2021,"month":5,"day":7,"energy_wh":15447},{"year":2021,"month":5,"day":8,"energy_wh":17338},{"year":2021,"month":5,"da
Power:         2024W
Charge status: C
Enabled:       true
Charged:       unexpected end of JSON input
`

andig, 2021-05-28:
Looks like we're not reading enough data in https://github.com/andig/evcc/blob/master/charger/tplink.go#L186. Could you try 4k instead of 2k?

/cc @thierolm

thierolm, 2021-05-28:
I tried 4k on Joachims pi and it's working:
```sh
pi@raspberrypi:~/git/evcc $ sudo ./evcc charger -l trace
[main  ] INFO 2021/05/28 14:07:39 evcc 0.55 (b437433)
[main  ] INFO 2021/05/28 14:07:39 using config file /etc/evcc.yaml
hs110
-----
[tplink] TRACE 2021/05/28 14:07:40 recv: {"emeter":{"get_realtime":{"current":0.072928,"voltage":237.572770,"power":0.538312,"total":0.181000,"err_code":0}}}
[tplink] TRACE 2021/05/28 14:07:40 recv: {"emeter":{"get_realtime":{"current":0.072928,"voltage":237.572770,"power":0.538312,"total":0.181000,"err_code":0}}}
[tplink] TRACE 2021/05/28 14:07:40 recv: {"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.2.6 Build 200727 Rel.120821","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS110(EU)","mac":"50:C7:BF:42:60:9B","deviceId":"80068B6B73AAD8C4A4D0F7B9AB5F8B1B1838EEAC","hwId":"45E29DA8382494D2E82688B52A0B2EB5","fwId":"00000000000000000000000000000000","oemId":"3D341ECE302C0642C99E31CE2430544B","alias":"Heizstab","dev_name":"Wi-Fi Smart Plug With Energy Monitoring","icon_hash":"","relay_state":1,"on_time":8826,"active_mode":"none","feature":"TIM:ENE","updating":0,"rssi":-45,"led_off":0,"latitude":51.054749,"longitude":13.660813}}}
[tplink] TRACE 2021/05/28 14:07:40 recv: {"emeter":{"get_daystat":{"day_list":[{"year":2021,"month":5,"day":21,"energy":0.001000},{"year":2021,"month":5,"day":22,"energy":0},{"year":2021,"month":5,"day":23,"energy":0},{"year":2021,"month":5,"day":24,"energy":0},{"year":2021,"month":5,"day":25,"energy":0},{"year":2021,"month":5,"day":26,"energy":0},{"year":2021,"month":5,"day":27,"energy":0.170000},{"year":2021,"month":5,"day":28,"energy":0.010000}],"err_code":0}}}
Power:         0W
Charge status: B
Enabled:       true
Charged:       0.0kWh
```

thierolm, 2021-05-28:
@xpcone U can download my 4k patch evcc version from this link: [evcc.gz](https://thierolfs-universe.de:5001/d/s/622493046344300758/6ZFwiqOo3IyAtTLX8GWBVKsYO4nsFe8z-.7iAIS6Kowg_)
Pls test. (Link is valid until 30.05.2021)
Be aware U have to unzip it ...

andig, 2021-05-28:
Fixed in https://github.com/andig/evcc/commit/9f19d420d981a6f9d2ae767b3214190bab9d2b97

closed 2021-05-28 by andig

xpcone, 2021-05-28:
Hi,
thanks for the help. Fixed the issue for my version 1 and 4 devices but found the issue on my version 2 ....
HS110
sw_ver:"1.5.6 Build 191125 el.083657" 
hw_ver:"2.0"

There is a overflow issue on the device which caps to 1018-1023 Chars per response..

andig, 2021-05-28:
Please open new issue- not sure how that would be fixed on application side.

thierolm, 2021-05-28:
> Please open new issue

... and provide trace log again