evcc-io/evcc #1100: Influxdb logging is mixing up vehicle SoC values

thierolm, 2021-05-29:
Having two vehicles configured on evcc loadpoint, evcc influxdb logging of SoC is wrong. Wrong SoC is stored for a given vehicle.  

Steps to reproduce the behavior:
1. Configure two vehicles on a loadpoint
2. Start charging the first vehicle for a while an monitor SoC values in influxdb
3. Stop charging the first vehicle, connect the second vehicle and check what happens in influxdb (SoC of second vehicle will be stored for first vehicle. 
4. See screenshots below:

On 17:25h today, I stopped charging my Zoe and connect my Audi A3. evcc started charging in pv mode:
[image]
In influxdb the final 95% SoC for my Zoe, was logged correctly since 17:24h. The grafana query has a WHERE filter to get only  the Zoe vehicle values). But afterwards the lower A3 vehicle SoC was logged:
[image]
By filtering on vehicle A3 ... influxdb provides a completely strange value:
[image]

andig, 2021-05-29:
Ich vermisse das Logfile aus dem hervor geht, ob evcc überhaupt eine Chance hatte die Autos zu erkennen? Welche sind das? Haben die das notwendige Api? Gibts ein Trace log?

thierolm, 2021-05-29:
Ich stelle das nochmal mit trace mode nach und liefere das Log.
Im UI hat evcc den Vehicle Wechsel mitbekommen. Siehe Screenshot. In
Richtung Influxdb dann aber das falsche Vehicle für das Speichern des SoCs
genutzt.

andig, 2021-05-30:
Ich kann nur vermuten, dass das Auto gar nicht erst erkannt wird und das Ganze mit Influx nix zu tun hat

thierolm, 2021-05-30:
Ich habe meinen evcc Entwicklungs-Shelly in eine meiner Garagensteckdosen verabaut, den Audi Steckdosen Charger dort angeschlossen und einen neuen loadpoint nur für den Audi angelegt. Mein Zoe ist jetzt als einziges vehicle für den goe-e charger konfiguriert. Damit ist das logging Problem für mich keins mehr.

Ich werde bei Gelegenheit aber das Infuxdb Logging mit mehreren vehicles nochmal testen, nachstellen und Logs liefern. Denke nicht, dass es an den APIs gelegen hat. Die beiden wurden erkannt.
Ich mach den Issue aber erst mal zu.

closed 2021-05-30 by thierolm

andig, 2021-05-30:
Wäre trotzdem schön die Ursache zu finden- ich kann das bei mir leider nicht diagnostizieren. Logfile wäre deshalb hilfreich.

thierolm, 2021-05-30:
Ich hab' vor mir eine Test-Config zu bauen und den vehicle Switch  zu simulieren. Brauch bloß etwas Zeit dazu.

thierolm, 2021-06-05:
Hallo @andig , ich habe heute eine Test-Config gebaut (siehe unten) und das Influxdb Soc Logging mit zwei EVs an einem Loadpoint getestet.
Gute Nachricht:
Ich konnte den Fehler oben nicht nachstellen. 

Ich habe über die in der Config eingebundenen Files (v1status, v2status) den Wechsel der Fahrzeuge simuliert.

In meiner Influxdb kamen die Logdaten sauber getrennt per loadpoint und vehicle an, so dass man sie in Grafana ordentlich mit einem entsprechenden WHERE Filter darstellen kann.

[image]

```yaml
uri: 0.0.0.0:7071  # uri for ui
interval: 15s  # control cycle interval
# log settings
log: debug
levels:
  core: debug # debug
  lp-1: debug # debug

javascript:
- vm: shared
  script: |
    state = {
      residualpower: 200,
      pvpower: -3000,
      batterypower: 200,
      gridpower: -2000,
      chargepower: 0,
      maxcurrent: 0,
    };
    function get() {
      console.log("state:", JSON.stringify(state));
    }
    function set() {
      console.log(param+":", val);
      console.log("state:", JSON.stringify(state));
    }

meters:
- name: grid
  type: default
  power:
    type: js
    vm: shared
    script: state.gridpower = state.pvpower + state.chargepower + state.residualpower + state.batterypower;

- name: pv
  type: default
  power:
    type: js
    vm: shared
    script: state.pvpower += 100*Math.random();

- name: battery
  type: default
  power:
    type: js
    vm: shared
    script: state.batterypower;
  soc:
    type: js
    vm: shared
    script: "30"
- name: charge
  type: default
  power:
    type: js
    vm: shared
    script: state.chargepower;
  soc:
    type: js
    vm: shared
    script: "30"

chargers:
- name: demo
  type: default
  enable:
    type: js
    vm: shared
    script: |
      set();
      state.enabled = val;
      if (state.enabled) state.chargepower = state.maxcurrent * 230; else state.chargepower = 0;
  enabled:
    type: js
    vm: shared
    script: |
      state.enabled;
  status:
    type: js
    vm: shared
    script: |
      if (state.enabled) "C"; else "B";
  maxcurrent:
    type: js
    vm: shared
    script: |
      set();
      state.maxcurrent = val;
      if (state.enabled) state.chargepower = state.maxcurrent * 230;

vehicles:
- name: demo1
  title: ZOE Experience R135/Z.E. 50
  type: default
  charge:
    source: script
    cmd: /bin/bash -c "cat /home/thierolm/evcc/git/evcc/v1soc"
    timeout: 5s
  status:
    source: script
    cmd: /bin/bash -c "cat /home/thierolm/evcc/git/evcc/v1status"
    timeout: 5s
- name: demo2
  title: Audi A3 Sportback TFSI e
  type: default
  charge:
    source: script
    cmd: /bin/bash -c "cat /home/thierolm/evcc/git/evcc/v2soc"
    timeout: 5s
  status:
    source: script
    cmd: /bin/bash -c "cat /home/thierolm/evcc/git/evcc/v2status"
    timeout: 5s

site:
  title: Demo
  meters:
    grid: grid
    pv: pv
    battery: battery

loadpoints:
- title: Carport
  charger: demo
  meters:
    charge: charge
  vehicles:
  - demo1
  - demo2
  soc:
    poll:
      mode: always

# influx database
influx:
  url: http://localhost:8086
  database: test
  user: evcc
  password: xxxxx
```