evcc-io/evcc #1105: improved notifications aren't grouped

astrakid, 2021-05-31:
hi,
my charger is currently not connected, so i get expected error messages.
with evcc 0.56 these messages should be grouped, but they aren't:

[image]

`[site  ] DEBUG 2021/05/31 21:06:51 ----
[site  ] DEBUG 2021/05/31 21:06:51 pv power: 240W
[site  ] DEBUG 2021/05/31 21:06:51 grid power: 144W
[site  ] DEBUG 2021/05/31 21:06:51 site power: 144W
[lp-1  ] ERROR 2021/05/31 21:06:52 charge timer: dial tcp 10.x.x.112:502: i/o timeout
[lp-1  ] ERROR 2021/05/31 21:06:53 charger: dial tcp 10.x.x.112:502: connect: no route to host
[site  ] DEBUG 2021/05/31 21:07:01 ----
[site  ] DEBUG 2021/05/31 21:07:01 pv power: 240W
[site  ] DEBUG 2021/05/31 21:07:01 grid power: 150W
[site  ] DEBUG 2021/05/31 21:07:01 site power: 150W
[lp-1  ] ERROR 2021/05/31 21:07:02 charge timer: dial tcp 10.x.x.112:502: i/o timeout
[lp-1  ] ERROR 2021/05/31 21:07:03 charger: dial tcp 10.x.x.112:502: i/o timeout
[site  ] DEBUG 2021/05/31 21:07:11 ----
[site  ] DEBUG 2021/05/31 21:07:11 pv power: 230W
[site  ] DEBUG 2021/05/31 21:07:11 grid power: 147W
[site  ] DEBUG 2021/05/31 21:07:11 site power: 147W
[lp-1  ] ERROR 2021/05/31 21:07:12 charge timer: dial tcp 10.x.x.112:502: i/o timeout
[lp-1  ] ERROR 2021/05/31 21:07:13 charger: dial tcp 10.x.x.112:502: i/o timeout
`

naltatis, 2021-05-31:
Grouping doesn't apply here because your get two different messages. But I think we should aim for a state where no errors/warnings are thrown in normal operation. Not quite sure what's a good solution in you case.

astrakid, 2021-05-31:
so grouping occurs only in case of serial order? i would expect to group them in any case, and order the notifications by their occurence.

naltatis, 2021-05-31:
Before the change grouping happened just as you described. Only unique messages where shown.

The issue with that approach is, that it hides errors and makes it hard/impossible to see the timing of events.

astrakid, 2021-05-31:
i understand. but the last timestamp would be enough to see the last occurence (and order grouped errors by timestamp desc).
however, i can live with the situation, usually i don't have this error and shouldn't need it very often (i hope). ;)

closed 2021-05-31 by astrakid

naltatis, 2021-05-31:
Currently evcc's error handling quite rudimentary which is perfectly fine of the software is used by technical people. But if we move to a mainstream audience we need to be more deliberate/sophisticated with our errors/warnings. Some errors may be perfectly fine as long as evcc is able to recover from them in x-time. Others might be more serious (car couldn't be reached for 2 days). Mid-term we need to come up with better and more meaningful representations of these states. Showing them where they make sense for the user.

Exposing the raw messages more clearly in the UI is a first step for me to start these discussions and come up with better ideas.