evcc-io/evcc #1123: MQTT-Client-ID optional auch über die Config vergeben

blenni, 2021-06-09:
In meiner Anwendung (Verbindung über MQTT zu FHEM) habe ich das Problem, dass bei jedem Neustart von EVCC eine neue Client-ID erzeugt wird und damit in FHEM ein neues Device angelegt wird (bzw. im alten Device die Readings nicht mehr aktualisiert werden).

Aktuell wird (wenn ich das richtig verstanden habe) in [evcc/provider/mqtt/client.go/](https://github.com/andig/evcc/blob/master/provider/mqtt/client.go#L23) bei jedem Neustart von EVCC eine zufällige ID berechnet:

```
// ClientID created unique mqtt client id
func ClientID() string {
	pid := rand.Int31()
	return fmt.Sprintf("evcc-%d", pid)
}
```

Es wäre schön, wenn man diese Client-ID optional auch über die config vergeben könnte.

andig, 2021-06-09:
Ahhhh. Das ist by Design- anderenfalls könnten sich mehrere EVCC Instanzen nicht bei einem MQTT Server anmelden. Wir könnten das aber konfigurierbar machen ;)

andig, 2021-06-09:
Magst Du einen PR machen? Die Änderung müsse in setup.go, configureMQTT passieren.

PR #1128 (2021-06-10): Custom MQTT Client ID
Add option to config file to use a fixed MQTT Client ID

Fix #1123
closed 2021-06-10 by andig